alexa-pi/AlexaPiDEPRECATED #4: Some questions

marksev1, 2016-01-17:
What would it take to run this script on a normal netbook with Ubuntu LTS. Of course the button would need to be replaced with a key on the keyboard. Any chance you would support this option also, Sam?

sammachin, 2016-01-18:
The code should work although as you say the GPIO stuff would need to be replaced with another form of input, probably via the keyboard. I'm not sure how easy it would be to have a script listen for keyboard input in the background, you would probably have to run the code in a terminal window and click into that window to trigger the key.

I've already built an alexa client which runs in a web browser (with python back end) I would suggest using this as an easier interface on netbooks, you can either use the version I run on heroku or install your own, running on localhost on Ubuntu should be nice and fast.
https://github.com/sammachin/alexaweb

closed 2016-01-18 by sammachin

marksev1, 2016-01-18:
So I can use it locally?

sammachin, 2016-01-18:
Yeah you can run the webserver locally, you'll need redis installed as well as thats what stores the keys, then just all the python modules in requirements.txt

groetingc, 2016-02-02:
I managed to get it work on my raspberry pi one. However I had some issues with mpg123 which added a 'Splat' at the end of each response.mp3 file. Exchanging mpg123 with mpg321 solved that issue.
One additional question: How would a 'Set Timer' functionality be realized? Looking into the AVS response I just can see Audio data and no parse-able content which could be processed locally on my raspberry (e.g. setting up an alarm timer). What would be You expectation related to such feature?

sammachin, 2016-02-02:
What version of raspbian & mpg123 are you running? And which model of Pi? I
didn't have that problem but would be interested to look into it, the CHIP
version only worked with mpg321.

Regarding the timer Amazon haven't explained things like that yet, I would
expect the JSON that accompanies the response coming back to contain some
relevant data but it doesn't seem to have anything. That way you could
start the timer as soon as the 'starting Now' part of the response has
played. This is just one of the areas where AVS differs from the full Echo
experience  at the moment.

groetingc, 2016-02-02:
The used mpg versions: 
mpg123 1.20.1.
mpg321 version 0.3.2

cat cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 2.00
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 000d
Serial		: 00000000861c82db

My distro:
RASPBIAN JESSIE LITE
Minimal image based on Debian Jessie
Version:November 2015
Release date:2015-11-21
Kernel version:4.1

Btw. setup.sh script has some minor issue in my environment:
I had to change
ip = `ifconfig eth0 | grep "inet addr" | cut -d ':' -f 2 | cut -d ' ' -f 1`
to
ip=`ifconfig eth0 | grep "inet addr" | cut -d ':' -f 2 | cut -d ' ' -f 1`
(It looked like that my shell doesn't like the spaces in the variable assignment
Moreover, I had to manually create the creds.py

One minor change for me is to use a tmpfs location for saving the recording and storing the response rather than write to SD.