alexa-pi/AlexaPiDEPRECATED #9: Main.py Doesn't run with Python 3

chiragd, 2016-02-07:
I guess the code was written for python2.

I'm bypassing the start procedure and calling alexa method to start. Referencing a local recording.wav file I created using the command line.

I've managed to get most of it working however I'm stuck reading the response from Amazon.

File "main.py", line 103, in alexa
    data = r.content.split(boundary)
TypeError: Type str doesn't support the buffer API

Would appreciate a nudge in the right direction.

Thanks!

Macksi, 2016-04-14:
i’m stuck at the same problem. did you manage to solve it by now?

chiragd, 2016-04-14:
Yes u rewrite entirely for pythin 3

Macksi, 2016-04-14:
Would you mind if i take a look at your code? Can't get it done yet.

chiragd, 2016-04-14:
Be my guest. Will have to send it to you later. It's a bit of a mess. It
only runs once because I didn't have a button. Then you have to interrupt
it and try again. It does work though :)

C

Macksi, 2016-04-14:
thanks a lot!

chiragd, 2016-04-14:
Will try in the evening. Can you email me ?

sammachin, 2016-04-30:
Currently still targettig with Python 2.7.9 for this, is there a good reason you need support for 3?

closed 2016-04-30 by sammachin

chiragd, 2016-04-30:
I'm running other things on the Pi such as https://github.com/home-assistant/home-assistant

You can write home assistant skill's for alexa so I wanted to use alexa to turn lights on and off and more.