alexa-pi/AlexaPiDEPRECATED #10: playback cutting off

TW79, 2016-02-08:
After making a request, Alexa begins playback but abruptly stops and the playback light turns off.

TW79, 2016-02-08:
I think playback from the Alexa service is working, but when I ask it to play music which requires streaming (Amazon Music), the stream stops at the end of Alexa's response, and doesn't start streaming music. Is there something I can enable to get this working?

lennysh, 2016-02-27:
There isn't any streaming method added into the code yet.  I'm currently writing a .NET windows application just so I can learn what all is needed to get everything up and running. Once figured out, I can try and add the streaming parts into this code if not already done by then. I do know that a streaming url is returned in the response,  and VLC will more than likely be needed to stream the audio. Amazon does require regular updates on the streaming playback, so that part will also need to be added as well. I'll try to keep everyone updated here on my findings.

lfzguud, 2016-03-01:
Remember when we had to do the setup in the developer site at Amazon?  There was one box for Amazon music that required a check for streaming which reads: "My device supports both HTTP Live Streaming (HLS) and progressive download as means to render music playback. *"  So, I'm guessing that we need to enable the HLS to make this work?

lennysh, 2016-03-01:
Yeah,  I can get VLC to stream the first song,  but then it cuts off. Going to work on it some more today.

lennysh, 2016-03-01:
I think some of that though has to do with not checking back in during the playback.

lfzguud, 2016-03-01:
That might be part of the "progressive download" that it requires.

maso27, 2016-03-08:
I think this might play into Audible support too.  If I ask her to play a book, she says OK, then silence.

lennysh, 2016-03-08:
Indeed it is... the script isn't currently requesting the next audio stream.

lennysh, 2016-03-21:
Well, I've gotten the to-do list and shopping list working!  lol

maso27, 2016-03-21:
That's great! 
Here's hoping you have continued success!

lennysh, 2016-03-22:
Just got to get VLC working now to attempt to get the streaming part working.  With my code changes that allowed the shopping list to work is like 90% of what's needed to get music working.  Just got to find an alternative audio playing app that play's local mp3's and various types of streams including those from m3u files.  VLC for some reason is giving me nothing but trouble...

Anybody successfully get VLC working on their Pi's?

tube0013, 2016-03-22:
On jessie, I did an apt-get install vlc.

than can play m3u streams via the commandline:  

cvlc http://stream.m3u

I have the pi hooked up to a monitor right now, with the X  up and running, and had to select my audio out as Analog before I got any sound.

tube0013, 2016-03-22:
I just tried supping cvlc in for mpg123 for playback for the response and hello.mp3, and I'm getting an error that vlc won't run as root.

lennysh, 2016-03-22:
I'm attempting to use vlc by importing vlc into the script.  If you can get that working in place of mpg123, would be a big help.  I'll try more when I get home.  There's not but so much I can do from work remotely, lol...  I can get VLC to play from SSH to a stream, but can't using the same inside a script.  I can't even get hello.mp3 to play with vlc.

tube0013, 2016-03-22:
Okay, to get around the root issue (long term wish this script did not have to run as root):

sed -i 's/geteuid/getppid/' /usr/bin/vlc

I hard coded the analog output in the string as well these are the 2 lines in main.py, and I'm successfully getting the hello.mp3 and responses out.

		os.system('cvlc -A alsa,none --alsa-audio-device hw:CARD=ALSA,DEV=0 {}response.mp3'.format(path, path))

...

	os.system('cvlc -A alsa,none --alsa-audio-device hw:CARD=ALSA,DEV=0 {}hello.mp3'.format(path, path))

lennysh, 2016-03-22:
Try it with the Python Binding.  I got the vlc.py script from here:

http://git.videolan.org/?p=vlc/bindings/python.git;a=tree;f=generated;b=HEAD

lennysh, 2016-03-22:
Def going to try using your method until then though.  Thanks...

lennysh, 2016-03-22:
I bet this might be my answer...

```
import vlc
i = vlc.Instance('--aout=alsa', '--alsa-audio-device=front:CARD=Speakers,DEV=0')
mrl = 'file:///home/user/folders/sample.mp3'
m = i.media_new(mrl)
mp = m.player_new_from_media()
mp.play()
```

tube0013, 2016-03-22:
I'll give that a whirl.  and let you know how far I get.  I have my pi with me at work and it's slow, so I have some time.

lennysh, 2016-03-22:
try this code.  I just tried it remotely, and didn't get any error's this time, but have no idea if it's playing or not, lol.

```
import vlc
def play_mp3(file):
	i = vlc.Instance('--aout=alsa,none', '--alsa-audio-device=hw:CARD=ALSA,DEV=0')
	p = i.media_player_new()	
	mrl = "{}{}".format(path, file)
	m = i.media_new(mrl)
	p.set_media(m)	
	mm = m.event_manager()
	mm.event_attach(vlc.EventType.MediaStateChanged, state_callback, p)
	p.play

def state_callback(event, player):
	print("Player State: {}".format(state))
```

tube0013, 2016-03-22:
That runs with no error, but I get no audio out put.

running something like this does generate output:

>>> p = vlc.MediaPlayer("file:///path/to/track.mp3")
>>> p.play()

lennysh, 2016-03-22:
I tried that method yesterday but kept getting an error...  ugh...  I need to bring my Pi to work!  lol

lennysh, 2016-03-22:
so does the following work for you then? (in place of the play_mp3 line.)

```
p = vlc.MediaPlayer("file://{}response.mp3".format(path))
p.play()
```

tube0013, 2016-03-22:
I got your code to work, it was missing a () on the p.play

def play_mp3(file):
    i = vlc.Instance('--aout=alsa,none', '--alsa-audio-device=hw:CARD=ALSA,DEV=0')
    p = i.media_player_new()    
    mrl = "{}{}".format(path, file)
    m = i.media_new(mrl)
    p.set_media(m)  
    mm = m.event_manager()
    mm.event_attach(vlc.EventType.MediaStateChanged, state_callback, p)
    p.play()

def state_callback(event, player):
    print("Player State: {}".format(state))

I do get a lot of errors out in the log, re the callback state, but I'm guessing that's code you have in your program that I'm missing:

Traceback (most recent call last):
>>>   File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/root/AlexaPi/vlc.py", line 1432, in _callback_handler
    call(event.contents, *args, **kwds)
  File "/root/AlexaPi/test.py", line 29, in state_callback
    print("Player State: {}".format(state))
NameError: global name 'state' is not defined
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/root/AlexaPi/vlc.py", line 1432, in _callback_handler
    call(event.contents, *args, **kwds)
  File "/root/AlexaPi/test.py", line 29, in state_callback
    print("Player State: {}".format(state))
NameError: global name 'state' is not defined
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/root/AlexaPi/vlc.py", line 1432, in _callback_handler
    call(event.contents, *args, **kwds)
  File "/root/AlexaPi/test.py", line 29, in state_callback
    print("Player State: {}".format(state))
NameError: global name 'state' is not defined

lennysh, 2016-03-22:
I feel like an idiot...  Been looking at this code all day. lol...

Here's my output now (on screen at least)
```
Checking Internet Connection...
Connection OK
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
```
Def looks promising...  Hopefully I'll have music working shortly then!

tube0013, 2016-03-22:
Awesome.  Happy to help.

tube0013, 2016-03-22:
LOL just got an Amazon Developer email, saying they've made some changes/updates to the Alexa Service:

> We are excited to announce an important new update to the Alexa Voice Service (AVS) that will enable you to enhance the user experience on your Alexa-enabled products. Last year, we launched a developer preview of AVS to introduce you to the benefits of voice-powered experiences. With this update, we are making architectural improvements that include updated APIs and message structures, Amazon Alexa App (iOS and Android) support and the ability to send server-initiated messages.
> 
> Here are some of the new features:
> 
> - Volume Control: Ability to adjust and mute/unmute volume on your product using voice or buttons.
> - Media Control: Ability to control media playback (e.g. play/pause) on your product using voice or buttons.
> - Timers and Alarms: Ability to create and manage timers/alarms on your product using voice.
> - Amazon Alexa App Enhancements: Control volume, media playback (e.g. Amazon Music, Audible), as well as manage timers/alarms on your product using the Amazon Alexa App.

tube0013, 2016-03-22:
this just went up to today, a java implementation including music using vlc for playback, it looks like it uses a gui to trigger teh recording, not a button press.

https://github.com/amzn/alexa-avs-raspberry-pi

lennysh, 2016-03-22:
Yeah I seen the email, but not the github link...  Very interesting...

lennysh, 2016-03-22:
Just played my first song from Amazon Prime Music!

lfzguud, 2016-03-22:
Ok, I've been following - now I'm uber excited!
So other than installing the VLC, was there anything else you had to do?

lennysh, 2016-03-22:
Yes, some of the changes have already been added to my music-patch-1 branch on my fork. I'll do a full pull request once completed.

lfzguud, 2016-03-22:
Your work on this is really outstanding.  Once you're done, I'll give it a
try on mine. Again, thanks for the work on this!

lennysh, 2016-03-23:
Well, I seem to have music working fluently and "most" of the progress reporting working (going to have to use timers to do the rest).  One thing I am noticing when playing the Amazon Prime Music streams is an error coming from vlc.  Here's the output I'm seeing:
```
Checking Internet Connection...
Connection OK
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Player State: State.Opening
[757e5508] httplive stream: HTTP Live Streaming (d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=219142429&c=cf&f=ts&t=10&bl=256k&s=true&e1=1458746100000&e2=1458747000000&v=V2&h=a377c8f0cebf4a2d93bd6ea04cfb94e2457ecadc44f0e28bb38e0459eadf551a)
[757f05a0] ts demux: MPEG-4 descriptor not found for pid 0x100 type 0xf
Player State: State.Playing
[757ecd18] packetizer_mpeg4audio packetizer: AAC channels: 2 samplerate: 44100
[757f05a0] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[757f05a0] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[757f05a0] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
```

Anyone have any idea's?  It may not be something that I can fix.  Doesn't appear to affect the performance of the audio output though, as the music seems to play just fine.  I'd hate for that to constantly flood someone's log file though.

tube0013, 2016-03-23:
If it's playing I'd not worry, wonder if there is anything in the vlc Python api to limit the output.

The sample implementation from Amazon used vlc for playback as well.

Does it work with any other services - Pandora etc?

lennysh, 2016-03-23:
Neither Pandora nor Spotify are supported through the API (yet?).  I've not tried iHeartRadio yet, but it did seem to attempt to try to play from TuneIn but failed yesterday. I think they use a M3U format, so it may be related to that directly.  That was in my early testing though.  I'll try again as soon as I get home.

I also cannot test Audible (no subscription), but in my [AlexaDotNet](http://www.lennysh.com/alexadotnet/) implementation, Kindle Books worked fine without any additional coding.  So I'll have to try that today as well.

lennysh, 2016-03-23:
I'll be merging my music changes into my music-patch-1 on my fork tomorrow AM EST. Got to make the code as readable as possible and remove the unneeded code. Just please keep in mind that I'm not a professional python developer. Lol

The only issues I am having right now that I'm still trying to fix are: 

1. Slow responses between shopping/to-do items. Was faster with mpg123. Might be best to use mpg123 for mp3 responses and only use VLC for streaming responses. 
2. TuneIn doesn't stream. Seems to be related to M3U links not currently working. Not sure if it's an issue with the binding or my coding, lol.

I'm sure there's more, so I'll add later as it comes to me, lol.

lfzguud, 2016-03-23:
Lenny - Thanks!  Soon as you post it I'll set up a new pi with it to test
it out and see what I can debug regarding the VLC entries in the log.
Really excited about this one!

lennysh, 2016-03-24:
I'm loving it...  Really hoping others can help improve my code for sure. It'll help me improve my skills. 

As for the VLC errors, iHeartRadio doesn't have them. Only Amazon Music.

lennysh, 2016-03-24:
I've merged my changes into my music-patch-1 branch located [here](https://github.com/lennysh/AlexaPi/tree/lennysh-music-patch-1).

I'm not going to send a pull request yet because Sam is going to update his script to work with the new API first.  Once that is done, and I get my side updated the same way, then I'll do a pull request unless asked otherwise.

lfzguud, 2016-03-25:
Lenny, I've got mine running with your changes and it plays Amazon Prime
music fine.  I realize that Pandora isn't supported (as Echo says) in the
current API.  I did try Tune-in "Play WWJ from Tune-in" and she responded
as if she was going to play it - but then nothing.
I played a few songs from Amazon Prime, then stopped them.  Interestingly,
after I'd stop one, it would show the following in my VSSH window upon
playing the next song.  Could that be overflow
because I stopped it before completion?  Or, could it be it was trying to
still send through the Tune-in while the next song was playing?
Regardless, I stopped it (Ctrl C) then rebooted, restarted the main, and
she's working fine and playing Prime music just fine.

[image: Inline image 1]

lennysh, 2016-03-25:
Yeah, TuneIn still has issues. It almost gets in an endless loop and can't stop without ctrl-c.

lennysh, 2016-03-25:
What's weird is that TuneIn just returns an M3U file that VLC plays fine via command line.  It must be an issue with the VLC binding itself. Going to have to do some research on that one.

lfzguud, 2016-03-25:
Yeah, I'm not getting any luck with iHeartRadio either.  She acknowledges
the request just fine, and it looks like she wants to play it - but -
nothing.  Still, even just having Prime playing is leaps and bounds over
where it was!

lfzguud, 2016-03-25:
Lenny, here's another possibility with why mine isn't doing iHeart or
Tune-in - I'm doing it on an A+ (strictly for the footprint size).
I'm going to do the same build on a Pi2 B tomorrow and see if the extra
processing power/speed has anything to do with it.

lennysh, 2016-03-25:
iHeartRadio works great on my Pi2B

lfzguud, 2016-03-25:
Yup, I'm betting that's exactly it.  Gonna switch it over now and see.

lfzguud, 2016-03-25:
Nope, that wasn't it.  No problem though - I'll wait for the final version
and give it a try then.

flooie, 2016-04-05:
@lennysh tunein - doesn't return an m3u file for me - any guesses why?

lennysh, 2016-04-05:
Still looking into this issue

flooie, 2016-04-05:
@lennysh thanks

sammachin, 2016-05-01:
I've integrated lennys vlc code and refactored the response parser, it now works with some tune in stations but it's a bit of a hack, I've tested with BBC stuff but I expect others will break so let me know which stations work and which don't, it's all in the version1.1 branch which I'll promote to master in the next few days.

closed 2016-05-01 by sammachin

Rian9771, 2016-09-01:
@sammachin , is there any effort going on to integrate Spotify into AlexaPi? I am using @flooie 's rep. It would be sweet to talk to spotify and play music. Apologies in advance if i am not posting at the right place.

sammachin, 2016-09-01:
@Rian9771 
I havn't looked at Spotify yet but I suspect it won't be possible without some commercial arrangement due to things like licensing and DRM.

FYI next time better to post this isn a new issue than using an existing one thats not about the subject.