alexa-pi/AlexaPiDEPRECATED #11: Installation Error

atonu, 2016-02-12:
Hello,
I am installing AlexaPi for the first time.
I am getting this error all the time i try installing AlexaPI

_________________________________
Object "=" is unknown, try "ip help".
python: can't open file './auth_web.py': [Errno 2] No such file or directory
Open http://:5000
You can now reboot
__________________________________

please help me. let me know where i am going wrong.

i am installing as super user.

groetingc, 2016-02-13:
setup.sh script has some minor issues in my environment:
I had to change
ip = ifconfig eth0 | grep "inet addr" | cut -d ':' -f 2 | cut -d ' ' -f 1 (In backticks)
to
ip=ifconfig eth0 | grep "inet addr" | cut -d ':' -f 2 | cut -d ' ' -f 1 (In backticks)
(It looked like that my shell doesn't like the spaces in the variable assignment
Moreover, I had to manually create the creds.py

TW79, 2016-02-15:
If you're using a WiFi adapter, you'll need to change setup.sh to ip=ifconfig wlan0 | ...

lennysh, 2016-03-23:
If this has been resolved, please close this issue.

Stephen-Neal, 2016-03-30:
I have the same issue. Latest Raspbian install on an RPi 2 (30th Mar 2016)
Have done an apt-get update and upgrade.
Have logged in as root by sudo su and moved to /root using cd /root
I get the Object "=" is unknown, try "ip help" error too.

If I remove the spaces as suggest by groetingc I get 
./setup.sh : line 31 : unexpected EOF whilst looking for matching ' '
./setup.sh : line 37 : syntax error: unexpected end of file

Help appreciated.

sammachin, 2016-04-30:
The setup.sh and authweb.py scripts are much improved in 1.1 to make things a bit easier, you can either grab those files from the 1.1 branch and try now or sit tight for the proper 1.1 update which has lots of changes in the alexa client too.

Please don't try and install and run the entire 1.1 branch as its still WIP and almost certainly broken!

closed 2016-04-30 by sammachin