alexa-pi/AlexaPiDEPRECATED #12: 500 error on /code?

elvisthedj, 2016-02-21:
I go to serverip:5000 and get redirected to Amazon login. I enter credentials and am then taken to the "click Okay" screen. Amazon then gives me this error:

We're sorry!
An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly.

I've waited, restarted browser, computer, pi, etc...same error.

I tried going to serverip:5000/code and when I do I get a 500 error:

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/_cprequest.py", line 670, in respond
    response.body = self.handler()
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/lib/encoding.py", line 217, in __call__
    self.body = self.oldhandler(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/_cpdispatch.py", line 60, in __call__
    return self.callable(*self.args, **self.kwargs)
  File "./auth_web.py", line 29, in code
    code = urllib.quote(cherrypy.request.params['code'])
KeyError: 'code'

Not sure if that is why Amazon can't do it's thing, so I thought I would ask to see if that is normal behavior.

HumbertoL, 2016-03-04:
Make sure you have serverip:5000/code in the Allowed Return URLs section of the security profile.

lennysh, 2016-03-23:
Has this been resolved?  If so, please close this request.

elvisthedj, 2016-03-23:
Hi Lenny -

No, I have the same issue. I received a denial note from Amazon since I had chosen to use Amazon music, so I am going to try to make a new project without selecting the music option and see if that works.  Perhaps it was all configured right, it was just pending their approval.

thesingingmoose, 2016-04-05:
Did you find a solution to this? I am having the exact same problem

joetech, 2016-04-06:
Make sure your ProductID in creds.py (and everything else in there) match your profile in your Amazon developer account.  I changed this to match and it worked.

sammachin, 2016-04-30:
We really need a set of screenshots or a walkthough video on setting up your Amazon account to make this easier for people, any volunteers to create one?

closed 2016-04-30 by sammachin