alexa-pi/AlexaPiDEPRECATED #13: "UnBoundLocalError : local variable 'inp'referenced before assignment " while pressing the button in GPIO 18

SoorajKrishna, 2016-02-27:


HumbertoL, 2016-03-04:
This seems to happen when the button is not connected properly. You get this error when Pin 18 is "floating". Make sure you have it connected correctly to the button. Refer to this diagram to make sure you connected the pins to the correct pins on the button itself.

[image]

RamblingGeekUK, 2016-03-25:
This also seems to happen if you have the speaker connected to the Audio out of the sound card rather then the jack on the Pi

doubledrat, 2016-03-26:
I've found it happens if you press the button before "Alexa's ready" :-)

seriously though, when it says "hello", it's not quite ready for input.  In my experience, waiting for about 3-5 secs before pressing the button and you don't get this error

sammachin, 2016-04-19:
Not sure why it would occur using the sound card output instead of the audio jack, I've tested with both and not had and issue.

@doubledrat is right the Hello does trigger a little early, there's also 3 flashes of the green LED whcih is just after the hello and is the better indication that its ready.

@SoorajKrishna does this happen all the time for you or just occasionally?

closed 2016-04-30 by sammachin