alexa-pi/AlexaPiDEPRECATED #16: Apt-get error after alexa install

lennysh, 2016-03-05:
I am getting the following when I try to do an apt-get upgrade on my pi sheet installing the alexa service...

Reading package lists... Done
Building dependency tree
Reading state information... Done                                                   Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.                      3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.                    Do you want to continue? [Y/n] y
Setting up bluez (5.23-2+rpi1) ...                                                  insserv: warning: script 'S99alexa' missing LSB tags and overrides
insserv: warning: script 'alexa' missing LSB tags and overrides                     insserv: There is a loop at service alexa if started
insserv: There is a loop between service plymouth and mountnfs if started           insserv:  loop involving service mountnfs at depth 4
insserv:  loop involving service networking at depth 3                              insserv: There is a loop between service plymouth and udev if started
insserv:  loop involving service udev at depth 1                                    insserv: There is a loop between service plymouth and mountall if started
insserv:  loop involving service mountall at depth 4                                insserv:  loop involving service checkroot-bootclean at depth 3
insserv: There is a loop at service plymouth if started                             insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!                                                              insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!                                                              insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!                                                              insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!                                                              insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Max recursions depth 99 reached
insserv:  loop involving service memcached at depth 1
insserv:  loop involving service kbd at depth 8
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
[... 10 lines cut ...]
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
[... cut ...]
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!
insserv: exiting now without changing boot order!
update-rc.d: error: insserv rejected the script header
dpkg: error processing package bluez (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of bluetooth:
 bluetooth depends on bluez; however:
  Package bluez is not configured yet.

dpkg: error processing package bluetooth (--configure):
 dependency problems - leaving unconfigured

Any ideas?

lennysh, 2016-03-07:
Fixed with my pull request...

ck37, 2016-04-14:
Thanks, I ran into this issue too and can confirm that the pull request resolves the error.

sammachin, 2016-04-19:
merged

closed 2016-04-19 by sammachin

larryhackett, 2016-05-13:
Sorry - how to fix this with a pull request? Please, can I get more detail, as I have same issue. thanks so much.

sammachin, 2016-05-13:
Hi @larryhackett 

I've merged the pull request so if you try installing from the master branch of my repository that should now fix the error.

larryhackett, 2016-05-13:
Perfect! Many thanks now it's fixed. Appreciate the help!

youneverknow1, 2016-10-17:
I reinstalled the AlexaPi cloning the master branch, but I am unable to get rid of udev error..
`Setting up udev (215-17+deb8u5) ...
addgroup: The group `input' already exists as a system group. Exiting.
update-initramfs: deferring update (trigger activated)
insserv: warning: script 'S99alexa' missing LSB tags and overrides
insserv: warning: script 'alexa' missing LSB tags and overrides
insserv: There is a loop at service alexa if started
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Max recursions depth 99 reached
insserv: There is a loop between service minidlna and mountnfs if started
insserv:  loop involving service mountnfs at depth 8
insserv:  loop involving service networking at depth 7
[... cut ...]
insserv: exiting now without changing boot order!
update-rc.d: error: insserv rejected the script header
dpkg: error processing package udev (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for initramfs-tools (0.120+deb8u2) ...
Errors were encountered while processing:
 udev
E: Sub-process /usr/bin/dpkg returned an error code (1)
`
I am unable to install any more packages.. nor upgrade to Raspbian pixel..  how do I get rid of this issue?

maso27, 2016-10-17:
Try df -h and see if you have any space left on your SD card.

youneverknow1, 2016-10-17:
Yes there's plenty. ~20gb left.

maso27, 2016-10-18:
So where are these errors cropping up?  During install, while _setup.sh_ runs?  After typing _python main.py_?

And you say that you can't install more packages.  You mean _apt-get_ has stopped working?

All these errors seem unique.  I'm surprised that minidlna is coming up at all.  Are you running other things on this Pi as well?  If not, it might be worth re-imaging the Pi and running setup.sh again.

youneverknow1, 2016-10-19:
Yes apt-get has stopped working. I am also running few other programs on my Pi - CUPS (for cloud printing), pi-hole etc. I uninstalled minidlna but the error is still there. Is there any way I could cleanup AlexaPi?

maso27, 2016-10-19:
How did you uninstall minidlna if _apt-get_ isn't working?  
Have you also tried disabling it: `sudo systemctl disable minidlna`?

"Cleaning up" AlexaPi could be quite complicated, because other things that you already have running might depend on things that _setup.sh_ installed for AlexaPi.  But you can try debugging some stuff.

You used the master branch of sammachin/AlexaPi, right?  Then you can see if things clear up first by just killing the python process.

So, `ps ax | grep main.py` to see if it's running.
(Note that this will also show the grep process itself, which isn't an issue).
Then, `sudo kill xxx`, where xxx is whatever process number was running main.py.
Another `ps ax | grep main.py` should show that it's gone.
(Note that it will try to restart after a reboot, so you might have to kill it again.)

Try apt-get and see if things are improved.
If it fixes things and you just want AlexaPi to stop auto-starting, you can disable that by `update-rc.d -f AlexaPi remove`.

Another thing to do is consider how your audio is configured.  What are you using for in/out?  By default it expects a USB mic and speakers at the 3.5mm jack.

youneverknow1, 2016-10-22:
No luck!
Tried systemctl command here are the error messages:

`Synchronizing state for minidlna.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d minidlna defaults
insserv: warning: script 'S99alexa' missing LSB tags and overrides
insserv: warning: script 'alexa' missing LSB tags and overrides
insserv: There is a loop at service alexa if started
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Starting alexa depends on minidlna and therefore on system facility `$all' which can not be true!
insserv: Max recursions depth 99 reached
insserv: There is a loop between service minidlna and mountnfs if started
insserv:  loop involving service mountnfs at depth 8
insserv:  loop involving service networking at depth 7
insserv:  loop involving service alsa-utils at depth 10
[... cut ...]
insserv:  loop involving service mountall-bootclean at depth 1
insserv:  loop involving service kbd at depth 10
insserv:  loop involving service urandom at depth 8
insserv:  loop involving service alexa at depth 1
insserv: There is a loop between service minidlna and mountnfs-bootclean if started
insserv:  loop involving service mountnfs-bootclean at depth 8
insserv: exiting now without changing boot order!
update-rc.d: error: insserv rejected the script header
*** Error in `systemctl': double free or corruption (fasttop): 0x81214fc8 ***
`