alexa-pi/AlexaPiDEPRECATED #18: Raspberry pi3

Nauttor, 2016-03-08:
Hi Sam, 

I get AlexaPi working on a new pi3 with the jessie-Lite but not without some workarounds.

First of all, It´s a bit confuse that you need to install it on /root as root, because by default on the Raspbian root user can´t log in. You need to login with "pi" user, then sudo bash, and go on with the install. 

Then when I execute the setup.sh it does not finish properly as others said in another issue. 
Finally i proceed with the "advanced" install and place the credentials IDs on the creds.py and execute the auth_web.py manually. Then I access http://myRaspIP:5000 and it show an amazon page, then just click ok and it finish properly. 

Finally i start alexa via /etc/init.d/alexa and the leds blink as expected and i hear HELLO. So alexa it´s running. I´ll try some voice commands and it reply to me, so all appears to be fine.

So, first of all thank for your work. 

Now my question is, on the developers area i read that you can use the android Alexa app to interact and setup alarms and other things, but even if the alexa Android app and alexa.amazon.com show my AlexaPi as a device, it always said NOT CONNECTED. I don´t know if its something i´m doing wrong or if the Alexa Pi it´s not intended to work with the Android App or even the web. 

Thank for your support.

Regards.

Nauttor, 2016-03-10:
No one made it work from the alexa app or portal?

lennysh, 2016-03-23:
It works with the Alexa app just fine.  Alarms and Timers were on;y just added to a newer version of the AVS API itself.  This script hasn't been updated to use the new API yet, and thus cannot use the Alarms and Timers part of the app yet.

sammachin, 2016-04-18:
Yep alarms and timers coming in v1.2 or maybe v2 depending how I get on with the new API, 
The setup script is 'sub optimal' and will be a focus for improvement.
I've just got an RPi3 so I'll be testing on that too but I don't think there's any real differences

closed 2016-04-18 by sammachin