alexa-pi/AlexaPiDEPRECATED #20: Music

tube0013, 2016-03-09:
Is is possible to get Amazon's Music Services working with the script?  Could you use something like ffmpeg or vlc to stream handle the streams?

lennysh, 2016-03-09:
It's being worked on by me at least...

groetingc, 2016-03-10:
Could You please explain me in some words what I need to do that Music-Service is also working for me?

tube0013, 2016-03-11:
I got approved for Music on my amazon dev account, so if there is any code to test I'd be happy to try.

SpiderSpartan, 2016-03-16:
Amazon has an [Alexa service implementation guide](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/docs/reference-implementation-guide), I haven't had a chance to dig into this, but it does reference and walk you through installing VLC on linux.  If I ever do get around to reading through all this and get music working I'll share what I find.

dliloch, 2016-03-17:
I too got approved for Music on my amazon dev account and would be happy to try. Looking at the implementation guide it appears that you have to create a "Music Event" then I guess you would 
point it to VLC. VLC is installed on my PI but I don't know how to code the linkage between the main.py code and the VLC.. If I ask Alexa to play a song "she" says fine but of course nothing. Before I was approved I would get an unauthorized for this service message...

lennysh, 2016-03-17:
Music is very tricky as every track has to be requested individually. And you don't need to request the next track until the previous track is done. I've got this working in my .Net version already.

dliloch, 2016-03-17:
this is the article I was referring to:
https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/rest/audioplayer-events-response#Audio%20Item%20JSON%20Object

lennysh, 2016-03-17:
The other issue is the progress reporting. It has set parameters on when to do the first report, and how far apart the next reports need to be.

lennysh, 2016-03-22:
Just played my first song from Amazon Prime Music! 

Now just to get the progress reporting working so the next song will play properly.

dliloch, 2016-03-22:
Wow...amazing.. It is Python I assume... Can I try it?

dliloch, 2016-03-22:
Did you see this email from Amazon..

Valued AVS Developer,

We are excited to announce an important new update to the Alexa Voice Service (AVS) that will enable you to enhance the user experience on your Alexa-enabled products. Last year, we launched a developer preview of AVS to introduce you to the benefits of voice-powered experiences. With this update, we are making architectural improvements that include updated APIs and message structures, Amazon Alexa App (iOS and Android) support and the ability to send server-initiated messages.

Here are some of the new features:

Volume Control: Ability to adjust and mute/unmute volume on your product using voice or buttons.
Media Control: Ability to control media playback (e.g. play/pause) on your product using voice or buttons.
Timers and Alarms: Ability to create and manage timers/alarms on your product using voice.
Amazon Alexa App Enhancements: Control volume, media playback (e.g. Amazon Music, Audible), as well as manage timers/alarms on your product using the Amazon Alexa App.
Learn how to integrate Alexa into your product with the Getting Started Guide. If you already have a solution, see the API Migration Guide(v20160207).

Have Questions? We are here to help! Visit us on the AVS Forum to discuss specific questions with one of our experts.

Alexa is always learning, and there is a lot more to come!

- Alexa Voice Service Team

lennysh, 2016-03-22:
It's the same script but modified. 

I got to iron out a few issues, and then I'll post the code.

dliloch, 2016-03-22:
Thanks so much..

dliloch, 2016-03-23:
Lenny,
One thing I noticed with the existing code is it sort of does a spin loop on the button GPIO pin. I wonder if it could wait on the pin to change state..on a pi 2 one of the cores is pegged constantly...ie. Instead of polling use GPIO.wait_for_edge(18,GPIO.RISING)?

lennysh, 2016-03-23:
I think someone already posted a pull request making this very change.  I still need to look over it myself to see how it performs in comparison. I'm also assuming Sam hasn't had much of a chance to look it over either in order to merge it.

lennysh, 2016-03-24:
I've merged my changes into my music-patch-1 branch located [here](https://github.com/lennysh/AlexaPi/tree/lennysh-music-patch-1).

There is a boolean at the top to enable debug mode if you have any issues.  Please enable it prior to uploading any script output for troubleshooting.

I'm not going to send a pull request yet because Sam is going to update his script to work with the new API first.  Once that is done, and I get my side updated the same way, then I'll do a pull request unless asked otherwise.

dliloch, 2016-03-24:
downloaded the code and redid the setup script. Then I tried it and it
said connect OK but I did not hear the voice. I then asked for the time
and it took about a minute to come back. I did not have this issue with the
original script, which I still have and it still works fine (it is in
another directory). I asked ALEXA to play fleetwood mac and it appears to
connect but no music. I set debug=1 and I've attached the  output.
Please let me know if you need any further diagnostics.
Thanks for all your effort here ... this is beyond my coding capabilities
.. but I can contribute by testing...
this is running on a raspy-3 using the wii-fi connection..

-don

comparing to the main.py without music...
alexa's response to questions very slow
when I first start it up python main.py .. it connects but I do not get the
audio on the hello.mp3..
I asked to play fleetwood mac and as you can see it appeared to connect but no
sound ..

pi@raspialexa:~/Downloads/AlexaPi-lennysh-music-patch-1 $ python main.pyChecking Internet Connection...
Connection OK
Play_Audio Request for: hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"directives":[{"namespace":"SpeechSynthesizer","name":"speak","payload":{"contentIdentifier":"amzn1.as-ct.v1.Domain:Application:Notifications#ACRI#DeviceTTSRenderer_07573134-f5e3-4bbe-8ce1-0232a2ec865b","audioContent":"cid:DeviceTTSRenderer_07573134-f5e3-4bbe-8ce1-0232a2ec865b_1050288615"}}]}}
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"directives":[{"namespace":"SpeechSynthesizer","name":"speak","payload":{"contentIdentifier":"amzn1.as-ct.v1.Domain:Application:Knowledge#ACRI#KnowledgePrompt.f40c15bf-1aa7-4f7b-a4d6-7d44f0765581","audioContent":"cid:KnowledgePrompt.f40c15bf-1aa7-4f7b-a4d6-7d44f0765581_664830369"}}]}}
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"directives":[{"namespace":"SpeechSynthesizer","name":"speak","payload":{"contentIdentifier":"amzn1.as-ct.v1.Domain:Application:Knowledge#ACRI#KnowledgePrompt.487ce0c6-a3aa-400a-a75d-79410c9b2eb0","audioContent":"cid:KnowledgePrompt.487ce0c6-a3aa-400a-a75d-79410c9b2eb0_1863796510"}}]}}
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"directives":[{"namespace":"SpeechSynthesizer","name":"speak","payload":{"contentIdentifier":"amzn1.as-ct.v1.Domain:Application:Music#ACRI#DeviceTTSRendererV4_ac21c594-3b04-4276-a3d8-3326ffdacce9","audioContent":"cid:DeviceTTSRendererV4_ac21c594-3b04-4276-a3d8-3326ffdacce9_728356133"}},{"namespace":"AudioPlayer","name":"play","payload":{"audioItem":{"streams":[{"streamUrl":"https://d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=203496952&c=cf&f=ts&t=10&bl=256k&s=true&e1=1458835200000&e2=1458836100000&v=V2&h=8c3885d96034192d43a153f41b32a72149f2ba0ee035e3789daa22e02bca2e2b","offsetInMilliseconds":0,"expiryTime":"2016-03-24T15:06:57+0000","progressReport":{"progressReportDelayInMilliseconds":1000,"progressReportIntervalInMilliseconds":0},"progressReportRequired":true,"streamId":"amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#3d81d0bf-d958-44d7-89c3-07aa3d2a11ff:54"}],"audioItemId":"amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#3d81d0bf-d958-44d7-89c3-07aa3d2a11ff:54"},"navigationToken":"amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#3d81d0bf-d958-44d7-89c3-07aa3d2a11ff:54","playBehavior":"REPLACE_PREVIOUS"}}]}}
navigationToken: amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#3d81d0bf-d958-44d7-89c3-07aa3d2a11ff:54
streamUrl: https://d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=203496952&c=cf&f=ts&t=10&bl=256k&s=true&e1=1458835200000&e2=1458836100000&v=V2&h=8c3885d96034192d43a153f41b32a72149f2ba0ee035e3789daa22e02bca2e2b
streamId: amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#3d81d0bf-d958-44d7-89c3-07aa3d2a11ff:54
playBehavior: REPLACE_PREVIOUS
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Sending Playback Progress Report Request...
Player State: State.Ended
Sending Playback Progress Report Request...
 Play_Audio Request for: https://d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=203496952&c=cf&f=ts&t=10&bl=256k&s=true&e1=1458835200000&e2=1458836100000&v=V2&h=8c3885d96034192d43a153f41b32a72149f2ba0ee035e3789daa22e02bca2e2b
[... 3 lines cut ...]
[74419c50] httplive stream: HTTP Live Streaming (d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=203496952&c=cf&f=ts&t=10&bl=256k&s=true&e1=1458835200000&e2=1458836100000&v=V2&h=8c3885d96034192d43a153f41b32a72149f2ba0ee035e3789daa22e02bca2e2b)
Recording...
Player State: State.Stopped
Recording Finished.
Sending Speech Request...
Processing Request Response...
[... cut ...]
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Sending Playback Progress Report Request...
Player State: State.Ended
Sending Playback Progress Report Request...
Play_Audio Request for: https://d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=258196818&c=cf&f=ts&t=10&bl=256k&s=true&e1=1458835200000&e2=1458836100000&v=V2&h=6fe8ccce6c53dafa52946f2272ffecea11f4d2635b2fd218d95326739cd96c53
Playback Progress Report was Successful!
 Player State: State.Opening

lennysh, 2016-03-24:
Does running it with sudo help?

lennysh, 2016-03-24:
Did you run the setup script?  Or did you manually install the new requirements?

dliloch, 2016-03-24:
I ran the setup script .. sudo ./setup.sh and then rebooted ..

lennysh, 2016-03-24:
I'm assuming then it also installed the service? Did you stop the service?

dliloch, 2016-03-24:
I could run the script again and capture the output if that would help...

lennysh, 2016-03-24:
You may have two Alexa services running now all trying to do the same thing

maso27, 2016-03-24:
I just updated, and so far it's working great!
She is a bit slower, but right now I'm grooving to some random jazz. I'm waiting for a second song to come around.

Here's how I did the update (not a reinstall):
1.) killall python, to stop the script
2.) place the new requirements.txt, setup.sh, and main.py in the AlexaPi folder
3.) apt-get update, apt-get upgrade, apt-get install vlc
4.) copy and paste the "wget..." line from the top of setup.sh
5.) "pip install-r requirements.txt" from setup.sh
6.) reboot

Great work!

lennysh, 2016-03-24:
Is the audio device listed differently on the RasPi3?  I've yet to purchase one but wouldn't of thought that it would have been different...  Might be why @dliloch is having an issue with audio playback...

dliloch, 2016-03-24:
cleared out creds.py .. reinstalled , rebooted ... and bingo .. I'm good ..
response is fine now... thanks amazing ..
:-)
wow

lennysh, 2016-03-24:
Glad to hear it @dliloch !

dliloch, 2016-03-24:
the output is coming out of my HDMI .. how can I redirect to the USB audio
dongle which goes to my Bose audio ...
The original main.py went out through the usb audio ...

lennysh, 2016-03-24:
The following line is the one used to determine the audio output device.  We need to tweak that line (second set of quotes) for your needs.

`i = vlc.Instance('--aout=alsa', '--alsa-audio-device=hw:CARD=ALSA,DEV=0')`

lennysh, 2016-03-24:
@dliloch What's the output of the below command?

`aplay -l`

maso27, 2016-03-24:
Yeah I can't help on audio.  I started from an install of Pi MusicBox so I could get Mopidy and AirPlay support as well.  You can configure the audio output from within the MusicBox webGUI.
But if you've got a desktop installed, can't you just open the ALSA mixer and set it from there?

So I take back what I said about her being slower.  I'd say her response time is comparable.

And this may be feedback for a new "Issue", but I can't pause or stop Audible books after they've started.  They do play, though.

This is really stellar work, @lennysh.

dliloch, 2016-03-24:
dmux errors ? is this important .. again it was because I am using
debug=1..see attached
by the way .. If I use the word album .. it will play an entire album ..
ie. alexa play stevie nicks album ..
playing popular stevie nicks album! ...

If ALEXA selects from "music library" get nothing .. must come from prime music..
sound does not have much bass ??
, expected 4) for PID 4096
[6adf5620] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[6e542898] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 4) for PID 4096
[6adf5620] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[6e542898] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 4) for PID 4096
[6adf5620] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[6e542898] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 4) for PID 4096
[6adf5620] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[6e542898] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 5) for PID 4096
[6adf5620] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[6e542898] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 5) for PID 4096
[6adf5620] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 14) for PID 4096
[6e542898] ts demux error: libdvbpsi (PSI decoder): TS discontinuity (received 0, expected 5) for PID 4096

lennysh, 2016-03-24:
@maso27 Simply pressing the record button and releasing it immediately "should" stop any current playback.  Not worked on a pause function yet, although a new button could be added to do a pause.

@dliloch The demux error's are unstoppable at the moment from any/all Amazon Music streams regardless if debug is on or not. Still researching that one, but seems to be related to only their streams.  For example, playing from iHeartRadio doesn't return those errors.

maso27, 2016-03-24:
@lennysh yes, playback of music definitely stops when the button is pressed.  The audible book does not.  I've reproduced it twice and had to reboot the Pi to recover.

lennysh, 2016-03-24:
If anyone prefers live chat, I'm currently online [here](https://webchat.freenode.net/?channels=AlexaDotNet).

lennysh, 2016-03-24:
@maso27 Can you turn debug mode on, start playing the audio book, and then press the button to attempt to stop playback, and then copy/paste the output here?

maso27, 2016-03-24:
Here it is.  I pushed the button twice, with no command the first time, then said "Stop" the second time. Also, I think I can hear Alexa talk over the book when I say "Stop":

`Sending Playback Progress Report Request...
Play_Audio Request for: response.mp3
 Play_Audio Request for: https://audiblecdn-vh.akamaihd.net/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/master.m3u8?start=29477&end=34349&hdnea=st=1458841649~exp=1458841709~acl=/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/*~data=29477,34349~hmac=6129fedd63cd2f8c8ef26e53da697259be55cf33bf267ea0b396fa44503a887f
[0x74f03c78] inhibit interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0x74f03c78] main interface error: no suitable interface module
[0x75b100e8] inhibit interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0x75b100e8] main interface error: no suitable interface module
Player State: State.Opening
Player State: State.Opening
Player State: State.Playing
Playback Progress Report was Successful!
Player State: State.Ended
[0x2245f30] stream_filter_httplive stream: HTTP Live Streaming (audiblecdn-vh.akamaihd.net/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/master.m3u8?start=29477&end=34349&hdnea=st=1458841649~exp=1458841709~acl=/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/*~data=29477,34349~hmac=6129fedd63cd2f8c8ef26e53da697259be55cf33bf267ea0b396fa44503a887f)
[0x2245f30] stream_filter_httplive stream: Meta playlist
[0x2245f30] stream_filter_httplive stream: bandwidth adaptation detected (program-id=1, bandwidth=30000).
Sending GetNextItem Request...
[0x2245f30] stream_filter_httplive stream: VOD Playlist HLS protocol version: 3
[0x2245f30] stream_filter_httplive stream: video on demand (vod) mode
[0x2245f30] stream_filter_httplive stream: Choose segment 0/488 (sequence=2951)
[0x2245f30] stream_filter_httplive stream: downloaded segment 2948 from stream 0
[0x2245f30] stream_filter_httplive stream: downloaded segment 2949 from stream 0
[0x2245f30] stream_filter_httplive stream: playing segment 2948 from stream 0
libdvbpsi error (PSI decoder): TS discontinuity (received 4, expected 0) for PID 0
[0x247b640] ts demux error: MPEG-4 descriptor not found
[0x2245f30] stream_filter_httplive stream: downloaded segment 2950 from stream 0
Player State: State.Playing
[0x2477c10] packetizer_mpeg4audio packetizer: AAC channels: 1 samplerate: 22050
[0x2245f30] stream_filter_httplive stream: downloaded segment 2951 from stream 0
[0x2245f30] stream_filter_httplive stream: downloaded segment 2952 from stream 0
[0x2245f30] stream_filter_httplive stream: downloaded segment 2953 from stream 0
[0x2245f30] stream_filter_httplive stream: downloaded segment 2954 from stream 0
[0x2245f30] stream_filter_httplive stream: playing segment 2949 from stream 0
libdvbpsi error (PSI decoder): TS duplicate (received 0, expected 1) for PID 4095
[0x2245f30] stream_filter_httplive stream: downloaded segment 2955 from stream 0
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"audioItem":{"streams":[{"streamUrl":"https://audiblecdn-vh.akamaihd.net/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/master.m3u8?start=34349&end=38998&hdnea=st=1458841660~exp=1458841720~acl=/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/*~data=34349,38998~hmac=5b83763c5b90ffb87d5f99ec3ea3a9a80a8b57acee2bdd51d9efdb42652fe715","offsetInMilliseconds":0,"expiryTime":"2016-03-24T17:48:40+0000","progressReport":{"progressReportDelayInMilliseconds":0,"progressReportIntervalInMilliseconds":30000},"progressReportRequired":true,"streamId":"amzn1.as-ct.v1.AudibleClientId#ACRI#url#ACRI#dc54cb67-2c7c-40a3-9d74-a44df3ace1b5:9"}],"audioItemId":"amzn1.as-ct.v1.AudibleClientId#ACRI#url#ACRI#dc54cb67-2c7c-40a3-9d74-a44df3ace1b5:9"},"navigationToken":"amzn1.as-ct.v1.AudibleClientId#ACRI#url#ACRI#dc54cb67-2c7c-40a3-9d74-a44df3ace1b5:9"}}
navigationToken: amzn1.as-ct.v1.AudibleClientId#ACRI#url#ACRI#dc54cb67-2c7c-40a3-9d74-a44df3ace1b5:9
streamUrl: https://audiblecdn-vh.akamaihd.net/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/master.m3u8?start=34349&end=38998&hdnea=st=1458841660~exp=1458841720~acl=/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/*~data=34349,38998~hmac=5b83763c5b90ffb87d5f99ec3ea3a9a80a8b57acee2bdd51d9efdb42652fe715
streamId: amzn1.as-ct.v1.AudibleClientId#ACRI#url#ACRI#dc54cb67-2c7c-40a3-9d74-a44df3ace1b5:9
Play_Audio Request for: https://audiblecdn-vh.akamaihd.net/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/master.m3u8?start=34349&end=38998&hdnea=st=1458841660~exp=1458841720~acl=/i/295890/audiblewords/content/bk/adbl/023112/V$174611$V/aax/bk_adbl_023112_22_32.mp4/*~data=34349,38998~hmac=5b83763c5b90ffb87d5f99ec3ea3a9a80a8b57acee2bdd51d9efdb42652fe715
[0x2527fc8] inhibit interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0x2527fc8] main interface error: no suitable interface module
Recording...
Player State: State.Opening
[0x2245f30] stream_filter_httplive stream: playing segment 2950 from stream 0
[... cut ...]
[0x24fe590] packetizer_mpeg4audio packetizer: AAC channels: 1 samplerate: 22050
Recording...
[0x2528ac0] stream_filter_httplive stream: downloaded segment 3437 from stream 0
Player State: State.Stopped
Recording Finished.
Sending Speech Request...
Processing Request Response...
Request Response is null (This is OKAY!)
[0x2245f30] stream_filter_httplive stream: playing segment 2951 from stream 0
libdvbpsi error (PSI decoder): TS duplicate (received 0, expected 1) for PID 4095
[0x2245f30] stream_filter_httplive stream: downloaded segment 2957 from stream 0
`

lennysh, 2016-03-24:
@maso27  Hmm...  Audible streaming might be a little different than music...  Just enough to cause problems.  I may have to get a trial period or something in order to do some testing myself.

dliloch, 2016-03-24:
*pi@raspialexa*:*~ $* aplay -l

**** List of PLAYBACK Hardware Devices ****

card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]

  Subdevices: 7/8

  Subdevice #0: subdevice #0

  Subdevice #1: subdevice #1

  Subdevice #2: subdevice #2

  Subdevice #3: subdevice #3

  Subdevice #4: subdevice #4

  Subdevice #5: subdevice #5

  Subdevice #6: subdevice #6

  Subdevice #7: subdevice #7

card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]

  Subdevices: 1/1

  Subdevice #0: subdevice #0

card 1: Device [Generic USB Audio Device], device 0: USB Audio [USB Audio]

  Subdevices: 1/1

  Subdevice #0: subdevice #0

dliloch, 2016-03-24:
if you ask for news it will start but does not get past the weather ... it
is some sort of audio stream ..

lennysh, 2016-03-24:
@dliloch change
`i = vlc.Instance('--aout=alsa', '--alsa-audio-device=hw:CARD=ALSA,DEV=0')`
To
`i = vlc.Instance('--aout=alsa', '--alsa-audio-device=hw:CARD=Device,DEV=0')`

lennysh, 2016-03-24:
@dliloch yeah, news was working the other day, but recently stopped. Still looking into that one.

dliloch, 2016-03-24:
Lenny,,
perfect! ALSA for HDMI and Device for USB !  ... great work!
-don

lennysh, 2016-03-25:
@dliloch I just merged in pull request #7 into my [music-patch-1](https://github.com/lennysh/AlexaPi/tree/lennysh-music-patch-1) branch.  Please let me know how it performs in comparison.

lennysh, 2016-03-25:
All, I'm still available for live chat [here](https://webchat.freenode.net/?channels=AlexaPi).

dliloch, 2016-03-25:
Lenny,
Seems faster .. I had to simply change the ALSA to Device .. you might make
it a start variable .. and default to ALSA .. no big deal though.
Asked about the new and ALEXA said getting NPR radio .. but I just got the
weather in Seattle.. It would be nice to have default location to where I
am actually at..
still get the demux error ..
when I play a album there seems to be a long wait between songs ..tracks
..but just played an album and you seem to have fixed that..
Again .. great work .. I'm going to put this in a case and replace the
radio on our kitchen sink! Thanks so much for you efforts .. If
I can provide any further help let me know..
-don

lennysh, 2016-03-25:
@dliloch For the location fix, you need to add your Address under your raspberry pi Alexa device on the Alexa app itself.

dliloch, 2016-03-25:
Lenny,
added my address and time, weather worked great thanks..

MiniguyBrendan, 2016-03-25:
Hey, what information did you put for the "buffering/caching mechanisms" on the developer amazon website?

lennysh, 2016-03-25:
Just VLC Media Player

dliloch, 2016-03-25:
works  great .. just set it up from crontab @reboot  python main.py ...
running headless .. when I hear the hello I am ready to go ...
still running on a pi3 which is a waste of resources .. going to try and
run on an old banana pi with integrated mic and wifi.. that would
make a perfect small form factor box..
-don

MiniguyBrendan, 2016-03-25:
They already denied it and there's no edit button to add VLC Media Player :( any ideas?

lennysh, 2016-03-25:
Just change the text and resave

MiniguyBrendan, 2016-03-25:
It's grayed out though.

MiniguyBrendan, 2016-03-25:
I think it's because it says it's pending. But amazon already sent me an email saying it was rejected. Weird.

lennysh, 2016-03-25:
Send me a screenshot

MiniguyBrendan, 2016-03-25:
http://imgur.com/UwLl01P

I didn't know it was required so I initially had just put "Doesn't have." I also didn't know what it meant, let alone that VLC handles that.

lennysh, 2016-03-25:
And you can't type new text in the boxes?

MiniguyBrendan, 2016-03-25:
Nope

MiniguyBrendan, 2016-03-25:
Should I just create a new Alexa product?

[38 more replies]

closed 2016-05-01 by sammachin