alexa-pi/AlexaPiDEPRECATED #21: Alexa does not able to interact with user. Why?

KataPi, 2016-03-10:
Hello dear, good job. Thanks to the installation video. It works!
I am able to interact with Alexa but she does not interact with me.
When you try to set an alarm clock, Alexa records but then the alarm does not work.
What do you think about?

Kata

maso27, 2016-03-11:
Amazon hasn't implemented the alarm/timer functionality for the api yet.  They say it's "coming soon," along with info on sports teams, scores, and schedules.

KataPi, 2016-03-11:
Hi maso27, thanks for the reply. I'm noob...
What do you mean schedules? So in alexa voice service, new skills can not control GPIO Raspberry?
I take this opportunity to take off a doubt: for new skills I must always use Lambda Service (for paid) or just a node.js without Lambda (for free)?
Tks, Kata

maso27, 2016-03-11:
Sorry, I was misleading.  I'm citing their [FAQ](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/docs/alexa-voice-service-developer-preview), and I meant it like sports [teams, scores, schedules].  
So sports schedules.
I'm still figuring this out, but I think new skills could only control Raspberry GPIO if it was written into main.py.  That script is what handles the button press and both LEDs.

I don't think main.py will support alarms or timers as it sits, even once Amazon adds it to their api.

I'm a noob too, but I'm hopeful we can get all this stuff figured out.

lennysh, 2016-03-23:
Sam is going to get the script updated to support the new API which will then allow the script to be modified to allow alarms and such.

KataPi, 2016-03-24:
Hi Lennysh, thanks for the information. I await the update but I think that the new API no longer support the set allarm function. Can you confirm or just happens only to me? Alexa says that my device does not support the function.

lennysh, 2016-03-24:
It's because your device (this script) is using the old api. Once Sam updates it, it'll then support it.

doubledrat, 2016-03-24:
also the script will need to have an alarm function written into it.  I'm guessing the new API just sends back the details of the alarm, which without further coding will do nothing.

lennysh, 2016-03-24:
Indeed.  That will "hopefully" be added once the new API is working with the script.

doubledrat, 2016-04-02:
just got an email that the new API is out -

Alexa Voice Service Update
Last week, we rolled out an important software update to Alexa Voice Service (AVS) that allows you to control volume/media playback and create alarms/timers directly on your product using voice. Learn more about adding voice to connected devices. [https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/reference/interaction-model#Client Interaction with the Alexa Voice Service](url)

sammachin, 2016-04-30:
Yep alarms are only functional in the new 2016 API and thats quite a lot of work to implement, I'm just cleaning stuff up to put lenny's work on music and some better parsing into a v1.1 which will then give me a less hacky codebase to migrate to the 2016 API for v2. 

Can't really give a timeframe for v2 at the moment I'm afraid as it depends on how much free time I have.

closed 2016-04-30 by sammachin

dlarue, 2016-05-30:
@sammachin  what part of CA will you be visiting? If in San Diego let me know and I can arrange a Fab Lab tour.