alexa-pi/AlexaPiDEPRECATED #24: python: can't open file './auth_web.py': [Errno 2] No such file or directory

tim14, 2016-03-20:
Hello guys,
Looks like a great project. Only I get stuck during installation. I use RASPBIAN JESSIE 4.1. The error that I get is python: can't open file './auth_web.py': [Errno 2] No such file or directory. After that he shows the IP from the pi and echo's that I can reboot the pi. Does somebody know how I can fix this?
Kind regards,
Tim

rndmsfree, 2016-03-20:
I saw the same thing. There's an "example_creds.py" file in the AlexaPi folder. Use nano to edit it and just manually enter all your information. Then exit and save it as "creds.py." Then manually launch the auth_web.py program and it is supposed to get you where you need to be (that part hasn't worked for me yet, no matter what I've tried - I think it may be a router issue, but I really don't know.)

The above is kind of confusing, but I'm a brand new linux/RPi user and it's taken me a day and a half to figure it out as far as I have.

This video makes it a little simpler to understand, but he's using a slightly different fork of the software (which has your issue worked out already): https://www.youtube.com/watch?v=frH9HaQTFL8

tim14, 2016-03-22:
Thanks that worked for my !

closed 2016-03-22 by tim14