alexa-pi/AlexaPiDEPRECATED #25: Raspberry Pi 3 - Installation error and missing "creds"

bismosa, 2016-03-20:
Hello,

I have tried to install it on my new Raspberry Pi 3.
Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux
But it wont work.

At installation i get:
_Enter your Security Client Secret:
xxxxx
Object "=" is unknown, try "ip help".
python: can't open file './auth_web.py': [Errno 2] No such file or directory
Open http://:5000
You can now reboot_

If i try to start main.py i get:
_root@raspberrypi:~/AlexaPi# ./main.py
Traceback (most recent call last):
  File "./main.py", line 10, in <module>
    from creds import *
ImportError: No module named creds_

How can i install "creds" ? What is wrong with the installation?

Thanks for this great project!

rndmsfree, 2016-03-20:
I saw the same thing.  There's an "example_creds.py" file in the AlexaPi folder.  Use nano to edit it and just manually enter all your information.  Then exit and save it as "creds.py."  Then manually launch the  auth_web.py program and it is supposed to get you where you need to be (that part hasn't worked for me yet, no matter what I've tried - I think it may be a router issue, but I really don't know.)

The above is kind of confusing, but I'm a brand new linux/RPi user and it's taken me a day and a half to figure it out as far as I have.

This video makes it a little simpler to understand, but he's using a slightly different fork of the software (which has your issue worked out already):  https://www.youtube.com/watch?v=frH9HaQTFL8

tim14, 2016-03-21:
Do you use a wifi module? Because than you have to change in de setup.sh ip =ifconfig eth0 to ip=ifconfig wlan0.

bismosa, 2016-03-22:
HI
Thanks for the help. I have now edit the example_creds.py and now it is working.

closed 2016-03-22 by bismosa