alexa-pi/AlexaPiDEPRECATED #27: Adding skills

rndmsfree, 2016-03-21:
So here's a big question - and I'm not asking you to solve it, but just looking for maybe a trail of breadcrumbs to follow:

Is it possible to add skills to this, admittedly handicapped, pseudo-Echo?

Cool work so far, by the way.  I actually finally got it working earlier today, but then I broke it somehow and haven't gotten it back up.  Wiped everything and am trying again.  (I'm a novice and it seemed easier to start fresh than to try and retrace my steps)

maso27, 2016-03-21:
The shortcoming of AlexaPi is that it's currently only a state machine that waits for a button press, then records a request and returns a response. Anything that requires server-initiated audio beyond that one response gets dropped on the floor. 
There's still a large amount of functionality that does work under those constraints.

I've installed a number of skills on mine, and most work.  I've had success with The Bartender, Teacher, Spelling Bee, and The Magic Door.

doubledrat, 2016-03-21:
how do you go about adding skills to AlexaPi?  I can't connect the amazon app to mine...

maso27, 2016-03-21:
For me, it was fairly seamless.

My Android phone is on the same network as the AlexaPi, and I installed the Alexa app on the phone.
When I went into "Settings/Alexa Devices" I saw both my Fire TV stick and the RaspberryEcho.  For some reason the RaspberryEcho is listed as Offline (most likely due to the state machine that just sits idle instead of constantly communicating with the server).  I set my address, connected calendars, and all the other setup stuff in spite of the Offline status, and everything works.

I can also connect to it from my wife's iPhone, FWIW.

doubledrat, 2016-03-21:
well this is odd.  today I go into the app and the alexapi is there without me having to do anything.  yesterday it wanted me to connect to the echo's wifi and allsorts.

lennysh, 2016-03-23:
Can this be closed now?

closed 2016-03-23 by rndmsfree