alexa-pi/AlexaPiDEPRECATED #28: Alexa's Response is Missing Information

MiniguyBrendan, 2016-03-24:
Hi,

I set up AlexaPi on my pi B+. Works great, thanks.

I only really have one thing that doesn't work as described: When I ask for information, it tends to give me the shortest answer possible, i.e.

"Alexa, what is anonymous?" "Anonymous: the hacking group."
"Alexa, what's the news?" "News from TuneIn."

Both responses stopped exactly there, and offered no more information. Is this not fixable, or does it have to do with how the response JSON is parsed?

Thanks.

lennysh, 2016-03-24:
Please try my music-patch-1 branch on my fork of this project.  Posted it this morning. You'll need to look at the setup.sh of the same branch for my updated requirements installed. Also pay attention to the requirements.txt changes.

MiniguyBrendan, 2016-03-25:
Thanks, it's a little better now. When I ask for the news, it responds with my "flash briefing," but only tells me the weather before ending the response.

Thoughts?

lennysh, 2016-03-25:
This was working the other day. Still looking into this one.

flooie, 2016-04-02:
Any word on this ?  I'm experiencing the same issue.

sammachin, 2016-04-30:
I'm refactoring the whole response parser for v1.1 which should iterate through multiple objects, please sit tight and retest with 1.1 which should be ready soon

closed 2016-04-30 by sammachin