alexa-pi/AlexaPiDEPRECATED #29: Reading Kindle Books

maso27, 2016-03-24:
This is separate from reading pre-recorded Audible audiobooks.
I ask Alexa to read a kindle book from my library, which doesn't have an Audible option, using branch [https://github.com/lennysh/AlexaPi/tree/lennysh-music-patch-1](url).  She says "reading Kindle book _____", then perhaps will open a commercial for Audible (or perhaps not), then nothing.

Debug log yields (this one didn't include a commercial):
`Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"directives":[{"namespace":"SpeechSynthesizer","name":"speak","payload":{"contentIdentifier":"amzn1.as-ct.v1.EBottsClientId#ACRI#DeviceTTSRendererV4_b0b1becb-76e8-4e64-8479-c3197d7ee90a","audioContent":"cid:DeviceTTSRendererV4_b0b1becb-76e8-4e64-8479-c3197d7ee90a_1989703376"}},{"namespace":"AudioPlayer","name":"play","payload":{"audioItem":{"streams":[{"streamUrl":"https://tinytts.amazon.com/1/37621e17-7a5d-4a82-b812-717680f6598d-082bed/2/1458944820817/2930c6b9a2c10c8faa375db567ba559e912687590e789b18120b5098d38b33e7/resource.mp3","offsetInMilliseconds":0,"expiryTime":"2016-03-24T22:28:00+0000","progressReport":{"progressReportDelayInMilliseconds":0,"progressReportIntervalInMilliseconds":0},"progressReportRequired":false,"streamId":"amzn1.as-ct.v1.EBottsClientId#ACRI#url#ACRI#f034aa51-14cf-4c53-affb-5d064e6b44e4:5362"}],"audioItemId":"ebotts_audio_resource_group_id"},"navigationToken":"amzn1.as-ct.v1.EBottsClientId#ACRI#url#ACRI#f034aa51-14cf-4c53-affb-5d064e6b44e4:5362","playBehavior":"REPLACE_PREVIOUS"}}]}}
navigationToken: amzn1.as-ct.v1.EBottsClientId#ACRI#url#ACRI#f034aa51-14cf-4c53-affb-5d064e6b44e4:5362
streamUrl: https://tinytts.amazon.com/1/37621e17-7a5d-4a82-b812-717680f6598d-082bed/2/1458944820817/2930c6b9a2c10c8faa375db567ba559e912687590e789b18120b5098d38b33e7/resource.mp3
playBehavior: REPLACE_PREVIOUS
Play_Audio Request for: response.mp3
[0x1b72798] inhibit interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0x1b72798] main interface error: no suitable interface module
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Play_Audio Request for: https://tinytts.amazon.com/1/37621e17-7a5d-4a82-b812-717680f6598d-082bed/2/1458944820817/2930c6b9a2c10c8faa375db567ba559e912687590e789b18120b5098d38b33e7/resource.mp3
[0x1c2f270] inhibit interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0x1c2f270] main interface error: no suitable interface module
Player State: State.Opening
[0x1c5ff78] gnutls tls client error: TLS handshake error: A TLS fatal alert has been received.
[0x1c5ff78] main tls client error: TLS client session handshake error
[0x1c60038] access_http access error: cannot establish HTTP/TLS session
[0x1c025c8] main input error: open of `https://tinytts.amazon.com/1/37621e17-7a5d-4a82-b812-717680f6598d-082bed/2/1458944820817/2930c6b9a2c10c8faa375db567ba559e912687590e789b18120b5098d38b33e7/resource.mp3' failed
[0x1c025c8] main input error: Your input can't be opened
[0x1c025c8] main input error: VLC is unable to open the MRL 'https://tinytts.amazon.com/1/37621e17-7a5d-4a82-b812-717680f6598d-082bed/2/1458944820817/2930c6b9a2c10c8faa375db567ba559e912687590e789b18120b5098d38b33e7/resource.mp3'. Check the log for details.
Player State: State.Error
`

lennysh, 2016-03-24:
That appears to be an mp3 link. Hmmm, wondering why it wouldn't play it? The issue appears to be with VLC, so might need to change the code a little bit to gey it to play remote mp3 files.

maso27, 2016-03-31:
Just tried it again, and it works now!

Alexa will read any "eligible" (i.e. not graphic novel or something, and not blocked by the publisher) book you have in your Kindle library, via Text to Speech.

closed 2016-03-31 by maso27