alexa-pi/AlexaPiDEPRECATED #30: Poor Audio Play Back

RamblingGeekUK, 2016-03-25:
When I play hello.mp3 using omxplayer all is loud and clear, when running main.py, very low with a hiss over the top.... 

Any suggestions?

Thanks,
Wayne

lennysh, 2016-03-25:
You using this branch or my music-patch-1 branch?

RamblingGeekUK, 2016-03-25:
This branch/orginal, shall I use yours then?

lennysh, 2016-03-25:
Mine supports playing music and such. It also uses VLC to play the audio instead of mpg123.

lennysh, 2016-03-25:
You just need to pay attention to the new requirements

RamblingGeekUK, 2016-03-25:
I've just used yours and the output is the same... I find it odd the audio playback is so bad...

RamblingGeekUK, 2016-03-25:
OK if I run main.py without SUDO output is via the USB sound and quality is good, if I run with SUDO output is via analogue and quality is bad.

RamblingGeekUK, 2016-03-25:
now the script hasn't got rights to right the recording.wav file

RamblingGeekUK, 2016-03-25:
mpg123 seems to be responsible for my poor audio play back.

RamblingGeekUK, 2016-03-25:
OK... I increased the sound quality using "alsamixer" - forehead slap

closed 2016-03-25 by RamblingGeekUK

lennysh, 2016-03-26:
You can change the output device in my version of the script.