alexa-pi/AlexaPiDEPRECATED #31: News - Multiple replies play at once

bhowell09, 2016-03-26:
I did a fresh install of Raspbian and the "lennysh-music-patch-1" branch and when I ask Alexa for the news she starts out by saying "Here is your Flash Briefing" and then starts to say multiple versions of the same or different news stories at the same time causing a ton of errors. I had this issue before the fresh install as well with this branch. 

Here is just a small snipet, but they repeat constantly until I kill the process and while Alexa is making a lot of noise: 

> [6f4dd6e8] mpgatofixed32 audio converter error: libmad error: Huffman data overrun
[69853848] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer
[6f4ae730] alsa audio output error: cannot open ALSA device "hw:CARD=ALSA,DEV=)" Device or resource busy
[6f4ae730] core audio output error: Audio output failed
[6f4ae730] core audio output error: The audio device "hw:CARD=ALSA,DEV=0" could not be used: Device or resource busy.
[6f4ae730] core audio output error: module not functional

Any ideas what I might be doing wrong?

bhowell09, 2016-03-26:
Oh, I forgot to mention that I can get iheartradio to play fine, but I can't get TuneIn to play (response says it's going to, but nothing happens), and I'm waiting for my Amazon Music request to be approved to try that.

bhowell09, 2016-03-27:
So I've done some more playing and noticed that when I ask for my Flash Briefing is tries to read all of the selections I have turned on in the App at one time. I turned all of them off except NPR and it just gets to "from TuneIn" and stops. I turned off that choice and tried "Top News" and it tried to read all of the headlines from that category at the same time. It seems that it's opening multiple instances of VLC at one time and they're stepping one each other. 

> Checking Internet Connection...
Connection OK
Play_Audio Request for: hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"directives":[{"namespace":"SpeechSynthesizer","name":"speak","payload":{"contentIdentifier":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:Say:DAILYBRIEFING:DailyBriefingIntroduction","audioContent":"cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:Say:DAILYBRIEFING:DailyBriefingIntroduction_666280791"}},{"namespace":"AudioPlayer","name":"play","payload":{"audioItem":{"streams":[{"streamUrl":"cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelIntroduction:0","offsetInMilliseconds":0,"expiryTime":null,"progressReport":{"progressReportDelayInMilliseconds":0,"progressReportIntervalInMilliseconds":0},"progressReportRequired":false,"streamId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelIntroduction:0"},{"streamUrl":"cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:0","offsetInMilliseconds":0,"expiryTime":null,"progressReport":{"progressReportDelayInMilliseconds":0,"progressReportIntervalInMilliseconds":0},"progressReportRequired":false,"streamId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:0"}],"audioItemId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:0"},"navigationToken":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:0","playBehavior":"REPLACE_PREVIOUS"}}]}}
navigationToken: amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:0
streamUrl: cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelIntroduction:0
playBehavior: REPLACE_PREVIOUS
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Sending GetNextItem Request...
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"audioItem":{"streams":[{"streamUrl":"cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:Separator:0:1","offsetInMilliseconds":0,"expiryTime":null,"progressReport":{"progressReportDelayInMilliseconds":0,"progressReportIntervalInMilliseconds":0},"progressReportRequired":false,"streamId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:Separator:0:1"},{"streamUrl":"cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:1","offsetInMilliseconds":0,"expiryTime":null,"progressReport":{"progressReportDelayInMilliseconds":0,"progressReportIntervalInMilliseconds":0},"progressReportRequired":false,"streamId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:1"}],"audioItemId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:1"},"navigationToken":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:1"}}
navigationToken: amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:1
streamUrl: cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:Separator:0:1
Play_Audio Request for: response.mp3
Player State: State.Ended
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Sending GetNextItem Request...
Play_Audio Request for: response.mp3
Play_Audio Request for: response.mp3
Player State: State.Opening
Player State: State.Playing
Processing Request Response...
JSON String Returned: {"messageHeader":{},"messageBody":{"audioItem":{"streams":[{"streamUrl":"cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:Separator:0:2","offsetInMilliseconds":0,"expiryTime":null,"progressReport":{"progressReportDelayInMilliseconds":0,"progressReportIntervalInMilliseconds":0},"progressReportRequired":false,"streamId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:Separator:0:2"},{"streamUrl":"cid:DailyBriefingPrompt.5dd38996-3dfc-4878-bd20-439d0fa96fd4:ChannelItem:0:2","offsetInMilliseconds":0,"expiryTime":null,"progressReport":{"progressReportD [... cut ...]
Player State: State.Opening
Player State: State.Playing
Sending GetNextItem Request...
Play_Audio Request for: response.mp3
[75705f10] mpgatofixed32 audio converter error: libmad error: Huffman data overrun
Sending GetNextItem Request...
Player State: State.Opening
Player State: State.Playing
Processing Request Response...
Request Response is null (This is OKAY!)
Processing Request Response...
Request Response is null (This is OKAY!)
Player State: State.Ended
Player State: State.Ended
Player State: State.Ended

iacos, 2016-04-02:
Hey,

I get the same problem/error. Have you manage to fix this in anyway?

bhowell09, 2016-04-02:
No, I haven't, but Lenny said in the Music thread that he was going to work on it. Amazon just released a new API which might slow it down though.

ggavalas, 2016-04-10:
Using the music-patch-01 from lennysh, and adding a delay and audioplaying check to the alexa_getnextitem function eliminated this issue for me. 

`def alexa_getnextitem(nav_token):
        # https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/rest/audioplayer-getnextitem-request
        time.sleep(0.5)
        if audioplaying == False:
                if debug: print("{}Sending GetNextItem Request...{}".format(bcolors.OKBLUE, bcolors.ENDC))
                GPIO.output(plb_light, GPIO.HIGH)
                url = 'https://access-alexa-na.amazon.com/v1/avs/audioplayer/getNextItem'
                headers = {'Authorization' : 'Bearer %s' % gettoken(), 'content-type' : 'application/json; charset=UTF-8'}
                d = {
                        "messageHeader": {},
                        "messageBody": {
                                "navigationToken": nav_token
                        }
                }
                r = requests.post(url, headers=headers, data=json.dumps(d))
                process_response(r)
`

flooie, 2016-04-10:
Totally worked for me thanks @ggavalas  + lennysh - now just gotta fix tunein

shadygrady, 2016-04-11:
Thank you both ggavalas and lennysh!! Been itching for music capability since Sam first put this up. One less Echo Dot I'll have to buy now!

lennysh, 2016-04-13:
I added your suggested fix to my music-patch-1 branch @ggavalas.

sammachin, 2016-04-30:
I'm reworking the parsing of responses for 1.1 and adding in much of the work @lennysh has done with music so I'll check this in the new build.

closed 2016-04-30 by sammachin