alexa-pi/AlexaPiDEPRECATED #34: Run on startup

bobdub19, 2016-03-27:
Hi.

I am having trouble getting Alxa to run on startup of my raspberry pi. the Main.py file is in home/pi/AlexaPi . Everything works fne when i manually run python main.py but on startup alexa service does not run.

any help is appreciated

thanks

doubledrat, 2016-03-27:
how are you trying to get it running at startup?
it should just be a case of -

sudo crontab -e

@reboot python /home/pi/AlexaPi/main.py &

bobdub19, 2016-03-27:
I tried something about the /etc/init.d/ but that did not work. Could you detail how exactly to do this method you are suggesting.

Thanks

bhowell09, 2016-03-27:
Check your log file and see if it says anything about failure to start start up: /var/log/alexa.log

Also, I made sure the start line in my init.d file pointed to the location of my main.py file which is located in the Downloads folder: 

> /home/pi/Downlaods/AlexaPi/main.py

doubledrat, 2016-03-28:
hi bobdub - I did detail exactly what to do!  add the @reboot line to the file you edit when you type the crontab -e command...

printgeek, 2016-03-28:
check your /etc/init.d/alexa file and see if it has a header or not. If not, add one and do an `update-rc.d alexa defaults` to get it working.

greenage, 2016-04-03:
I think the way Sam has written it your files should be at /root/AlexaPi/ not /home/pi/AlexaPi.

closed 2016-04-03 by bobdub19

Stuckeymax, 2016-06-10:
@bobdub19 
Step 1) Add the crontab line supplied by Sam.
Step 2) Comment out the startup line in initd_alexa.sh by placing # in front of it in nano or your fave editor, thus:
#     python /root/AlexaPi/main.py &
Save. After that, you're golden. Hit reboot and she'll start up first time, every time, just like she's got fuel injection and electronic ignition.

Grumpyguard, 2016-09-13:
First off I am a noob.  I have been working at this for a while and the pi won't connect to Alexa Voice Service.  I have completed the setup including getting my Amazon refresh token.  When I boot up the Pi it does not connect to AVS.  When I checked the log it said:  
"starting Alexa...
/etc/init.d/AlexaPi: line 19: /root/AlexaPi/monitorAlexa.sh: No such file or directory
python: can't open file '/root/AlexaPi/main.py': [Errno 2] No such file or directory"

Following the instructions from above here is my initd_alexa.sh file:
"#! /bin/bash

### BEGIN INIT INFO
# Provides:          AlexaPi
# Required-Start:    $all
# Required-Stop:     $all
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: AlexaPi Service
# Description:       Start / Stop AlexaPi Service
### END INIT INFO

exec > /var/log/alexa.log 2>&1
case "$1" in

start)
    echo "Starting Alexa..."
#    python /root/AlexaPi/main.py &
sudo crontab -e
   @reboot python /home/pi/AlexaPi/main.py &
;;

stop)
    echo "Stopping Alexa.."
    pkill -f AlexaPi\/main\.py
;;

restart|force-reload)
        echo "Restarting Alexa.."
        $0 stop
        sleep 2
        $0 start
        echo "Restarted."
;;
*)
        echo "Usage: $0 {start|stop|restart}"
        exit 1
esac"

What changes do I need to make to get this to work?

Thank you for your assistance.

maso27, 2016-09-13:
You have installed as _pi_, in the _/home/pi_ directory.  This install is designed to be run as _root_, in the _/root_ directory.

Does Alexa work when you type `python /home/pi/AlexaPi/main.py &`?
If so, you can edit _/etc/init.d/AlexaPi_ and change everything that says _/root_ to _/home/pi_.  Then it should start at boot.

Rian9771, 2016-09-19:
@maso27 , i am using your snowboy branch and i installed it in /home/pi as root following your directions. The install went well, after reboot, Alexa was not starting on it's own. I checked the /etc/init.d/AlexaPi and noticed the path was /root/... so i changed that to /home/pi/AlexaPi/main.py. I still cannot run main.py and it still does not start on it's own after reboot. I don't see any errors, in /var/log/alexa.log, all i see is "Starting Alexa".  

I can use arecord this.wav and aplay test.wav just fine. Yet, i don't hear anything. Here is the only config that works for me in ./asoundrc. What am i doing wrong? What do you suggest i need to have in "device =  " in main.py. 

pcm.pluged {
    type plug
    #this is your output device
    slave.pcm "hw:0,1"
}

#capturing
pcm.dsnooped {
    ipc_key 1027
    type dsnoop
    #this is your input device
    slave.pcm "hw:1,0"
}

#and this is the real magic
pcm.asymed {
    type asym
    playback.pcm "pluged"
    capture.pcm "dsnooped"
}

#a quick plug plugin for above device to do the converting magic
pcm.pasymed {
    type plug
    slave.pcm "asymed"
}

#a ctl device to keep xmms happy
ctl.pasymed {
    type hw
    card 0
}

#for aoss:
pcm.dsp0 {
    type plug
    slave.pcm "asymed"
}

ctl.mixer0 {
    type hw
    card 0
}

pcm.!default {
    type plug
    slave.pcm "asymed"
}

maso27, 2016-09-19:
@Rian9771 
Did you have a different version installed previously?  That init.d script shouldn't have been loaded with this install.
Regardless, the way to start the _snowboy_ version is different from the other versions.  It needs to be run with `python /home/pi/AlexaPi/start.py /home/pi/AlexaPi/Alexa.pmdl &`.  You can either edit _/etc/init.d/AlexaPi_ to match, or else move the _initd_alexa.sh_ file into its place. (So you'll have to change the name.)

Does your _/etc/init.d/AlexaPi_ have any reference to monitorAlexa.sh?  If so, take that out.  It isn't part of this version and will fail on boot.

This version was me playing with a way to avoid installing as _root_, and I fear it has caused confusion.

Sorry!  But please report back if things work or have issues--the mileage is good because I hope to generate a more robust revision of it.

Rian9771, 2016-09-20:
Running python /home/pi/AlexaPi/start.py /home/pi/AlexaPi/Alexa.pmdl & gives me the following. It does say listening but i don't hear anything. Let me try a clean install and see what happens.

python /home/pi/AlexaPi/start.py /home/pi/AlexaPi/Alexa.pmdl &
[1] 1007
pi@raspbian:~ $ ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pcm_dsnoop.c:557:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Listening... Press Ctrl+C to exit
^C

Rian9771, 2016-09-20:
It seems more like a ALSA error? The .asoundrc config was the only way i could get the aplay and arecord to work. Would you be able to suggest me on what i need to put in main.py for the capture? Getting the mic and speakers to work in ALSA has been a pain so far.

maso27, 2016-09-21:
ALSA takes a back seat in the snowboy stuff. It uses _pulseaudio_ instead. I can't give much as far as suggestions--seems like it can be a struggle if things don't just work out.
My setup is with speakers attached to the 3.5mm jack and a usb webcam attached for the mic. Works great after the install has finished.

A lot of folks worked on it in this thread: https://github.com/sammachin/AlexaPi/issues/82
Maybe there's something helpful in there?

flooie, 2016-09-21:
I can only second @maso27 - I used a bluetooth speaker system in my branch with a pi3, but it took me days to figure out the configuration that worked for me.

Rian9771, 2016-09-21:
Thanks @maso27 @flooie  for your input. I wiped the SD card, dumped a new image of Raspian 8.0, install Maso's Snowboy branch. 

The sudo .setup.sh ended with following before it asked for the Device ID Type.
cp: cannot stat ‘init.d_alexa.sh’: No such file or directory
update-rc.d: error: initscript does not exist: /etc/init.d/AlexaPi

Now, after i do  python /home/pi/AlexaPi/start.py /home/pi/AlexaPi/Alexa.pmdl , it detects me saying "Alexa" and quits with the following.  This is a fresh install. Not sure why it isn't working. 

 python /home/pi/AlexaPi/start.py /home/pi/AlexaPi/Alexa.pmdl
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

[... cut ...]
  File "/home/pi/AlexaPi/snowboydecoder.py", line 190, in start
    main.start(triggerword=True)
  File "/home/pi/AlexaPi/main.py", line 660, in start
    if triggerword: silence_listener()
  File "/home/pi/AlexaPi/main.py", line 517, in silence_listener
    frames_per_buffer=VAD_PERIOD)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 747, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 442, in __init__
    self._stream = pa.open(**arguments)
IOError: [Errno Device unavailable] -9985

maso27, 2016-09-21:
So I found a typo in setup.sh.  Explains the trouble I was having making it copy initd_alexa.sh.
It was "init.d_alexa.sh" when it should've been "initd_alexa.sh"

I've fixed it.

So that should solve the "Run on startup" issue.

As for your errors--it looks like something goes south right at this line:
`ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused`
I did a fresh pull and install on my pi, and everything up to there is expected gobbledygook. Maybe that can point you in a helpful direction? 
(My fresh install comes up and works, for what it's worth).

Rian9771, 2016-09-26:
Okay, thanks @maso27 , i will do a fresh install and update here. Hopefully, this will work! Thanks

Rian9771, 2016-09-26:
Fresh install @maso27  , snowboy branch. Is there anything else i could try?

Starting Alexa...
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=1$
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=1$
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=1$
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
INFO:snowboy:Keyword 1 detected at time: 2016-09-26 14:23:56
rm: cannot remove ‘/home/pi/AlexaPi/tmpcontent/*’: No such file or directory
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
[... cut ...]
  File "/home/pi/AlexaPi/snowboydecoder.py", line 190, in start
    main.start(triggerword=True)
  File "/home/pi/AlexaPi/main.py", line 660, in start
    if triggerword: silence_listener()
  File "/home/pi/AlexaPi/main.py", line 517, in silence_listener
    frames_per_buffer=VAD_PERIOD)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 747, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 442, in __init__
    self._stream = pa.open(**arguments)
IOError: [Errno Device unavailable] -9985

maso27, 2016-09-26:
Unfortunately, it seems to be a very hardware-specific task to get everything working smoothly.  If you look back into that thread I referenced, it looks like @olsonk had similar errors and eventually worked through them.
Some success was had with the `pulseaudio -D` command, but I'm afraid I'm not likely to be too helpful beyond what that thread revealed.