alexa-pi/AlexaPiDEPRECATED #35: Initiate a Alexa request by text possible?

adamdobson, 2016-03-29:
I'm using AlexaPi and skill kit with my home automation system and it works really well, but thinking it would be great if I could use AlexaPi to give my home automation system a voice for notifications rather than just as a responses.

Not sure, but I dont think there is a proper mechanism in the api for unsolicited messages from a skill. But if there was a way to initiate a Alexa request by text rather than audio I could generate a "Simon says " request and have AlexPi speak it out loud. Anyone know if this is possible?

doubledrat, 2016-03-30:
I think I read somewhere that they (Amazon) will be extending Alexa to include notifications in the future.

lennysh, 2016-03-30:
That will be sweet. Your only chance of doing this now is to use a TTS to generate a recording of the Simon says command and send it to Alexa. But then again, you could just use that same TTS to do the notification for you. Which home automation system are you using?

adamdobson, 2016-03-30:
Thanks guys.

Yeah I guess TTS might be an interim option. Passing it onto Alexa would al least give the benefit of a consistent voice.

@lennysh I'm using Home Assistant.

doubledrat, 2016-03-30:
would alexa be able to understand the TTS though?  I have trouble!

another way of doing it would be to have an alexa skill.  e.g. "make announcement".  Use alexapi to capture the wav file of you saying "ask make announcement".  which you can then resend to the alexa servers whenever you want to make an announcement using the code gleaned from alexapi.  your skill can then make a web connection back to base to find out what you want it to say, which you can then send back to alexapi the way you send all your responses.  bit circuitous, but should work

I do all the heavy lifting in a python web server and I just send back as part of the HTML "<SPEAK>whatever I want Alexa to say</SPEAK>" which my skill pulls from the HTML response and uses to populate

e.g.
`def sendHome(params):
# connect to home webserver.  tell it the command and read the voice
# response out of the returned html
    opener=urllib.urlopen(homeWebserver+params)
    result=opener.read()
    response=re.findall('<SPEAK>(.*)</SPEAK>', result)
    if response:
        return urllib.unquote(response[0])
    else:
        return ""
.
.
  speech_output=sendHome(r"?heating=up")
`
HTH

sammachin, 2016-04-25:
Until there's proper support for push notifications and alerts in the AVS API I don't really want to hack something onto AlexaPi.

However if you just want to have announcements played out through your RasPi you can do that totally separate of Alexa, I've built a little app which could run on the same Pi and listens for events on a PubNunb channel then plays that text out using the Ivona text to speech engine from Amazon which i believe  is the same one that powers Alexa although I can't quite match the voices.

https://github.com/sammachin/PubSpeak

closed 2016-04-25 by sammachin