alexa-pi/AlexaPiDEPRECATED #36: Alexa starts, but won't listen to input

rmeskill, 2016-03-30:
I've successfully started Alexa, and she might be listening to input, but I don't think so. On boot, I get a 'starting Alexa' in the log, a 'hello' greeting and 3 flashes of my green LED. If I then push the button (or jumper the GPIO directly together, or start the RPi with the GPIO jumpered from GPIO18>GND and remove the jumper), my LED turns to amber for almost exactly a second and then flashes red and shuts off. This is regardless of how long I hold the button/jumper. Now I bought the button as an individual part without a spec sheet, and it has 4 poles, but I've tried all different combinations of jumping that to the GPIO18>GND with seemingly no effect. I'm not certain the microphone is working, here's my output of arecord -L:
```
plughw:CARD=USB,DEV=0
    Jabra BIZ 2400 USB, USB Audio
    Hardware device with all software conversions
```
And here's what I have in my main.py file:
```
#Settings
button = 18 #GPIO Pin with button connected
lights = [24, 25] # GPIO Pins with LED's conneted
device = "plughw:USB" # Name of your microphone/soundcard in arecord -L
```
(I tried plughw:0 and :1 and I get this error:
```
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 156, in <module>
    start()
  File "/root/AlexaPi/main.py", line 124, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: No such file or directory
```

Thoughts, anyone?

rmeskill, 2016-03-30:
Some further testing confirmed my USB mic is indeed at plughw:
`sudo arecord -D plughw:USB -r 48000 test.wav`
did indeed record sound. Now just to figure out why Alexa doesn't want to listen to me for more than a second...

rmeskill, 2016-03-30:
Wow, and now weirder, my entire system seems to have filled up...
```
pi@raspberrypi:/boot $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.2G  1.2G     0 100% /
devtmpfs        459M     0  459M   0% /dev
tmpfs           463M     0  463M   0% /dev/shm
tmpfs           463M  6.3M  457M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           463M     0  463M   0% /sys/fs/cgroup
/dev/mmcblk0p1   60M   20M   41M  34% /boot

```

doubledrat, 2016-03-30:
sounds like it's started recording but never stopped.  Look for the
recording wav file in your alexapi folder.

I have to say that booting your machine with the pin already jumpered is
probably not a great idea as the loop that checks for a button press relies
on a state change, so starting with the button "pressed" might well confuse
it horribly.

rmeskill, 2016-03-30:
I've completely wiped my machine out (seems I'd filled my filesystem by not expanding it and it wouldn't let me resize it with alexa running, so I thought I'd start from scratch) and reinstalled. I haven't rebooted yet, so I'm sitting post-install. Alexa says 'hello' and when I push the button the LED flicks to yellow for a full second, then flicks red for a brief moment before shutting off. The recording.wav file is exactly 1000 bytes. I tried doing a chmod 777 on it just in case, but no dice. Here's what my AlexaPi directory looks like:
```
root@raspberrypi:~/AlexaPi# ls -l
total 96
-rw-r--r-- 1 root root 16853 Mar 30 17:04 1sec.mp3
-rwxr-xr-x 1 root root  1581 Mar 30 17:04 auth_web.py
-rw-r--r-- 1 root root  8866 Mar 30 17:04 beep.wav
-rw-r--r-- 1 root root   874 Mar 30 17:10 creds.py
-rw-r--r-- 1 root root  1054 Mar 30 17:10 creds.pyc
-rw-r--r-- 1 root root   219 Mar 30 17:04 example_creds.py
-rw-r--r-- 1 root root 19629 Mar 30 17:04 hello.mp3
-rwxr-xr-x 1 root root   405 Mar 30 17:04 initd_alexa.sh
-rw-r--r-- 1 root root  1078 Mar 30 17:04 LICENSE
-rwxr-xr-x 1 root root  3950 Mar 30 17:04 main.py
-rw-r--r-- 1 root root  3873 Mar 30 17:04 README.md
-rwxrwxrwx 1 root root  1000 Mar 30 17:15 recording.wav
-rw-r--r-- 1 root root    83 Mar 30 17:04 requirements.txt
-rwxr-xr-x 1 root root   933 Mar 30 17:05 setup.sh
```

lennysh, 2016-03-30:
Have you tried changing
`device = "plughw:USB"`
to
`device = "plughw:CARD=USB"`
or
`device = "plughw:CARD=USB,DEV=0"`
?

lennysh, 2016-03-30:
I use a usb microphone, and the default
`device = "plughw:1"`
works just fine for me.  What did it do for you?

rmeskill, 2016-03-30:
Just tried all of the above, including :1, to no avail. About 1 second of amber light, then a green flash (not sure if I have my green/red LEDs plugged into the correct GPIO pins), and I'm left with a new, if identical-sounding, 1000 byte recording.wav file.

lennysh, 2016-03-30:
Use my music-patch-1 branch and turn on debug in the code.

rmeskill, 2016-03-30:
Got a slew of errors trying to install that-mainly around the install of VLC. I should also note that, while the button *usually* works, it doesn't always. I didn't get anything resembling a fancy button-just the cheapest one they had at my local electronics shop-I wonder if it's just a momentary button and not one that is stateful for if I keep it depressed. Looks like this: http://arduinotutorial.altervista.org/wp-content/uploads/2011/03/MomentaryPushButton-01a-L.jpg

rmeskill, 2016-03-30:
Equally curiously, running htop shows alexa's consuming 100% of one of my RPi CPUs if it can, which I can't imagine is a good thing, both if I run it manually as 'python main.py' or on startup automatically:
```
  PID USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command
 2236 root       20   0 18588 14544  7252 R 100.  1.5  0:52.20 python main.py
```

lennysh, 2016-03-30:
Replace the start command with the one that's in my branch. Then try.

lennysh, 2016-03-30:
Remove the if statement that has the p.stop command

rmeskill, 2016-03-30:
I'm beginning to think it's an issue with my button-pushing it 90% of the time now amounts to no response, 10% of the time it does register, but only for an absolute moment of a second, no matter how long I hold the button down for:
```
root@raspberrypi:~/AlexaPi# python main.py 
Checking Internet Connection
Connection OK
Ready to Record.
Recording...
Recording Finished.
Ready to Record.
```

lennysh, 2016-03-30:
What's the CPU look like now though

rmeskill, 2016-03-30:
Revised my previous comment-CPU was around 0-15%. However, now when I click the button and it registers, my LED goes amber and stays that way unless I restart alexa, with no output in the logs. No amount of button-mashing has any effect without restarting alexa. And my recording.wav file is now 0 bytes, not 1000

lennysh, 2016-03-30:
I'd check your button then

rmeskill, 2016-03-30:
Yeah, guess so-I'll go pick a new one up tomorrow. Anything special I should be looking for? Do they make momentary buttons and buttons that maintain their state for as long as they're depressed, or should they all be like the latter?

lennysh, 2016-03-30:
They make both. Look for the latter.

rmeskill, 2016-03-30:
Hmm...further troubleshooting (bypassing the button by just jumpering between GPIO18<>GND results in this:
```
root@raspberrypi:~/AlexaPi# python main.py
Checking Internet Connection
Connection OK
Ready to Record.
Recording...
Traceback (most recent call last):
  File "main.py", line 335, in <module>
    start()
  File "main.py", line 272, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy
```

lennysh, 2016-03-30:
Are you manually running main.py?

rmeskill, 2016-03-30:
To get this output, yes

rmeskill, 2016-03-30:
Not sure if directly jumpering GPIO18<>GND should have the same effect as a button, but if I run main.py with those jumpered, then remove and re-jumper them, I still only get a split second of attempted record:
```
root@raspberrypi:~/AlexaPi# python main.py
Checking Internet Connection
Connection OK
Ready to Record.
Recording...
Recording Finished.
Ready to Record.
Recording...
Recording Finished.
Ready to Record.
Recording...
Recording Finished.
Ready to Record.
Recording...
Recording Finished.
Ready to Record.
```
Looks to potentially not be the button...

lennysh, 2016-03-30:
You running it with sudo?

rmeskill, 2016-03-30:
running it logged in as root for testing

doubledrat, 2016-03-30:
this jumpering has me concerned.  can you just attach two wires one to 18
and one to ground, then touch them together while you speak?

and you do realise that gpio 18 is not pin 18 don't you?

rmeskill, 2016-03-30:
I have absolutely no idea what happened, but I reinstalled novaspirit's fork from scratch and all is well. Perhaps the button was just being flaky or there was a weirdly-installed package, but Alexa's working now. Thanks all for the support!

Sivakumarmohan, 2016-04-02:
Here is the solution for what I experienced with my USB headset. The USB headset would not give me hello.mp3 welcome message. I found that the mpg123 app does not redirect the sound to the default device on the system. 

In main.py needs a temp patch ( Hope Sam fixes the issue with a option for device ) 
anywhere you see os.system('mpg123 -q ..... 
suffix you device using option -a , example my hardware is plughw:H340,DEV=0 , os.system('mpg123 -a plughw:H340,DEV=0 -q ....

WR
Siva

dony71, 2016-04-23:
lennysh,
i'm using AlexaPi from your github and get this error below in /var/log/alexa.log
any idea what happened?
----------------------------------------
Starting Alexa...
Checking Internet Connection...
Connection OK
Ready to Record.
Recording...
Recording Finished.
(process_response Error) Status Code: 403
Ready to Record.
Recording...
Recording Finished.
(process_response Error) Status Code: 403
Ready to Record.
Recording...
Recording Finished.
(process_response Error) Status Code: 403
Ready to Record.
Recording...
Recording Finished.
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 426, in <module>
    start()
  File "/root/AlexaPi/main.py", line 401, in start
    alexa_speech_recognizer()
  File "/root/AlexaPi/main.py", line 111, in alexa_speech_recognizer
    r = requests.post(url, headers=headers, files=files)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 94, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 49, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 457, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 569, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/adapters.py", line 407, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', gaierror(-2, 'Name or service not known'))

lennysh, 2016-04-23:
Has it ever worked for you? Could be an issue with yours creds.

dony71, 2016-04-23:
no, it never works. one thing i doubt about my creds, if generate using setup.sh, then there is not any
refresh_token = ''
but without this, i have error below in log
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 421, in <module>
    token = gettoken()
  File "/root/AlexaPi/main.py", line 69, in gettoken
    refresh = refresh_token
NameError: global name 'refresh_token' is not defined

so i add this refresh_token in creds.py manually
what is refresh_token value? should be just empty or?

lennysh, 2016-04-23:
No, that value comes from the Amazon login. If yours is blank, then that's your problem.

dony71, 2016-04-23:
i already suspect somehow token is not authenticated.
when i run setup.sh, while the script running auth_web.py somehow stuck like below and never finish
i'm scratching my head why it stuck? notice also message said empty config, is it normal? 
by the way, does the product ID has to be named AlexaPi ?

[23/Apr/2016:11:48:10] ENGINE Listening for SIGHUP.
[23/Apr/2016:11:48:10] ENGINE Listening for SIGTERM.
[23/Apr/2016:11:48:10] ENGINE Listening for SIGUSR1.
[23/Apr/2016:11:48:10] ENGINE Bus STARTING
CherryPy Checker:
The Application mounted at '' has an empty config.

[23/Apr/2016:11:48:10] ENGINE Started monitor thread '_TimeoutMonitor'.
[23/Apr/2016:11:48:10] ENGINE Started monitor thread 'Autoreloader'.
[23/Apr/2016:11:48:10] ENGINE Serving on http://0.0.0.0:5000
[23/Apr/2016:11:48:10] ENGINE Bus STARTED

ppokharkar3, 2016-04-25:
Hi, I am getting this error on pressing the button, any clues how to resolve it?
 File "main.py", line 156, in <module>
    start()
  File "main.py", line 130, in start
    l, data = inp.read()
alsaaudio.ALSAAudioError: Capture data too large. Try decreasing period size

Thanks!

sammachin, 2016-04-25:
@dony71 At that point in setup then the web server is ready for you to begin the auth process, goto http://[IP of your Pi]:5000 and you should be prompted to enter your amazon login, once that completes it will add a refresh token to your creds.py.

I've improved this in the next version so its a bit clearer.

sammachin, 2016-04-25:
@ppokharkar3 This looks like a totally separate issue, I'm going to close this thread as its got way too many things in it, can you raise as a new issue.

closed 2016-04-25 by sammachin

erm3nda, 2016-05-22:
@ppokharkar3 Which value did you used on the input device?
Try to set `inp.setperiodsize(xxx)` to a value from 32 to 160 or something like that but 32 as low.

eurokid21, 2016-09-18:
@dony71 I was having the same issue and realized that I was using **https** instead of **http** on my Amazon security profiled for allowed origins/allowed return URLS.