alexa-pi/AlexaPiDEPRECATED #37: Red led and Green led connections

iacos, 2016-03-31:
I'm a noobie. 

I don't have a dual led. I am using separate red and green LEDs from an old PC case. They were used as HDD and POWER indicators. They should be fine, I guess. 

On the instructions it says: "... you could also use separate LEDS, Red is connected to GPIO 24 and green to GPIO 25."

Does that mean RED is GPIO 24 + GROUND, right? The same question goes for GREEN too.

Thanks,

C

doubledrat, 2016-03-31:
yes.  and ideally you should also use an appropriate resistor if you want the LEDs to last.  "Appropriate" very much depends on the LED's characteristics, which I'm asssuming you won't know!

PS LED's need to be wired the correct way around or they won't work

iacos, 2016-03-31:
All clear. Thank you.

closed 2016-03-31 by iacos