alexa-pi/AlexaPiDEPRECATED #40: Missing part of the sound at the beginning of response from Alexa

iacos, 2016-04-02:
I am a noobie on Raspbian/Linux or any type programming. I have a Raspberry Pi 2 siting around for nothing for some time in the house. Recently I saw all over the news Amazon's github on how to use a Pi for an Echo using a Java engine and I said to give it a try. After a few hours I finally had it running but I didn't like the GUI starting interface.
This way I've reached Sammachin github on how to do this with a push button using a different engine (Python). I liked the idea more and gave it a try. After many hours and trials I finally had this working also. I ended up installing lennysh's version music-patch-1. 

Although Alexa is working I have a few issues bothering me

1. The beginning part (maybe 3-4 seconds) of the response is missing. The shorter responses do not get at all.

2. If I am asking for the news it sounds like many different responses are played on top of each other making it impossible to understand anything. 

3. If I ask her to play NPR on Tunein it gives a short answer ending in Tunein, this part being the only one I get. 

I am new to github and as far as I comprehend it is not the usual forum-like website where anybody can ask a stupid question and whoever feel like can revert with an answer. Trying to make this work I have posted "issues" asking for answers a few times and some of them have simply been ignored. Somehow I understand. Probably most of these questions are simply dumb for someone experienced who doesn't want to waist his/her time. If I could find answers elsewhere I'd go there, but I couldn't find any on this matter and I really want to make this work. Hoping some of you will understand my noobie position, I am looking forward to some answers. If not I'll just close this issue like those other before it.

lennysh, 2016-04-02:
Try my music-patch-1 branch

maso27, 2016-04-02:
You're right--github isn't intended to be like a forum--more a place to track bugs in the code.  But communication is necessary to figure things out, so it becomes forum-like at times.

It sounds like you're already using the music-patch-1 branch.

So perhaps Question 1 -- the missing first part of the response -- is a new issue.  
The script from @sammachin stuffs the response with a leading 1 second of silence.  My setup, with the music-patch-1 branch that @lennysh made, hasn't had problems playing the whole response so I assumed that vlc doesn't have the same difficulty that mpg123 had with initializing.  But maybe that depends on the audio hardware you're using.

Question 2 is listed in Issue 31--typically they get closed when they're resolved, so that's a decent way to track it.

Question 3 is also mentioned in issue 31.

Does that help at all?

iacos, 2016-04-02:
Thank you taking the time to respond.

Question 1
So, if I replace the 1sec.mp3 file with another mp3, let´s say 3 or 4 seconds long, same name, of course and silent, would this fix the problem?
I have attached the response.mp3 file (weather in Seattle :) so someone can compare and see how long the silence is in the beginning. 

Question 2 and 3
I understand now how it works. Thank you. I will keep an eye on this issue. Letś hope someone will find the cure. 

It helps A LOT!

maso27, 2016-04-02:
The music-patch-1 branch doesn't use the 1sec.mp3 at all, so the main.py script would have to be modified to play it before playing Alexa's response.

I don't see the attachment, but the way it works its that it plays the 1sec.mp3 first, then response.mp3, so there shouldn't be any audio missing from response.mp3.  If there is, that's something entirely different.

Also, I've seen other projects that had to pre-stuff silence too because sometimes the Raspberry pi is slow to initialize outputs.
Can you try a different audio output?  Like HDMI instead of the headphone jack or something?

iacos, 2016-04-02:
Noted. As I see this, there is no reason to extend the length of 1sec.mp3. 

Indeed there is audio missing from the beginning of the response.mp3 file. I don´t know why the attachment does not go through. Probably the Epiphany Browser I´ḿ using. 

I am actually using HDMI out at the moment. I do not know how to derail the sound to the audio jack so I can try listen in headphones.

bhowell09, 2016-04-02:
If you have access to the Desktop GUI then you can change the output by right clicking on the speaker in the lower right corner and changing HDMI to Analog. 

About half way down Issue #20 @Lennysh details how to change the output in the main.py file by determining the output device number using the `aplay -l` command and updating this line
 `i = vlc.Instance('--aout=alsa', '--alsa-audio-device=hw:CARD=ALSA,DEV=0')`

iacos, 2016-04-03:
You were right. It was the audio output. On audio jack Alexa works like a charm. Thank you.

I will close this issue now.

closed 2016-04-03 by iacos