alexa-pi/AlexaPiDEPRECATED #43: 400 Error after auth_web

thesingingmoose, 2016-04-03:
After I run auth_web.py and try to access the socket (serverIP:5000/) I get a 500 error saying "amazon is working to fix this error". Error details "400 Bad Request
The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal."

However, I have this address on the allowed origins and allowed return URLs (originally just on the origins but put it on the return to troubleshoot).

Has anyone had to deal with this?

maso27, 2016-04-04:
I saw this error when I tried to use the built-in browser on the Raspbian desktop.  Tried in Chrome on my desktop and it worked with no issues.

thesingingmoose, 2016-04-04:
I had the error on the Raspbian browser as well as chrome on a Windows laptop. Could the browser matter?

maso27, 2016-04-04:
Did you put all four of the addresses into developer.amazon.com?  

http://localhost:5000
http://<ip_address>:5000

http://localhost:5000/code
http://<ip_address>:5000/code

And did your IP address change for your raspberry pi maybe?

thesingingmoose, 2016-04-04:
Yeah, had first 2 as origin and second 2 as return. I am sshing into the pi at the same time, so I dont think IP changed

tdicola, 2016-04-04:
I just ran into this and what was happening was me accessing http://raspberrypi:5000/ and it trying to redirect to http://raspberrypi:5000/code without that URL being set in the security profile web settings.  Make sure your security profile web settings have exactly the URL you're using to access the Pi.  The instructions call for using localhost but it seems like that will only work if you're using the browser on the Pi itself to access localhost.  Probably best to add http://raspberrypi:5000 and http://raspberrypi:5000/code so you can access under that URL from another machine.

thesingingmoose, 2016-04-05:
I have both localhost:5000 and RPi_IP:5000 added in the web settings (and /code for both for return). 

I now get to the login screen, login, hit "OKAY" to access my device, and then hit the error.

thesingingmoose, 2016-04-05:
There's also no longer a 500, just an error message and a link to Amazon home page

burf2000, 2016-04-06:
Has anyone found a fix for this?

thesingingmoose, 2016-04-07:
Worked when I made a new device in Amazon Dev Portal

closed 2016-04-07 by thesingingmoose

chimeranzl, 2018-10-18:
> Has anyone found a fix for this?

Old post I know, but for anyone else... Delete the line refresh_token: "_token_code_"

and replace with...

refresh_token:

(no double quotes after it)  then try again.  It seems the internal error 500 is caused by the token line existing in the /etc/opt/Alexpi/config.yaml file.