alexa-pi/AlexaPiDEPRECATED #44: Software Button

HUGEwindow, 2016-04-03:
Would it be possible to create a software button for opposed to hardwired? 
Thanks in advance.

lennysh, 2016-04-04:
OK, I'll bite, what would the advantage be?

bhowell09, 2016-04-04:
Amazon actually has a Java based version that uses a GUI button: https://github.com/amzn/alexa-avs-raspberry-pi 

If you are just looking for a way to activate Alexa without building a GPIO button setup and have a Bluetooth adapter and Wiimote, I found some code here on GITHUB and incorporated it into @lennysh music patch. Now I can activate Alexa by holding down the "A" button on a Wiimote that's been paired. I'm going to try and work on a way to use either the GPIO or Wiimote, but my coding is really rusty so it'll be a while.

HUGEwindow, 2016-04-04:
@lennysh I have created a prototype remote. The remote & associated application are not running on the pi. I would like have the application trigger the Alexa opposed to a hardwired button. In any other scenario I would prefer a bluetooth connected device to kick off the interaction as well as a hardwired button on unit. 

@bhowell09 I would be interested in what you have set up. Although I would attempt use the bluetooth device I am hacking.

Thanks for your responses.  Much appreciated.

doubledrat, 2016-04-04:
just to throw my 2p in, I am using AlexaPi with an IR remote using lirc

I'm using one button press to start the conversation and one to end it.  If there's no end press, AP will give up after a certain time and process what it has.

It would be nice if we could end up with some code that supported all 3 options?

marksev1, 2016-04-05:
doubledrat can you show the code that you are using for doing it with an IR remote, Thats a good idea.

Also good idea the Wiimote support.

I'kk throw another idea: https://github.com/sammachin/AlexaPi/issues/45

creator72, 2016-04-05:
+1 on the code to implement the WII remote solution

doubledrat, 2016-04-05:
<PRE>import lirc
sockid = lirc.init("alexa", blocking=True)
</PRE>
I'm also using variables for the LEDs so it's more obvious what's being done in the code.
<pre>
listenLED = 24
processingLED = 25
lights = [listenLED, processingLED] # GPIO Pins with LEDs connected
</pre>
most of the changes are to "start".  

<PRE>
def start():
        recorded=False
        listenCount=0
        while True:
                val = lirc.nextcode()
                if (val  or listenCount>249): # we have an IR code or we've given up waiting for a stop code
                        if (val == ['stop'] or val == ['listen'] or listenCount>249) and recorded == True: # respond to stop or listen again.  or give up waiting
                                rf = open(path+'recording.wav', 'w')
                                rf.write(audio)
                                rf.close()
                                inp = None
                                print "sending to Alexa"
                                alexa()
                                recorded=False
                                listenCount=0
                                lirc.set_blocking(True, sockid) # nextcode will now wait until there's a code rather than consume all the CPU looping
                        elif val == ['listen']: # 'listen' pressed.  start recording
                                os.system('aplay -q {}what.wav'.format(path, path))
                                listenCount=1
                                GPIO.output(processingLED, GPIO.HIGH)
                                inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
                                inp.setchannels(1)
                                inp.setrate(16000)
                                inp.setformat(alsaaudio.PCM_FORMAT_S16_LE)
                                inp.setperiodsize(500)
                                audio = ""
                                l, data = inp.read()
                                if l:
                                        audio += data
                                recorded = True
                                lirc.set_blocking(False, sockid) # don't block while waiting for next IR command
                elif recorded: # no IR code but we're recording, so get more input from the mic
                        l, data = inp.read()
                        listenCount=listenCount+1
                        if l:
                                audio += data

</PRE>

I've configured lirc like this

<PRE>
egin
remote = squeezebox
button = KEY_PAUSE
prog = alexa
config = listen
flags = quit
end

begin
remote = squeezebox
button = KEY_STOP
prog = alexa
config = stop
flags = quit
end
</PRE>
(the remote I'm using is from my "squeezebox"

creator72, 2016-04-05:
Thats a pretty good solution, only problem i have is i'm currently using my IR for my media center, thats why the WII remote would be better for me :-)

doubledrat, 2016-04-05:
just in case you didn't know - you can have lirc recognise multiple remotes...

creator72, 2016-04-05:
mean i don't have a free IR Receiver to hand.

marksev1, 2016-04-05:
They are cheap from ebay :). Thanks doubledrat hopefully i'll be able to copypaste that into the correct places :-)

doubledrat, 2016-04-05:
creator72:  I *think* you could configure lirc to do both the media center and alexa.  If you use the pause button like I have, then you have the benefit of pausing whatever media you might be playing so you can talk to alexa - or you can use a button that you're not using for anything on your media center.  Just an idea :)

marksev1: let me know if you have any trouble, I can send you the .py file in it's entirety.  I'll try and work out how to create a branch once I've tested it further and if people are interested.  Thing is, it's missing the music playing and hopefully the upcoming alerts & timers functionality ;), so I might wait until that appears and mod that code.

creator72, 2016-04-06:
I have managed to hack together the wii remote pairing and using the button 'A' to initiate alexa. 

all the changes are after start():

```
def start():
        recorded = False
        last = 8
        while True:
                val = wm.state['buttons']
                if val != last:
                        last = val
                        if val == 0 and recorded == True:
                                rf = open(path+'recording.wav', 'w')
                                rf.write(audio)
                                rf.close()
                                inp = None
                                alexa()
                        elif val == 8:
                                GPIO.output(25, GPIO.HIGH)
                                inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsa$
                                inp.setchannels(1)
                                inp.setrate(16000)
                                inp.setformat(alsaaudio.PCM_FORMAT_S16_LE)
                                inp.setperiodsize(500)
                                audio = ""
                                l, data = inp.read()
                                if l:
                                        audio += data
                                recorded = True
                elif val == 8:
                        l, data = inp.read()
                        if l:
                                audio += data
if __name__ == "__main__":
        GPIO.setwarnings(False)
        GPIO.cleanup()
        GPIO.setmode(GPIO.BCM)
        GPIO.setup(button, GPIO.IN, pull_up_down=GPIO.PUD_UP)
        GPIO.setup(lights, GPIO.OUT)
        GPIO.output(lights, GPIO.LOW)
        GPIO.output(25, GPIO.HIGH)
        wm=cwiid.Wiimote()
        wm.rpt_mode = cwiid.RPT_BTN
        GPIO.output(25, GPIO.LOW)
        Rumble = False
        while internet_on() == False:
                print "."
        token = gettoken()
        os.system('mpg123 -q {}1sec.mp3 {}hello.mp3'.format(path, path))
        for x in range(0, 3):
                time.sleep(.1)
                GPIO.output(24, GPIO.HIGH)
                time.sleep(.1)
                GPIO.output(24, GPIO.LOW)
```

the Red LED will light up to tell you when to press 1+2 on the remote to pair it (i have the script running in the background so i can exit the terminal & keep the script running using 'sudo nohup ./main.py &' its a bit hack rough but so far its been working solid for 2 days now

lennysh, 2016-04-06:
Doesn't the Wii remote have a mic on it? Or am I mistaken?

creator72, 2016-04-06:
no it has a speaker on it... good idea tho

lennysh, 2016-04-06:
Yeah...  Google confirmed it doesn't, lol...  Worth a thought...  :)

creator72, 2016-04-06:
a mobile phone with a touch button would be the next best thing... just no idea how to incorporate that lol

lennysh, 2016-04-06:
Another user has already done something similar... Uses the phone mic for recording...

creator72, 2016-04-06:
hmm, will go take a look!

marksev1, 2016-04-07:
Lennysh do you have a link to that guys implementation?

GemBro, 2016-04-15:
I should of posted it here first ... but I replied to Issue #50 above (similar question) before I read this ... also posted here: https://www.raspberrypi.org/blog/amazon-echo-homebrew-version/

With AlexaPi the issue is with the 2 states it requires to work properly ... other than using a simple button ... so (from what I can see) we are restricted to what's allowed by Amazon or the API changes to allow other triggers ...

Other activation processes, other than the simple switch, would need 2 states, to trigger the EchoPi, when asking a question … voice activation would not really work or be efficient enough (imho) because of the required 2 states, to make the EchoPi work properly.

iR could work if you have one button for activate and one for de-activate, you then have your 2 states, other than that it will be the same as the voice activation issue.

EXAMPLE (Using voice activation):

State 1:
Using a button: Button is pressed – Amazon is now waiting for your question.
Using Voice Activation Process: Voice activation has been initialised with a voice command – an electronic switch (transistor or whatever you decide to use) is engaged (closes the button circuit).

Intermediate state: Ask a Question: “Alexa what’s the weather like……..”

State 2:
Using a button: Button is de-pressed – You’re now waiting for the response from Amazon.
Using Voice activation process: Voice activation is still on (active) – therefore the electronic switch (transistor or whatever you decide to use) is STILL engaged (a closed circuit, i.e. button still pressed).

One workaround for voice activation, for it to work properly, would require a set time-delay (window) in which your your question is asked … then after that set time-delay it would switch off the electronic switch (transistor or whatever you decide to use), then ‘State 2’ would kick in and take effect and you would get your answer from Amazon.

I can see this to be a pain if you want to be quick and have a short question, plus getting in a long question (within that set time) may not be good enough for Amazon to process and could lead to errors. Even if you have another voice command to end the question, that could then be mistaken as part of the main question.

Just a thought … but would love to hear any work arounds

doubledrat, 2016-04-15:
I have it working with IR and I just use one button.  One press to start recording another to stop it.  Also amazon themselves have released some code that answers the initial question, which is can it be done with a software button.

As for the request in issue 50, it's against the terms of use, but it would be possible using functionality akin to "jarvis" to capture the phrase and send it to alexa.

GemBro, 2016-04-15:
There you go ... you have the 2 states ;) ...

sammachin, 2016-04-18:
I like the IR remote button option thats neat! I'm trying to work out whats the best way to maintain the project with some of these cool customisations, I don't really want to have lots of forks as there's things to fix/improve in the main AVS code. 

I've got a custom build running here thats playing the responses back through my office Sonos so I don't have to plugin an extra speaker. 
It feels like there could end up being one very large project with lots of custom flags for things like sonos/IR remote/software button etc. Anyone got any ideas/suggestions how to maintain that?

marksev1, 2016-04-19:
+1 for IR support, also you could maybe add this - https://www.hackster.io/fcooper27/beagleman-56f174 :)

MQTT support would be nice if possible since most home automation software use MQTT (like Openhab, Node-red, Domoticz,..)..So if Alexa's answer could be relayed to mqtt as a string it would be super-nice.

And some way to interact with it with textual input from other python scripts or other terminal progams would be cool. Thats my wish list so far :)

sammachin, 2016-04-19:
Yeah wake word detection is probabbly the most requested feature, leaving aside the questions about it in the AVS licence I havn't seen one yet that doesn't have a high false positive rate so would keep waking up and feeding bits of random conversation/tvshows to alexa! That video is quite cool but I'd be curious to know if it works well day to day.

I'm not quite sure how MQTT fits in here? Are you talking about using it to control stuff like lights? in which case thats better done via an alexa skill or the home automation skills rather than this alexa front end.
However sending an MQTT event to start/stop recording could be interesting?

Yeah text is hard as AVS only takes voice files, so you'd end up generating a sound file to upload and I'm not sure how you'd parse the response I guess it would need some speech recognition, I think that would end up very clunky and error prone.

marksev1, 2016-04-19:
Regarding MQTT, does Alexa send back the spoken text string, if so I'd like that packaged in MQTT, so I could then act on that with other software.

lennysh, 2016-04-19:
No, it returns an audio response

sammachin, 2016-04-19:
The Alexa skill gets the spoken commands text but AVS just deals with
audio, it sounds like you'd be better off implementing a skill with its own
push channel back to another app running on the Pi, maybe use something
like pub nub to connect the skill on lambda to the Mqtt gateway on the Pi

tube0013, 2016-04-20:
Regarding pocketsphinx, I was having a horrible time with it, then disabled
the auto gain on my microphone input, and it was picking up my keyword
nearly every time. This was in my office where we have a stand alone AC
unit running all the time, so lots of background white noise too.

sammachin, 2016-04-20:
@tube0013 The problem i had was with it picking up normal words in conversation and thinking that was the wake word, try leaving some talk radio on in the background for a bit and see if it triggers.

tube0013, 2016-04-20:
@sammachin that's a valid point, but I would think being able to configure the trigger word to something less commonly spoken is a valid mitigation.  A bigger issue, would be how to deal with any Skill that requires further responses, like wikipedia, and the tell me more command.  You have to issue the trigger word for each response.  Also if it was working with music, how would you issue the pause/next/stop commands.  These are all issues Amazon has worked out with the Echo.  But they also fall victim to the false positives for the Alexa keyword.  I think it's a running joke that their Echo Commercials set off peoples echos in their homes.

sammachin, 2016-04-20:
The false positives I was seeing in my testing wernt due to bad choice of
keyword, even using alexa or nightingale or sausage I was getting it
triggered my conversation that didn't contain that word!

The interactive skills is pretty easy, the response from AVS indicates if
its expecting further input and you can just switch into recording mode
without looking for the wake word.

GemBro, 2016-04-20:
Maybe an active filter would help on the mic input, to reduce such audio discrepancies (false triggers)? ... tune it like, say, a bandpass filter ... and then fine tune it to the wake word maybe? ... this and a mic array (quite simple to build) are things I'd like to try when I get 5mins of electronic time ...

HUGEwindow, 2016-04-26:
I am not getting the initial 'hello'. It hangs on 'connection ok'  Any suggestions?

HUGEwindow, 2016-04-27:
Does AVS return the text of the response?

doubledrat, 2016-04-27:
In a word, no.

sammachin, 2016-04-30:
This thread it huge and covers so many different topics! I'm going to close it down as I think all the outstanding questions are at least answered if not solved.

If you have a new topic please start a new separate issue.

closed 2016-04-30 by sammachin