alexa-pi/AlexaPiDEPRECATED #45: Initiate listening with swiping hand over HC-SR04 ultrasonic sensor

marksev1, 2016-04-05:
Perhaps an idea for a more streamlined listening initiation would be to swipe hand over a HC-SR04 ultrasonic sensor and it would initiate listening for predefined time. Anyone up for implementing it? :)

ProfessionalParanoid, 2016-04-05:
Yup, it's on my todo list.  I have my Alexa integrated into a magic mirror with a motion sensor already working the monitor screen blanking.  I'm thinking I'll add a second one inside, back set behind a small hole in the mirror frame so you'll have to wave specifically in front that small hole to trigger. We'll see how it works out.

marksev1, 2016-04-05:
Awesome looking forward to it. Do you have the magic mirror code also hosted somewhere?

ProfessionalParanoid, 2016-04-05:
I used this one as a starting point: http://michaelteeuw.nl/tagged/magicmirror but used nginx vice apache.  also got rid of the random accolade in the center.  mirror and alexa in action:  http://other.bagpipez.com/Projects

marksev1, 2016-04-07:
This is so cool. Need to build one too :D

sammachin, 2016-04-19:
There's lots of different ideas for triggering the recording, ultrasonic sensors are quite a cool idea:)

I think I need to find a way to implement a generic interface for control and lights that then allows people to create plugins to support whatever controller they like. I'll close this for now and look at it as part of a wider concept.

closed 2016-04-19 by sammachin