alexa-pi/AlexaPiDEPRECATED #46: Error when finishing setup.py

bytedeez, 2016-04-10:
here is the error:
Object "=" is unknown, try "ip help".
python: can't open file './auth_web.py': [Errno 2] No such file or directory

Also at the begining i get this error: Cleaning up...
ln: failed to create symbolic link ‘S99alexa’: File exists

CDSLabs, 2016-04-13:
I am also getting the same error, any help?
Object "=" is unknown, try "ip help".
python: can't open file './auth_web.py': [Errno 2] No such file or directory

sureshrink, 2016-04-13:
I am also getting this error. Checked and rechecked my following the instructions.

jchisholm59, 2016-04-13:
Unless your Python environment variables you may get that "No such file" error.
Just modify the setup.sh to point to the absolute path of your auth_web.py
I got past that but I'm getting an alsa overflow error.
Dunno about the "object=" error, I'm getting that too.
The more complex java version runs great for me however.

sureshrink, 2016-04-22:
I was able to fix this by removing the spaces before and after the = sign.

dony71, 2016-04-23:
script stuck at below, then i have to press CTRL-C to end
is this normal?
-------------------------------------------------------------------------------------------------------
CherryPy Checker:
The Application mounted at '' has an empty config.

[22/Apr/2016:18:40:19] ENGINE Started monitor thread '_TimeoutMonitor'.
[22/Apr/2016:18:40:19] ENGINE Started monitor thread 'Autoreloader'.
[22/Apr/2016:18:40:19] ENGINE Serving on http://0.0.0.0:5500
[22/Apr/2016:18:40:19] ENGINE Bus STARTED

cgioconda, 2016-04-23:
The next step is to go to a webbrowser, to the address you entered in the security setup in Amazon (i.e. I used http://localhost:5000). What will then happen is Amazon will populate the refresh token in the creds.py file, and you should be able to restart your Pi, and use the button to contact Alexa.

dony71, 2016-04-23:
I already add this 2 in amazon, but auth_web.py still stuck there
Allowed Origins : http://localhost:5000
Allowed Return URLs : http://localhost:5000/code

bytedeez, 2016-04-23:
For those of us who are limited in their coding knowledge,  github.com/novaspirit has a fork with modifications added to get rid of the errors. He also has a YouTube video that will guide you through the setup.

dony71, 2016-04-23:
i just try install following novaspirit and still got same error
this means i already try 3 variant of AlexiPi below with same error
https://github.com/sammachin/AlexaPi
https://github.com/lennysh/AlexaPi
https://github.com/novaspirit/AlexaPi

Don't understand why, I must be very stupid maybe?
below is where is stuck with novaspirit git running setup.sh

Open http://128.168.1.43:5000
[23/Apr/2016:14:33:21] ENGINE Listening for SIGHUP.
[23/Apr/2016:14:33:21] ENGINE Listening for SIGTERM.
[23/Apr/2016:14:33:21] ENGINE Listening for SIGUSR1.
[23/Apr/2016:14:33:21] ENGINE Bus STARTING
CherryPy Checker:
The Application mounted at '' has an empty config.

[23/Apr/2016:14:33:21] ENGINE Started monitor thread 'Autoreloader'.
[23/Apr/2016:14:33:21] ENGINE Started monitor thread '_TimeoutMonitor'.
[23/Apr/2016:14:33:21] ENGINE Serving on http://0.0.0.0:5000
[23/Apr/2016:14:33:21] ENGINE Bus STARTED

ericw357, 2016-04-23:
dony71,
if you are folowing the  directions from those forked gits, note that in the directions of those gits they hane not altered the readme to instruct you to clone their forked git.

Clone this repo to the Pi git clone https://github.com/**sammachin**/AlexaPi.git Run the setup script ./setup.sh

if you want to use the   https://github.com/lennysh/AlexaPi  then the instruction would be:

Clone this repo to the Pi git clone https://github.com/lennysh/AlexaPi.git Run the setup script ./setup.sh

dony71, 2016-04-23:
Thank's all
The problem found because I have tried to install lots of time
lennysh git will create /etc/init.d/AlexaPi while novaspirit will create /etc/init.d/alexa
So there are double startup script in my RPi causing double refresh token and fails authentication
I remove one of the startup script then everything fine

closed 2016-04-24 by sammachin