alexa-pi/AlexaPiDEPRECATED #49: Question: is it possible to control devices?

marksev1, 2016-04-14:
Is it possible to control devices with this AlexaPi? So one could say "Alexa, discover my devices" and it would discover the ones it supports.

Like Wemo, hue, possibly control things with mqtt protocol?

doubledrat, 2016-04-14:
yes, but it will involve coding it yourself if there isn't already an alexa "skill" for your device.

check out [https://developer.amazon.com/public/solutions/alexa/alexa-skills-kit/docs/smart-home-skill-api-reference](url)

I have to admit that I've failed to do this myself at the stage of linking the alexa account to my smart home skill :-(

marksev1, 2016-04-14:
I found this http://m.instructables.com/id/Hacking-the-Amazon-Echo/ could we maybe levarage this. It links to a github repository.

doubledrat, 2016-04-14:
that's one way of doing it.  i took the route in the footnote and did a custom skill, but the link I quoted means you don't have to say (in my case) "tell the house to turn the heating up".  Unfortunately I've been unable to get the new method to work due to the requirement to link the skill

marksev1, 2016-04-14:
So what about this way of doing it from the instructables link, do you think it should work?

doubledrat, 2016-04-14:
I don't see why not.  It's just pretending to be a device that Alexa already knows how to interact with.

doubledrat, 2016-04-14:
having looked at the instructable, it's only good for on/off.  If you want to set your heating to a particular temperature or your lights e.g. to 50% it won't do that.

if all you're after is on/off though, it looks about the simplest way of achieving that.

marksev1, 2016-04-14:
Partly, so I maybe should combine this with the other way you described. I've found this: https://github.com/nivertech/lambda-mqtt-test could this be of any use with Alexa (since its amazons service)?

doubledrat, 2016-04-14:
sorry, I've misled you.  I realised as I was driving home that this won't work with AlexaPi at all as the code required isn't within AlexaPi's capabilities.

i doubt the other mqtt will work either as there's no mqtt support in AlexaPi

I think you're back to having to use  a custom skill I'm afraid.

sammachin, 2016-04-19:
There are 2 types of home automation devices, ones that alexa talks to 'Cloud 2 cloud' which you have to  first of all oAuth to their web app with, things like LIFX and SmartThings are this category. 
The second are ones that can be discovered just on your local network by the echo.

AlexaPi should work fine for the first type as its all on the back end, for the local ones it certainly won't currently work as there's no local network discovery code in my client I havn't seen that in the API so it might be an Echo feature not an Alexa feature (they're not quite the same) It might be something in the V2 API and if it is then I'll get to it eventually! First of all I need to get the basics working with V2 though.

closed 2016-04-19 by sammachin