alexa-pi/AlexaPiDEPRECATED #50: Without button to listen?

sabas1080, 2016-04-14:
Hi

Is there any option to not press button and Alexa listen?

Use voice command "Alexa ..." 

Thanks

GemBro, 2016-04-15:
I posted this on raspberrypi.org ... these are the issues I see using anything else part from a button ...

Other activation processes, other than the simple switch, would need 2 states, to trigger the EchoPi, when asking a question … voice activation would not really work or be efficient enough (imho) because of the required 2 states, to make the EchoPi work properly.

iR could work if you have one button for activate and one for de-activate, you then have your 2 states, other than that it will be the same as the voice activation issue.

EXAMPLE (Using voice activation):

State 1:
Using a button: Button is pressed – Amazon is now waiting for your question.
Using Voice Activation Process: Voice activation has been initialised with a voice command – an electronic switch (transistor or whatever you decide to use) is engaged (closes the button circuit).

Intermediate state: Ask a Question: “Alexa what’s the weather like……..”

State 2:
Using a button: Button is de-pressed – You’re now waiting for the response from Amazon.
Using Voice activation process: Voice activation is still on (active) – therefore the electronic switch (transistor or whatever you decide to use) is STILL engaged (a closed circuit, i.e. button still pressed).

One workaround for voice activation, for it to work properly, would require a set time-delay (window) in which your your question is asked … then after that set time-delay it would switch off the electronic switch (transistor or whatever you decide to use), then ‘State 2’ would kick in and take effect and you would get your answer from Amazon.

I can see this to be a pain if you want to be quick and have a short question, plus getting in a long question (within that set time) may not be good enough for Amazon to process and could lead to errors. Even if you have another voice command to end the question, that could then be mistaken as part of the main question.

Just a thought … but would love to hear any work arounds

rastarobbie, 2016-04-15:
Can you please link the relevant T&C? I read through the ones I could find and never saw one about having to use a button. 
I would like to use jasper to listen for the trigger word and record the command and then pass it on to AVS.

doubledrat, 2016-04-16:
I'm sure I've seen it somewhere and someone else said so too.  I can't seem
to find anything now though.  Perhaps they've removed that requirement?

tube0013, 2016-04-18:
There is a project that looks like it forked Sam's code, and has voice activation:  https://www.hackster.io/fcooper27/beagleman-56f174
https://github.com/fcooper/beagleman

I tried integrating the code for sphinx into the lenny-music-1 patch but could not get it to work.

sammachin, 2016-04-19:
I've done some testing with wake word detection, putting aside the T&C's issue for now;

Jasper seems to use google voice for speech rec whcih means everything that is said in the vicinity of your Pi is being streamed up to their cloud, this doesn't work for me from a bandwidth or privacy perspective.

I did look at Pocket Sphinx which is based on the same CMU sphinx project that beagleman uses however I found it had avery high false positive rate which also made it impractical for real world use.
I'd like to try Sphinx a bit more, if anyone fancies creating a simple project (ideally in pyhton) that just listens and lights the LED for 5 secs when it hears the wake word that could be good for testing.

Oh the other problem is that you also have to do silence detection to figure out when the user has finished their command but this is easier to implement just might need tuning for certain scenarios

closed 2016-04-19 by sammachin