alexa-pi/AlexaPiDEPRECATED #51: Installation Issues

dolanwashere, 2016-04-15:
I ran ./setup.sh and I put in all of the correct information, but it is stuck and says:

[15/Apr/2016:10:49:20] ENGINE Started monitor thread 'Autoreloader'.
[15/Apr/2016:10:49:20] ENGINE Started monitor thread '_TimeoutMonitor'.
[15/Apr/2016:10:49:21] ENGINE Serving on http://0.0.0.0:5000
[15/Apr/2016:10:49:21] ENGINE Bus STARTED

Once it gets to this point, it just stays here. Is there something else I need to do?

dolanwashere, 2016-04-15:
Now I have a new issue. It told me to reboot after it finally finished. I rebooted, so now what so I do?

jaysharm, 2016-04-16:
How did you resolve this?

dolanwashere, 2016-04-18:
I'm not really sure honestly. What I did was use Ctrl+C to stop the installation, reboot, and ran the command again and it took a few minutes but then it finally finished and then I rebooted again. Hope it works for you too.

sammachin, 2016-04-18:
The setup script isn't the best at the moment, I'm going to re-visit that for the next version. At that point you've got the webserver running to do the oAuth dance with amazon and get link your AlexaPi to your alexa account, you need to goto either http://localhost:5000 on the Pi's browser or if you don't have a desktop attached to the Pi then goto http://[IP ADDRESS OF PI]:5000 on another machine and sign in to your amazon account

closed 2016-04-18 by sammachin