alexa-pi/AlexaPiDEPRECATED #52: PiFace Digital Support?

QuillzLouis, 2016-04-16:
^

sammachin, 2016-04-19:
Can you give me some more details?

QuillzLouis, 2016-04-19:
So that you can use a piface digital as the button for alexa

sammachin, 2016-04-19:
Interesting idea, I don't think it would be too hard to swap out the code from the GPIO to use the piface library but it would then require maintaining either 2 versions or some kind of config and switch. It also feels a little bit overkill to use a PiFace for this when it only needs a very simple push button and Red/Green LED The piface is around £20 isn't it?

I'm going to close this as I don't see that I'm likely to implement it as I don't have a PiFace to test with, however there are a number of comments about using different methods to 'push the button' so if I find a way to support lots of different options which keeping the codebase clean I'll look to include piface as well.

closed 2016-04-19 by sammachin