alexa-pi/AlexaPiDEPRECATED #53: Offline - but functioning?

jchisholm59, 2016-04-18:
Why would my device (which is working fine for requests such as "what time is it", etc) be reported by 
alexa.amazon.com as being Offline? I have also built the Amazon AVS java version, and it reports as "Online"
Also, requests for Tunein are silent whereas on the Java version they play fine.
Alexa.amazon.com's Tunein reports me as being "Currently Offline"

tube0013, 2016-04-18:
I believe this has to do with AlexaPi currently using the first version of the AVS API from Amazon.  The 2nd version which the Amazon AVS Java Sample is based on includes a "client" side webserver that is always running to answer server initiated requests (etc alarms, which aren't supported in ver 1).  This is just my understanding from looking at both projects release notes around the AVS version 2 API.

sammachin, 2016-04-18:
Yeah the Amazon example almost certainly uses the new HTTP/2 API, I've started looking into updating mine to support this and more of the Alexa features but I'm having issues with the underlying HTTP/2 client for python so not sure of an ETA yet.

closed 2016-04-18 by sammachin