alexa-pi/AlexaPiDEPRECATED #54: 100% CPU ?

jchisholm59, 2016-04-18:
Is main.py supposed to be running at %100 cpu?

doubledrat, 2016-04-18:
I think it's because it's constantly looping waiting for the button press.  It did it to me too.  One of the branches may have been coded differently so as to avoid this issue.

jchisholm59, 2016-04-18:
Thanks doubledrat

jchisholm59, 2016-04-18:
the AVS Java version runs ~ .02 %

sammachin, 2016-04-18:
The Amazon Java version uses a GUI and requires you to click a button on the screen to talk though wheras mine is using hardware. However I might well not be polling the GPIO in the most efficient method as it was ported over from the CHIP which had very crude initial GPIO support, will look to improve this in V2

jchisholm59, 2016-04-19:
I have my AVS version set up to boot headlessly, start automatically with a few xdotool scripts and use a pushbutton via an RPi.GPIO python routine. Works flawlessly!

sammachin, 2016-04-19:
I've incorporated some pull requests so we're managing the GPIO better now, on my test box when idle the AlexaPi script doesn't even register in TOP and when its actively recording its only about 3-4%

closed 2016-04-19 by sammachin

jchisholm59, 2016-04-19:
Cool, I'll try it now! Any idea why though.. even when responding to queries as expected, the python version is reported as being "Offline" which prevents musics and TuneIn play back for example. Thanks for your efforts!

sammachin, 2016-04-19:
Yeah I'm using the very early version of the API before things like online/offline support was a thing, this is more an issue for setting alarms or control from the companion app.

The music issue is separate again when I wrote this version the API didn't document things like Music, someone has figured some of it out so I'm going to try and put music support into v1.1 that I'm working on with this API but I might skip ahead and put it in V2.0 which will use the brand new AVS API but there's a few dependancies I'm waiting on for that. Either way I hope to have music (and hopefully alarms/timers) in the next few weeks, Online support might be a little longer

jchisholm59, 2016-04-19:
Great, thought I was doing something wrong. I sure appreciate the ease of setup of your version.
Hey I'm in Canada (that Amazon hasn't really heard of apparently) so most of the online music stuff doesn't work anyway :/

sammachin, 2016-04-19:
I'm in the UK so in a similar position to you, I find Tune in and Spotify
works on my real Echo

jchisholm59, 2016-04-19:
Great, my Echo is in the mail. 'Course you can't actually "officially" buy one in Canada from Amazon..grrr..

jchisholm59, 2016-04-19:
Wow much better, running at .02!
Cheers