alexa-pi/AlexaPiDEPRECATED #56: Problem with Amazon Login

sureshrink, 2016-04-22:
Everything seems to load and run without error but when I click Okay of the Amazon screen which provides connectivity to Alexa 

When you click "Okay", we'll provide AlexaPi3:
Connectivity to Alexa so that you can interact with it using your voice.

I get the following error and nothing seems to help.

We're sorry!
An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly.
 Go to Amazon.com's Home Page

sureshrink, 2016-04-22:
After reviewing issue 12 I corrected the product if in creds.py and everything worked. Hot damn this is good.

sammachin, 2016-04-23:
Yeah the ProductID catches a lot of people out (me included), will try and make sure the docs are improved in future versions :)

closed 2016-04-23 by sammachin