alexa-pi/AlexaPiDEPRECATED #62: Alexa audio input issue; audio error

ppokharkar3, 2016-04-25:
Hi, when I run main.py I am successfully getting the 'hello' audio output.

But when I press the button(connected b/w GPIO 18 and GND), it throws the following error:

Traceback (most recent call last):
  File "main.py", line 156, in <module>
    start()
  File "main.py", line 130, in start
    l, data = inp.read()
alsaaudio.ALSAAudioError: Capture data too large. Try decreasing period size

Any clues on resolving this?

sammachin, 2016-04-25:
What soundcard/mic are you using?

Can you post the output of arecord -L from the commandline

ppokharkar3, 2016-04-25:
Im using a Logitech headset.

Output of arecord -L is:
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=H340
    Logitech USB Headset H340, USB Audio
    Default Audio Device
front:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    Front speakers
surround21:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    Direct sample mixing device
dsnoop:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    Direct sample snooping device
hw:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    Direct hardware device without any conversions
plughw:CARD=H340,DEV=0
    Logitech USB Headset H340, USB Audio
    Hardware device with all software conversions

In the main.py I have tried all combinations of:
plughw:1
plughw:1,0
plughw:H340,0
plughw: CARD=H340,DEV=0

sammachin, 2016-04-25:
In main.py line 121 try changing the value of inp.setperiodsize(500) down to 400 or 300 and see how that works.

ppokharkar3, 2016-04-25:
I did try that, but still gives the same error..

Not sure if the following might be affecting anything?:
In the file /home/pi/.asoundrc: I have modified card 0 to be card 1:
pcm.!default {
	type hw
	card 1
}

ctl.!default {
	type hw           
	card 1
}

instead of card 0 , so that it outputs through the usb headset.
(Output of arecord -l is:
**** List of CAPTURE Hardware Devices ****
card 1: H340 [Logitech USB Headset H340], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
)

sammachin, 2016-04-25:
You could try playing about with arecord from the command line to see if you can get some options that work with your mic. For a start the settings used in the script are the same as:
arecord -r 16000 -c 1 -d 5 -f S16_LE -D plughw:1 /tmp/test.wav

HUGEwindow, 2016-04-26:
I am getting the same error. I get audio capture from : arecord -r 16000 -c 1 -d 5 -f S16_LE -D plughw:1 /tmp/test.wav

HUGEwindow, 2016-04-26:
I am getting a similar error:

 File "main.py", line 135, in start
    l, data = inp.read()
UnboundLocalError: local variable 'inp' referenced before assignment 

I get audio capture from test suggestion : arecord -r 16000 -c 1 -d 5 -f S16_LE -D plughw:1 /tmp/test.wav

ppokharkar3, 2016-04-26:
The recording and playback are working correctly for the above settings.

I also added the following for mpg123 in the main.py:
('mpg123 -a plughw:1 -q {}1sec.mp3 ........')

ppokharkar3, 2016-04-27:
Hey, so today instead of decreasing the periodSize, like the error was telling me to, I increased the periodSize to 800, and it is now working magically :)

thanks for the suggestions!

closed 2016-04-27 by ppokharkar3

samiultahir, 2017-02-09:
I increased the periodSize in main.py to 800 too and it works like a charm ;)

To do this you Edit in IDLE into main.py and CTLR+F

A popup window will open then type 'period' and press 'find next'

Increase the periodSize from (500) to (800)

Please leave a like and comment if this works for you! :)