alexa-pi/AlexaPiDEPRECATED #63: Does this work with Philips Hue?

benblb, 2016-04-27:
Does this implementation work correctly with Philips Hue without using IFTTT as a workaround?

The official Amazon implementation does not: https://github.com/amzn/alexa-avs-raspberry-pi

Thanks!

sammachin, 2016-04-28:
I don't have a hue to test with but I expect the Echo/Hue integration uses something to connect directly within your LAN from the Echo to the Hue bridge instead of a 'cloud to cloud' connection, therefore this won't work currently with AlexaPi.

doubledrat, 2016-04-28:
I would expect the same Sam.  Although Ben, if you can program it is
possible to write your own smart home skill that calls back to your house
and operates your equipment.  I've integrated about 4 technologies this way
that aren't natively supported.

dony71, 2016-04-28:
below is java based hue bridge emulator and i test it's not working with AlexaPi
it needs echo devices to discover this emulator
https://github.com/bwssytems/ha-bridge

doubledrat, 2016-04-29:
no, it won't work as AlexaPi doesn't have the code to discover and interact with smart home devices

closed 2016-04-30 by sammachin