alexa-pi/AlexaPiDEPRECATED #64: python Error.. I don't know.

zzeri, 2016-04-28:
python Error.. I don't know.  Please.....

pi@raspberrypi ~/Downloads/BlexaPi/AlexaPi $ python main.py

Traceback (most recent call last):
  File "main.py", line 10, in <module>
    from creds import *
  File "/home/pi/Downloads/BlexaPi/AlexaPi/creds.py", line 6

KonstantinSchubert, 2016-04-28:
Your paste does not include the actual error message. It should come before the stack trace (the part which starts with "Traceback")

sammachin, 2016-04-30:
@zzeri not a lot of detail there but it looks like an issue with the creds file which should be created by running the setup script, you'll need to get a set of developer credentials from Amazon and put those in when prompted.

Did you run setup.sh and input as prompted, then go through the web oAuth process to sign AlexaPi into your account?

brownedon, 2016-05-27:
Same issue here.  Successfully connect to amazon before reboot.  Now getting
pi@raspberrypi:~/AlexaPi $ cat /var/log/alexa.log
Starting Alexa...
Traceback (most recent call last):
  File "/home/pi/AlexaPi/main.py", line 10, in <module>
    from creds import *
  File "/home/pi/AlexaPi/creds.py", line 8
    refresh_token = "Atzr|IQEBLjAsAhRfy2kxJ47ObVRDxgjxCFtMhAv9QwIUNEcYK3mbCb6zjgQ6ZsPodg0-gfAswFNSXeDbJ-kTGmsBqvij3h1z3oCO_Is3WBabYV1Kph11LsgB49q4zaX_PSHvtvsUxwR_MuzFz0JO_pHYyy8QoiFJxSr2nSV-ZZ4mmd997o5SK2kv1wvAPj2alPCmoPCLBsi9uHrJiFgeNqe1XjiLWIlTsCabBoG4qHeabsr_zQiJ0d-PCZo87dgEDmNbbsRvKx_tKAeWpSS8Jlz7HQlBujjqD_Il_5cW5Rvgwkc7nwXakq7avAKdB9bmbSYaGR9r6JYisinJ8FP7hpANtGEF8wwyKYKaX-PlntSKGwoVk2mnodjRZDPvZakZ_3Mer3iTSMXFgpIdlJf_morvogrwzqwtCodMixAPH30PCT7gFxk1bLj01YNh9qMwTVu31wg_nfHpnPzE4lYva0OGak5v7zAUTqcN0augv8MBsKewrDt4PysdK1-4_Fog_xN5-Y1SL8pimd_axXLXDdoc1fXt6SmcxJ7FxzXkLPcesm5Uzyc"refresh_token = "Atzr|IQEBLjAsAhQIJvl6dE6dByxGhdvoyFZk5iHZ9AIUObge4G7lF5Q5z1Al7xxbaaGuljzjM0tjwMh0llrP8gkRAKBUnrc7m565...8"

SyntaxError: invalid syntax

brownedon, 2016-05-27:
Looks like this issue comes from doing the authentication (ex. localhost:5000) more than once.  I you look at my refresh_token above, you'll see there are multiple ones there.  Just delete all but one.

fernando-vera, 2016-07-26:
I ran into the same thing and that's how I was able to fix it.

renekliment, 2016-10-26:
Seems like fixed -> closing. Reopen if needed.

closed 2016-10-26 by renekliment