alexa-pi/AlexaPiDEPRECATED #65: error: Cannot open /root/AlexaPi/1sec.mp3: File access error. (code 22)

kotobuki, 2016-04-29:
First of all, thank you very much for providing such an interesting example!

I run main.py without errors and heard Alexa's voice (i.e. hello) successfully.
> $ sudo python main.py
> Checking Internet Connection
> Connection OK

After pressing a button connected to my Pi, I could interact with Alexa. However I got an error as follows:
> [mpg123.c:634] error: Cannot open /root/AlexaPi/1sec.mp3: File access error. (code 22)

sammachin, 2016-04-30:
have you installed the code in /root/AlexaPi/  or somewhere else?

It looks like for some reason I hard coded the path on the 2nd 'padding' 1sec audiofile, no idea why but I've fixed that now in main.py so it should be dynamic this fixed in commit 03f16ca byt changing line 98, you can either download the whole thing again and just keep your creds.pyor patch that one line.

It shouldn't be causing alexa the 1sec is a silence file just to avoid some soundcards clipping the beginning and end of the audio.

closed 2016-04-30 by sammachin

kotobuki, 2016-04-30:
I've installed the code in ~/Desktop/AlexaPi. I just confirmed that now fixed. Thank you very much!