alexa-pi/AlexaPiDEPRECATED #67: ALSA Hanging After First Use

charges, 2016-04-30:
I can't get it to run more than once. Everything works well on first boot. Alexa says "hello." I press the button. Red light goes on. I ask a question. Release button. Red light goes off. Green light comes on. Alexa responds. Yay. Then ... Nothing. 

After the first interaction, nothing. Button push gets no red LED. No Alexa response. This is what the log says:

            Checking Internet Connection
            Connection OK
            Traceback (most recent call last):
              File "/root/AlexaPi/main.py", line 151, in <module>
                            start()
              File "/root/AlexaPi/main.py", line 117, in start
                            inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE,                        
                            alsaaudio.PCM_NORMAL, device)
                            alsaaudio.ALSAAudioError: Device or resource busy

Any ideas why ALSA gets bogged down after the first instance?﻿ I am using a USB min mic (http://www.amazon.com/dp/B00IR8R7WQ/ref=cm_sw_su_dp) for capture and the Raspberry Pi sound card for playback.

KonstantinSchubert, 2016-04-30:
Which branch of the repository are you using? Master?

sammachin, 2016-04-30:
Hmm I'm using the same mic and havn't seen that issue.

Can I suggest trying the version1.1 branch, there's A LOT of stuff changed in there and I'd rather work on getting that one working now, feature-wise its pretty much on par with the current master anyway, the music stuff is new and still a bit fragile.

charges, 2016-04-30:
Thanks. I will reinstall. I actually used NovaSpirit's fork at https://github.com/novaspirit/AlexaPi. I was getting some errors around the script not populating the release token variable and stumbled then across Novaspirit's video. For some reason, his version of the setup script solved that problem for me. I'll try starting from scratch with 1.1. Thanks.

charges, 2016-05-01:
Ok. Reinstalled with 1.1. Running from /root. Same problem. Here's the log entry:

Checking Internet Connection...
Connection OK
Play_Audio Request for: /root/AlexaPi/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Recording...
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 397, in <module>
    start()
  File "/root/AlexaPi/main.py", line 352, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy

I must have ALSA configured wrong.

sammachin, 2016-05-01:
can you post the output of `arecord -l` and `arecord -L`

charges, 2016-05-01:
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=Device
    USB PnP Sound Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Hardware device with all software conversions

Hoki-gcav, 2016-05-02:
I had a similar problem, with a usb sound card and a usb mic at the same time (probably more so if the sound card has a mic input option of any kind) the pi doesn't seem to be able to use both. Try using the aux out with a general speaker to test it with the usb mic still plugged in and then start AlexaPi again.

charges, 2016-05-02:
Sorry if I wasn't clear. I am using the aux out (headphone jack) for output and a USB mic for capture.

Hoki-gcav, 2016-05-03:
Oh.. I thought you meant external sound card when you said sound card..

sammachin, 2016-05-05:
@charges which OS are you using? And is AlexaPi running as root?

charges, 2016-05-05:
I'm using Raspbian Jesse, latest version and ran apt-get update and upgrade after install.

charges, 2016-05-15:
Ok. Been working on this. Started from scratch with Jesse Lite and 1.1. Everything worked perfectly for a half-day. You could ask question after question. All worked great. Now, it will work for a question or two. Then it craps out. No LED response. Here's the log for the last boot:

charges, 2016-05-15:
Sorry. Here's the log:  pi@raspberrypi:/var/log $ tail -50 alexa.log 
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Weather#ACRI#WeatherPrompt.33bdb625-81e8-4c35-bc9c-d3b8a26cca9b", "audioContent": "cid:WeatherPrompt.33bdb625-81e8-4c35-bc9c-d3b8a26cca9b_20839583"}}]}, "messageHeader": {}}
Play_Audio Request for: /root/AlexaPi/tmpcontent/WeatherPrompt.33bdb625-81e8-4c35-bc9c-d3b8a26cca9b_20839583.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Weather#ACRI#GlobalDomain_ActionableAbandon_93fbe147-b45e-428c-b59a-ab83c94f9055", "audioContent": "cid:GlobalDomain_ActionableAbandon_93fbe147-b45e-428c-b59a-ab83c94f9055_816914994"}}]}, "messageHeader": {}}
Play_Audio Request for: /root/AlexaPi/tmpcontent/GlobalDomain_ActionableAbandon_93fbe147-b45e-428c-b59a-ab83c94f9055_816914994.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Knowledge#ACRI#KnowledgePrompt.4b1832ca-1361-42a3-81bf-3fd12fd0631f", "audioContent": "cid:KnowledgePrompt.4b1832ca-1361-42a3-81bf-3fd12fd0631f_1238532806"}}]}, "messageHeader": {}}
Play_Audio Request for: /root/AlexaPi/tmpcontent/KnowledgePrompt.4b1832ca-1361-42a3-81bf-3fd12fd0631f_1238532806.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 410, in <module>
    start()
  File "/root/AlexaPi/main.py", line 380, in start
    alexa_speech_recognizer()
  File "/root/AlexaPi/main.py", line 111, in alexa_speech_recognizer
    r = requests.post(url, headers=headers, files=files)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 94, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 49, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 457, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 569, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/adapters.py", line 407, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', gaierror(-2, 'Name or service not known'))

sammachin, 2016-05-15:
That error looks like the http client can't resolve the hostname for the alexa voice service which generally points to a problem with your DNS server/ISP or possibly your home network.

Which ISP are you with?
Do you ever get problems when browsing the web and pages failing to load?

GemBro, 2016-05-16:
@charges

Are you using a wifi donglee or a network cable with the AlexaPi? ... if wifi it could be the dongle itself ... have you tried it with a network cable? ...

charges, 2016-05-16:
@GemBro Dongle. My second. The first one was even sketchier. I actually haven't had any more occurrences of that network error. I've noticed Alexa response time can be kind of slow and the delay seems proportion to the length of my statement/question but I'm not sure whether that's the Pi's processor, my network or Amazon's turn time. At any rate, it's working great now. iHeart radio works. Waiting for Amazon Music approval. TuneIn seems to work--Alexa replies with the selected station--but then no actual audio is returned. Need to fiddle with that a bit. Thanks for all you help and @sammachin killer code.

sammachin, 2016-05-16:
Ok gonna close this as it seems to be network related, yep the speed of network connection will affect response time as its a record then upload model although on decent broadband with 1meg or so up you should be fine.

closed 2016-05-16 by sammachin

GemBro, 2016-05-16:
Wifi is the bain for getting fast Internet speeds (when you want it) ... wifi bottlenecks all of your super fast Internet speeds down to practically nothing ... wifi Internet connection speed is governed by the actual wifi communication ... but by using a wired network connection you get your full Internet speed, right up to your devices :) ...

charges, 2016-05-16:
I'll go to wired when I install the AlexaPi in the wall behind one of these and wire the talk switch into the button: http://www.thinkgeek.com/product/e9b8/

That will also be when my wife leaves me.