alexa-pi/AlexaPiDEPRECATED #68: TuneIn not working with Version 1.1

Hoki-gcav, 2016-05-01:
When asking for Alexa to play radio one I get this error:

`Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "AudioPlayer", "name": "clearQueue", "payload": {"clearBehavior": "CLEAR_ENQUEUED"}}, {"namespace": "AudioPlayer", "name": "stop", "payload": {}}, {"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Music#ACRI#DeviceTTSRendererV4_8fa4e90a-cd60-4b25-b00a-f3a8bfd916de", "audioContent": "cid:DeviceTTSRendererV4_8fa4e90a-cd60-4b25-b00a-f3a8bfd916de_2140820245"}}, {"namespace": "AudioPlayer", "name": "play", "payload": {"playBehavior": "REPLACE_PREVIOUS", "audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 900000}, "streamUrl": "http://opml.radiotime.com/Tune.ashx?id=s267343&formats=aac,mp3&partnerId=!EALLOjB&serial=AHNQXQPR7RGLLSIBBLJMBRKN2U5A&latlon=47.631863,-122.34427", "offsetInMilliseconds": 0, "expiryTime": "2016-05-01T14:41:32+0000", "streamId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#2fb7c217-0c59-43fc-91bf-0d6db1ac0b05:1", "progressReportRequired": true}], "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#2fb7c217-0c59-43fc-91bf-0d6db1ac0b05:1"}, "navigationToken": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#2fb7c217-0c59-43fc-91bf-0d6db1ac0b05:1"}}]}, "messageHeader": {}}
Play_Audio Request for: /AlexaPi/tmpcontent/DeviceTTSRendererV4_8fa4e90a-cd60-4b25-b00a-f3a8bfd916de_2140820245.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Play_Audio Request for: None
Exception in thread Thread-9:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "main.py", line 270, in play_audio
    m = i.media_new(file)
  File "/AlexaPi/vlc.py", line 1607, in media_new
    if ':' in mrl and mrl.index(':') > 1:
TypeError: argument of type 'NoneType' is not iterable
`

When asking for Alexa to tell me the news I get this error:

`Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:Say:DAILYBRIEFING:DailyBriefingIntroduction", "audioContent": "cid:DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:Say:DAILYBRIEFING:DailyBriefingIntroduction_1519392521"}}, {"namespace": "AudioPlayer", "name": "play", "payload": {"playBehavior": "REPLACE_PREVIOUS", "audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 0}, "streamUrl": "cid:DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelIntroduction:0", "offsetInMilliseconds": 0, "expiryTime": null, "streamId": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelIntroduction:0", "progressReportRequired": false}, {"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 0}, "streamUrl": "http://opml.radiotime.com/Tune.ashx?id=t37047160&sid=p212&formats=aac,mp3&partnerId=!EALLOjB&serial=AHNQXQPR7RGLLSIBBLJMBRKN2U5A", "offsetInMilliseconds": 0, "expiryTime": "1970-01-01T00:01:00+0000", "streamId": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#url#ACRI#DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelItem:0:0", "progressReportRequired": false}], "audioItemId": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#url#ACRI#DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelItem:0:0"}, "navigationToken": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#url#ACRI#DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelItem:0:0"}}]}, "messageHeader": {}}
Play_Audio Request for: /AlexaPi/tmpcontent/DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:Say:DAILYBRIEFING:DailyBriefingIntroduction_1519392521.mp3
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[732b3970] rdp demux error: can't connect to rdp server
[02016fc0] core input error: open of `/AlexaPi/tmpcontent/DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:Say:DAILYBRIEFING:DailyBriefingIntroduction_1519392521.mp3' failed
[02016fc0] core input error: Your input can't be opened
[02016fc0] core input error: VLC is unable to open the MRL '/AlexaPi/tmpcontent/DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:Say:DAILYBRIEFING:DailyBriefingIntroduction_1519392521.mp3'. Check the log for details.
Player State: State.Error
Play_Audio Request for: /AlexaPi/tmpcontent/DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelIntroduction:0.mp3
Ready to Record.
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[7126cc30] rdp demux error: can't connect to rdp server
[70adc450] core input error: open of `/AlexaPi/tmpcontent/DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelIntroduction:0.mp3' failed
[70adc450] core input error: Your input can't be opened
[70adc450] core input error: VLC is unable to open the MRL '/AlexaPi/tmpcontent/DailyBriefingPrompt.57af664b-4dbe-4da5-a9d5-f6874dee07b8:ChannelIntroduction:0.mp3'. Check the log for details.
Player State: State.Error
Play_Audio Request for: 
Player State: State.Opening
libdvdnav: Using dvdnav version 5.0.1
libdvdread: Encrypted DVD support unavailable.
************************************************
**                                            **
**  No css library available. See             **
**  /usr/share/doc/libdvdread4/README.css     **
**  for more information.                     **
**                                            **
************************************************
libdvdread: Couldn't find device name.
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.IFO failed
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.BUP failed
libdvdread: Can't open file VIDEO_TS.IFO.
libdvdnav: vm: failed to read VIDEO_TS.IFO
[70a4dcc8] core demux error: option sub-original-fps does not exist
[75e23b30] core input error: no suitable demux module for `file/subtitle:///AlexaPi//requirements.txt'
Player State: State.Playing
Player State: State.Ended`

GemBro, 2016-05-01:
For some reason your setup is trying to read from a DVD (or DVDROM player) ... VIDEO_TS.IFO is a DVD specific file ... information file ... it's the first thing that is read by a DVD player (like a table of contents) ... have you got a DVD disc in the DVDROM device on your PC? ... is VLC picking up the last format that was played in the player? ... also your MP3's maybe associated with VLC in Windows itself hence why it's trying to play 'something' ... in this case a DVD? ...

Hoki-gcav, 2016-05-01:
Windows? Wait what? AlexaPi is for a RaspberryPi, i has nothing to do with windows and a raspberry pi doesn't have a DVD player unless you buy an external usb one for it. My best guess to why it mentions IFO's is they are something to do with the media streams from TuneIn.

GemBro, 2016-05-01:
DOH! ... of course you're right ... what was I thinking? ... edited it, as in scrapped that comment ... I'm too Windows orientated ... I forget sometimes ... ;) ...

sammachin, 2016-05-01:
@Hoki-gcav Does AlexaPi play any responses eg 'what time is it' ?

Looking at the output of the first error that should be playing back a locally cached file,

Hoki-gcav, 2016-05-01:
Yes, that works fine.

Hoki-gcav, 2016-05-01:
It does not play anything except for the dialogue message that comes before starting the radio or news broadcast. e.g. Radio One on tuneIn. *then silence*

flooie, 2016-05-01:
Have you tried changing which news sources in the Alexa app first.  I can't get some tune in stations but I can get others of Alexa reading.

Hoki-gcav, 2016-05-01:
I can't use the Alexa app for IOS in the UK yet, but I know that it works. Its a playback error of the content not a lack of content. (I think)

gregwa1953, 2016-05-01:
check to see if your audio is switching from the default mode to a
"indeterminate" state.  I have the problem when I ask for news.  I get the
"intro", then nothing and when I look at the audio output source (speaker
icon, upper right corner) it is set to neither HDMI or Analogue.  Once I
set it and readjust the volume things work for a bit...then go back to the
crazy state.

sammachin, 2016-05-02:
@gregwa1953 interesting, sounds like VLC doesn't like the auto audio output, if you run raspi-config you can lock it to the 3.5mm jack.

Hoki-gcav, 2016-05-02:
@sammachin I already did lock it to 3.5 jack because i have a kiosk mode webpage also running on the pi with HDMI so I had to set it to force.

Hoki-gcav, 2016-05-02:
oops pressed wrong button.. why is that button so close to the comment button.

gregwa1953, 2016-05-02:
@SAM,
It is already locked.  I think it's an issue with Pulse Audio trying to
over ride the Alsa system when things go to silence for more than a second
or so.  Just a thought right now...will be doing more investigating.

Hoki-gcav, 2016-05-02:
Closing this issue because no further investigation is required for this issue and further developments on the bug can continue in Issue #71. :))

closed 2016-05-02 by Hoki-gcav

flooie, 2016-05-03:
Just for the record @Hoki-gcav I'm having the exact same issue  - just here in the States