alexa-pi/AlexaPiDEPRECATED #69: alsaaudio.ALSAAudioError: Device or resource busy using Logitech webcam as mic

gregwa1953, 2016-05-01:


gregwa1953, 2016-05-01:
Everything works fine up to the point where the system tries to record audio from the mic.

I get...

File "./main.py", line 156, in <module>
    start()
  File "./main.py", line 124, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy

The mic is defined as card1 in /proc/asound/.
main.py is set as plughw:1 
alsaaudio.PCM_CAPTURE = 1

arecord -L shows:
hw:CARD=U0x46d0x9a1,DEV=0
    USB Device 0x46d:0x9a1, USB Audio
    Direct hardware device without any conversions
plughw:CARD=U0x46d0x9a1,DEV=0
    USB Device 0x46d:0x9a1, USB Audio
    Hardware device with all software conversions

using 'arecord -D plughw:1 --duration=10 -f cd -vv ~/rectest.wav' works fine.

Any suggestions?  I believe everything else will work fine once this issue is fixed.

gregwa1953, 2016-05-02:
Further confusion...
The above issue was encountered using a Pi 2.  Moved the microSD card to a Pi 3 and it doesn't give the error.
However, recording.wav is just noise.
Once I figure out the issue with that, then I'm sure I'll be humming along with everyone else.  It probably has to do with the Logitech webcam that I'm using as a USB Mic, but it works fine using the java version of Alexa for Pi.  :(

Hoki-gcav, 2016-05-02:
Do "sudo alsamixer" and press F6, if there is a sound card displayed there for the logitec webcam select it and make sure there is a mic volume bar you can adjust, if there isn't then this may be a lost cause. (If you've already done this then i don't mean to patronise)
I tried using a web cam mic before and the range for Alexa was god awful and it barely could figure out what I was saying. Im using a Blue Snowball mic and that works phenomenally, maybe even better than the echo it self. Don't cheap out on the mic, its well worth investing in a decent audio capture device for this. 
Sorry that I cant be of more help.

gregwa1953, 2016-05-02:
@Hoki-gcav,
Thanks for the reply.  Well, I've tried your first suggestion, and It does
work outside of Echo (not patronising at all).  In fact, it works well
using the "arecord" command since I have the webcam just on top of the
monitor and it's only 3 feet away from my big mouth.  Just not in the echo
app.  YET....

Unfortunately, when it comes to buying a "real" mic, I'm currently out of a
job, so I have to live with what I've got at this moment.  I'm sure that if
I could use a decent mic, this wouldn't be an issue.

Thanks again for your help and suggestions.

mholzi, 2016-07-16:
Hi guys,
I have the same issue. Is there a solution to the "device or resource busy" yet?

renekliment, 2016-10-26:
Hey, do you still have issues with this?

mohaggag, 2016-11-07:
i have the same issue and i can see that no one care

LittleMiller4, 2017-01-07:
Yes, I'm having issues with this error.

renekliment, 2017-01-07:
@LittleMiller4 Do you have a desktop version of the OS? If that is the case, PulseAudio is sitting on the sound card and you can't access it as another user. You'll have to set up PA in system-wide mode.

If you use the new repo (https://github.com/alexa-pi/AlexaPi), there is a how to: https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio

LittleMiller4, 2017-01-07:
I am running Raspbian Jessie on a Raspberry Pi A. After following that tutorial, Alexa starts and then crashes. And because I set it to restart after crashing in the setup, it's starting then crashing, starting then crashing. 
After running main.py:
```
pi@raspberrypi:/opt/AlexaPi/src $ sudo python main.py
[0134c470] pulse audio output error: PulseAudio server connection failure: Access denied
[0137e368] pulse audio output error: PulseAudio server connection failure: Access denied
Checking Internet Connection...
Connection OK
[01352030] pulse audio output error: PulseAudio server connection failure: Access denied
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

[01352030] alsa audio output error: cannot open ALSA device "default": Connection refused
[01352030] core audio output error: Audio output failed
[01352030] core audio output error: The audio device "default" could not be used:
Connection refused.
[01352030] core audio output error: module not functional
[b4111a00] core decoder error: failed to create audio output
ALSA lib conf.c:4705:(snd_config_expand) Unknown parameters CARD=pulse
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=pulse
Traceback (most recent call last):
  File "main.py", line 587, in <module>
    loop()
  File "main.py", line 505, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Invalid argument
```
Thanks!

renekliment, 2017-01-07:
>  PulseAudio server connection failure: Access denied

You probably did not add the alexapi user to the pulse-access group.

LittleMiller4, 2017-01-08:
This came up after trying again.
`
`The user `alexapi' is already a member of `pulse-access'.`

renekliment, 2017-01-08:
> EDIT: Actually, you're running this under the pi user, so you need to add the user pi there.

LittleMiller4, 2017-01-09:
It says the same as above for user 'pi'.

stu0219, 2017-01-09:
Hi:
Maybe the other application take your Mic already, in my experience
there is two way I tried and fixed:
1. Key "ps aux |grep pulse", and find out the pulseaudio process, and kill it, this is process which take my mic.
2. Try to change device name "PlughwX:X" to "Default", this might work too.

 this might be not best solution, but work for me.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment