alexa-pi/AlexaPiDEPRECATED #70: Inverting GPIO output?

Hoki-gcav, 2016-05-01:
I hooked up a mains relay to the GPIO pins but it seems to invert the lamp I hooked up. I want it to come on when alexa is talking and stay off if she is not. I got a little demo video of it in action.
[](http://www.umdio.com/wp-content/uploads/2016/05/HelloLamp.mp4)

GemBro, 2016-05-01:
A transistor can invert the output signal for you ... I take it you have some form of driver (for tyhe relay) already in line with the GPIO? ... is it possible to sketch up a schematic of what you currently have? ...

Hoki-gcav, 2016-05-02:
5V pin is going to VCC, GND goes to GND,  GPIO 25 goes to Relay input pin (IN1).
So basic that a diagram wouldn't be needed.

Hoki-gcav, 2016-05-02:
I'll try and find a transistor lying around.

GemBro, 2016-05-02:
Hoki ... I would advise putting a driver inline with the GPIO (the transistor will do the job) ... because at the moment, without a driver circuit, your GPIO pin will get fried over time, due to the back EMF from the relay's coil ... a small diode, at least, needs to go across the relay coil ... you will still need the diode when using a transistor too ...

Here's a driver circuit:
http://www.petervis.com/Raspberry_PI/Driving_Relays_with_CMOS_and_TTL_Outputs/relay-driver-circuit-using-npn-transistor.gif

sammachin, 2016-05-02:
@Hoki-gcav If its just that you want the playback light to be HIGH when off and LOW when speaking then you could just modify those lines in the code to change the values from HIGH to LOW.

However one of the things I want to do fairly soon is abstract the lights stuff into its own file and functions as people want to hook up all sorts of visual indicators so this should make it easier to modify, 

Feel free to have a play but its on my list of things to change soon.

Hoki-gcav, 2016-05-02:
Okay thanks Sam :).
Came up with a solution, the relay actually has an inverter function on board so Im going to just use that. I've put a resistor in place, thanks GemBro for that.

Hoki-gcav, 2016-05-02:
Closing because it has been resolved. :)

closed 2016-05-02 by Hoki-gcav