alexa-pi/AlexaPiDEPRECATED #71: TuneIn ... so close

GemBro, 2016-05-02:
Running AlexaPi 1.1 ... ask to play 'JFS Radio' and get the following stream error and the program (loop) ends ... as in crashes out to prompt ...

ERROR DUMP:

Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "main.py", line 265, in play_audio
    file = tuneinplaylist(file)
  File "main.py", line 256, in tuneinplaylist
    r = requests.get(req.content)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 60, in get
    return request('get', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 49, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 443, in request
    prep = self.prepare_request(req)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 374, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 304, in prepare
    self.prepare_url(url, params)
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 354, in prepare_url
    scheme, auth, host, port, path, query, fragment = parse_url(url)
  File "/usr/lib/python2.7/dist-packages/urllib3/util/url.py", line 143, in parse_url
    raise LocationParseError(url)
LocationParseError: Failed to parse: st8.webradioworld.net:8008
http:

MY GUESS:
"Failed to parse: st8.webradioworld.net:8008" ... because that is not the actual audio stream URL ...

It's close at playing the stream though ... that 'st8.webradioworld.net:8008' is the actual ShoutCast Admin page, which is correct ... but it's not the audio stream ... to stream the audio (to hear it) it must have http:// in front of that URL ... i.e. ... http://st8.webradioworld.net:8008 ...

If 'st8.webradioworld.net:8008' is put into VLC (on a PC) it errors out, like in AlexaPi ...

But if you put http://st8.webradioworld.net:8008 into VLC (or any media player for that matter) it does start streaming the audio ...

So it looks like the TuneIn audio stream URLs are being grabbed right but not parsing correctly because it's missing the vital bit at the beginning (the http://) to say it's actually audio ... that's what I'm getting from my log above ...

Hoki-gcav, 2016-05-02:
Hmm that could be the solution to my issue too :o

GemBro, 2016-05-02:
Hoki ... it looks like it ... I just asked to play Radio One and get the same error as I did for my station ...

s30.streamerportal.com:8448 ... in VLC does not stream audio ...

but ... http://s30.streamerportal.com:8448 ... does stream audio ...

ERROR DUMP:

Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "main.py", line 265, in play_audio
    file = tuneinplaylist(file)
  File "main.py", line 256, in tuneinplaylist
    r = requests.get(req.content)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 60, in get
    return request('get', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 49, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 443, in request
    prep = self.prepare_request(req)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 374, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 304, in prepare
    self.prepare_url(url, params)
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 354, in prepare_url
    scheme, auth, host, port, path, query, fragment = parse_url(url)
  File "/usr/lib/python2.7/dist-packages/urllib3/util/url.py", line 143, in parse_url
    raise LocationParseError(url)
LocationParseError: Failed to parse: s30.streamerportal.com:8448
http:

Hoki-gcav, 2016-05-02:
Can you tell me the name of a radio station that worked for you so that I can compare? I don't know any American radio stations.

GemBro, 2016-05-02:
Here's a list of USA Internet Radio Stations on TuneIn:
http://tunein.com/radio/United-States-r100436/

I tried one from New York ... asked to play WBGO ... and now Alexa loops without any audio ... keeps saying playback successful though ... but no audio at all ... I let it loop 3 times ...

LOG DUMP:

pi@raspberrypi:~/AlexaPi1.1 $ sudo python main.py

Checking Internet Connection...
Connection OK
Play_Audio Request for: /home/pi/AlexaPi1.1/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended

Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...

JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Music#ACRI#DeviceTTSRendererV4_3c67cfb7-f5ca-40e6-ab03-d5fe06939fec", "audioContent": "cid:DeviceTTSRendererV4_3c67cfb7-f5ca-40e6-ab03-d5fe06939fec_1744324029"}}, {"namespace": "AudioPlayer", "name": "play", "payload": {"playBehavior": "REPLACE_PREVIOUS", "audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 900000}, "streamUrl": "https://opml.radiotime.com/Tune.ashx?serial=AFIHU5JRSUSO3FIXP5L7MSTDFWQQ&formats=aac,mp3&partnerId=!EALLOjB&id=s27437", "offsetInMilliseconds": 0, "expiryTime": "2016-05-02T12:22:53+0000", "streamId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1", "progressReportRequired": true}], "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1"}, "navigationToken": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1"}}]}, "messageHeader": {}}
Play_Audio Request for: /home/pi/AlexaPi1.1/tmpcontent/DeviceTTSRendererV4_3c67cfb7-f5ca-40e6-ab03-d5fe06939fec_1744324029.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Play_Audio Request for: https://opml.radiotime.com/Tune.ashx?serial=AFIHU5JRSUSO3FIXP5L7MSTDFWQQ&formats=aac,mp3&partnerId=!EALLOjB&id=s27437
 Ready to Record.
Player State: State.Opening
Player State: State.Playing
Sending Playback Progress Report Request...
Player State: State.Ended
Sending Playback Progress Report Request...
Sending GetNextItem Request...
Playback Progress Report was Successful!
Playback Progress Report was Successful!
Processing Request Response...

JSON String Returned: {"messageBody": {"audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 900000}, "streamUrl": "https://opml.radiotime.com/Tune.ashx?serial=AFIHU5JRSUSO3FIXP5L7MSTDFWQQ&formats=aac,mp3&partnerId=!EALLOjB&id=s27437", "offsetInMilliseconds": 0, "expiryTime": "2016-05-02T12:22:59+0000", "streamId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1", "progressReportRequired": true}], "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1"}, "navigationToken": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1"}, "messageHeader": {}}
Play_Audio Request for: https://opml.radiotime.com/Tune.ashx?serial=AFIHU5JRSUSO3FIXP5L7MSTDFWQQ&formats=aac,mp3&partnerId=!EALLOjB&id=s27437
Player State: State.Opening
Player State: State.Playing
Sending Playback Progress Report Request...
Player State: State.Ended
Sending Playback Progress Report Request...
Sending GetNextItem Request...
Playback Progress Report was Successful!
Playback Progress Report was Successful!
Processing Request Response...

JSON String Returned: {"messageBody": {"audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 900000}, "streamUrl": "https://opml.radiotime.com/Tune.ashx?serial=AFIHU5JRSUSO3FIXP5L7MSTDFWQQ&formats=aac,mp3&partnerId=!EALLOjB&id=s27437", "offsetInMilliseconds": 0, "expiryTime": "2016-05-02T12:23:01+0000", "streamId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1", "progressReportRequired": true}], "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1"}, "navigationToken": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#36150927-4ae0-46a8-a183-d73e06b0e29d:1"}, "messageHeader": {}}
Play_Audio Request for: https://opml.radiotime.com/Tune.ashx?serial=AFIHU5JRSUSO3FIXP5L7MSTDFWQQ&formats=aac,mp3&partnerId=!EALLOjB&id=s27437
Player State: State.Opening
Player State: State.Playing
Sending Playback Progress Report Request...
Player State: State.Ended
Sending Playback Progress Report Request...
Sending GetNextItem Request...
Playback Progress Report was Successful!
Playback Progress Report was Successful!
Processing Request Response...

I exited out to prompt after 3 loops ...

sammachin, 2016-05-02:
Thanks @GemBro it looks like TuneIn returns all sorts of nested URL's for different stations and VLC doesn't seem to easily parse those so I might have to write code for each type to extract the real audio streams, this is what I did for the BBC stuff in the UK.

@Hoki-gcav When you say Radio One are you talking about BBC Radio 1 in the UK, that works for me as it was my first test of TuneIn?

Hoki-gcav, 2016-05-02:
Do you have to specify that its BBC? I will try that.

Hoki-gcav, 2016-05-02:
That worked! Wow I'm incompetent. I didn't factor in that there might be a different radio 1 beside the BBC's... .-.
This works brilliantly, I'm gonna need to get a proper sound card for this pi now cus I'm gonna be using this a lot ^-^

sammachin, 2016-05-02:
No worries, the the BBC stuff should work but I'm not surprised other
stations don't, I'll take a look at 'Radio One' whereever that is!

sammachin, 2016-05-02:
Note for me to look into, theres a VLC plugin for TuneIn that might add some parsing capabilities to the urls that alexa sends back for Tune IN, no idea if it will work with the py bindings or the command line.
http://addons.videolan.org/content/show.php/TuneIn+Radio?content=152788

Hoki-gcav, 2016-05-02:
Hmm, i'll give a shot if I get some time. I'm pretty happy with just Radio One, thats the only station I'd ever listen too. Waiting for someone to make a spotify control skill (which is possible through spotify's controller API) so I can have alexa play my playlists :)

GemBro, 2016-05-02:
That works!!! ... BBC Radio 1, 2, 3 etc ... that's cool to actually get a station playing now on TuneIn ... didn't think to use BBC either ;) ...

Some (UK stations) that work:
Play Kiss FM - works (plays on iHeart radio ... I'm not registered with iHeart!!!)

Play BBC Radio London - works (but has error)
Error = ([71a22e40] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer)

Play Capital FM - works (but has error)
Error = [75e1cdf0] mpgatofixed32 audio converter error: libmad error: bad main_data_begin pointer

Gonna try some others ... here's a list of UK Internet Radio Stations:
https://en.wikipedia.org/wiki/List_of_radio_stations_in_the_United_Kingdom

Cannot get the Android Alexa app to work though (downloaded APK) ... so cannot try iHeart or others ... wish there was a tutorial on how to use Alexa app with AlexaPi ...

sammachin, 2016-05-02:
@GemBro I struggled to signup for iHeart Radio I think you need to have a US ip address need to sort my VPN.

I don't think there's anything that the iOS/Android app can do that you can't do from the web, http://alexa.amazon.com

GemBro, 2016-05-02:
Ok ... thanks for the tips Sam ... 

The good news is, after the second attempt, I've now got approved for Amazon Music streaming through the AVS ... gonna tuck into this project properly and design some electronic addons ...

Hoki-gcav, 2016-05-02:
If you want to get the app for iOS or Android just make a chosen app  store account on US store with no payment method linked. I did this for iOS and managed to get the Alexa App on my Nexus 7 tablet, iPad & iPhone.

By the way Gem, how long did it take to get your Amazon Music streaming to go from pending to accepted? And what did you put for the fields asking what cache and buffering mechanisms your device uses?

GemBro, 2016-05-02:
Thanks for that USA hack Hoki ... I'll give that a go ;) ...

As for the Amazon Music section (on the Dev application page - device’s buffering/caching mechanisms) my 2nd attempt for approval seem to take a bit longer than the 1st ... 

I originally put in 'RaspberryEcho' (cannot remember where I saw that suggestion) ... obviously wasn't enough info for Amamzon hence got rejected ...

On the 2nd attempt I just put in 'VLC Media Player' ... that seems to work ...

I think Amazon just need to know that you don't intend to rip/copy/record the music ... so having an application that cannot do those things (intentionally) and does not void any copyright laws regarding the streamed audio from Amazon, is fair doo's ...

If I'm correct, I think the actual time (delay) for the max Amazon Audio stream length session is 10 seconds? ... then it dumps the cache and fills it up again ... I think, that's what's required by Amazon ... I read that somewhere too ...

GemBro, 2016-05-03:
Gonna close this now issue ...

flooie, 2016-05-03:
... I still don't have tune in working -

GemBro, 2016-05-03:
Flooie ... Sam's working on the URLs that cannot get parsed ... some of us in the UK can get some of our well known stations to play (see above) but none of the USA stations stream audio, because of the URL errors (I can't at the moment) ... I'm sure something will be sorted soon ;) ...

Have you tried the BBC stations above? ... if they work fine using TuneIn normally (web page) then they should work in AlexaPi ...

ProfessionalParanoid, 2016-05-03:
So, I'm having the same problem where you press the button and ask for a TuneIn station and it says it's going to play the station (NPR news in my case) and then nothing.  The -I'm talking- LED goes out and nada.  It sounds more like the problem described in "TuneIn not working with Version 1.1 #68" question vice what's generally described above.  I get some of the flash briefing stations to play, but not the ones with a TuneIn intro.  hmmmm....  I followed the TuneIn Radio addons link instructions.

flooie, 2016-05-04:
@GemBro - thanks for the update.  Totally works for BBC Radio London

flooie, 2016-05-09:
@sammachin forgive me - I'm new to this stuff- I've written code that lets you listen to podcasts - or atleast the most recent episode.

flooie, 2016-05-12:
I've pushed code that fixes the tune in issue for the stations I use.  And also lets you use podcasts from tune in - but only the most recent episode.

ProfessionalParanoid, 2016-05-12:
flooie for president!   You nailed it!  couple of minor tweaks for me; had to add the vlc.py (http://git.videolan.org/?p=vlc/bindings/python.git;a=tree;f=generated;b=HEAD) to the directory with my 'main.py' and create a directory called tmpcontent to the AlexaPi directory.  Then NPR Flash brief w/o the tunein intro.  My talking mirror is almost complete.  just got to figure out a motion or gesture trigger vice pressing the button.  Thanks, your upload was very helpful.  flooie's fix in action:  http://other.bagpipez.com/Projects/

flooie, 2016-05-12:
Big update coming.  Thanks for the kind words 

Law Office of William Palin
wpalin@gmail.com
617.272.0554
CambridgeLaw.org

sammachin, 2016-05-13:
@flooie Can you submit a PR with your changes soon I'd like to get them merged before i make any further changes to the branch.

Sam

GemBro, 2016-05-14:
**@flooie**

Interesting ... I'll have a play ... thanks ...

GemBro, 2016-05-16:
**@ProfessionalParanoid** 

You could have a simple iR (infra red) setup by either waving your hand, or placing your hand at/on the mirror, to replace the 'button' ... if you make the iR trigger a short distance (say a few inches) people standing in front of the mirror will not trigger it by mistake ... I see your mirror has a monitor behind it? so the 2 way light passage, for the iR, is possible ...

GemBro, 2016-05-20:
Closing this thread ...

closed 2016-05-20 by GemBro