alexa-pi/AlexaPiDEPRECATED #72: Setup.sh Runs and Hangs never launches web page to get token

fskemery, 2016-05-03:
I've triple checked all of my Amazon credentials for this device on the developers portal and compared them with what appears in the cred file and they match.

This is the last thing I see in terminal after I run setup.sh  Alexa Error log appears below.  Any idea what's wrong here?

Object "=" is unknown, try "ip help".
[02/May/2016:18:01:56] ENGINE Listening for SIGHUP.
[02/May/2016:18:01:56] ENGINE Listening for SIGTERM.
[02/May/2016:18:01:56] ENGINE Listening for SIGUSR1.
[02/May/2016:18:01:56] ENGINE Bus STARTING
CherryPy Checker:
The Application mounted at '' has an empty config.

Alexa Error Log

Starting Alexa...
python: can't open file ' /root/AlexaPi/main.py' : : [Errno 2] Mp such file or directory

I used the installer app and the path on my pi to the main.py file is /home/pi/AlexaPi/main.py

Thanks in advance for any insights you can provide?

sammachin, 2016-05-03:
The setup script isn't the best in version1.0, its improved in 1.1 and should give a few more pointers.

I think there are 2 issues, firstly if you havn't completed the web oAuth process to get a refresh token from amazon into the creds.py file you'll need to do that, you can either re-run the setup or just run `python ./auth_web.py ` which will start the webserver then goto http://[IP OF PI]:5000 to start the oAuth.

The init.d script expects AlexaPi to be installed in /root/AlexaPi which is why its failing to autostart, you can either re-install it there or edit the path in /etc/init.d/Alexa

fskemery, 2016-05-03:
Sam, 

Thanks for the quick response. Much appreciated. 

I was able to retrieve the refresh token using your suggestion and now my creds file is complete. 

I edited the initd_alexa.sh file to reflect my install path /home/pi/AlexaPi/main.py 
and htis is what it looks like now. 

start) 
echo "Starting Alexa..." 
python /home/pi/AlexaPi/main.py & 

;; 

stop) 
echo "Stopping Alexa.." 
pkill -f /home/pi/AlexaPi/main.py\/main\.py 

Rebooted pi and ran python ./auth_web.py and I"m still getting this as a response in termnal. 

pi@raspberrypi:~/AlexaPi $ sudo python ./auth_web.py 
[03/May/2016:12:13:11] ENGINE Listening for SIGHUP. 
[03/May/2016:12:13:11] ENGINE Listening for SIGTERM. 
[03/May/2016:12:13:11] ENGINE Listening for SIGUSR1. 
[03/May/2016:12:13:11] ENGINE Bus STARTING 
CherryPy Checker: 
The Application mounted at '' has an empty config. 

[03/May/2016:12:13:11] ENGINE Started monitor thread 'Autoreloader'. 
[03/May/2016:12:13:11] ENGINE Started monitor thread '_TimeoutMonitor'. 
[03/May/2016:12:13:11] ENGINE Serving on http://0.0.0.0:5000 
[03/May/2016:12:13:11] ENGINE Bus STARTED 

Not sure what I may be doing wrong here? 

Frank

sammachin, 2016-05-03:
> <javascript:_e(%7B%7D,'cvml','fskemery@comcast.net');>>, "Author" <

You don't need to run auth_web.py everytime it's only needed the first time
to sign in.

After a reboot AlexaPi should be ready to go, it auto starts on boot and
when ready you should hear a 'hello' sound through the speaker followed by
3 flashes on the playback LED.

Press and hold your button, speak a command then release.

fskemery, 2016-05-12:
Just got back to this project and connected the buton to the GPIO board. NO leds instaled yet. 

When I push the button I'm getting this error message 

pi@raspberrypi:~/AlexaPi $ python main.py 
Checking Internet Connection 
Connection OK 
Traceback (most recent call last): 
File "main.py", line 151, in <module> 
start() 
File "main.py", line 127, in start 
rf = open(path+'recording.wav', 'w') 
IOError: [Errno 13] Permission denied: '/home/pi/AlexaPi/recording.wav' 

Can you tell me what's wrong with my setup? 

Thanks 

Frank

closed 2016-05-14 by fskemery