alexa-pi/AlexaPiDEPRECATED #74: Alexa seems not recording

ahmadihamid, 2016-05-04:
I have tested my mic using arecord, and I can get it recording but when I short pin no 14 (gnd) and no 18 Alexa not give recording message just stuck at "connection ok" 

I don't have a button yet, could I just short the pin? 
Am I short the wrong pin? 

Huge thanks

GemBro, 2016-05-04:
ahmadihamid ... Pin12 (GPIO 18) needs to be connected to pin 14 (Ground) while you ask Alexa a question ... I think you may have your pins mixed up ... ;) ... 

Your pin18 (GPIO 24) (that you used) is actually an output that goes to one of the LEDs (RED LED) ... pin22 (GPIO 25) is used for the other LED (Green) ... the LEDs give a visual of the state Alexa is in ...

RED LED = Recording (asking a question while button is pressed) and the GREEN LED = Response from Alexa ... you don't need these LEDs for it to work though ...

Yes you can just short the 2 pins together instead of using a button ...

ahmadihamid, 2016-05-04:
Thanks for pointing my mistake, I'm shorting pin 12+14 now, but still don't get the recording message, 

I get resource busy error message if pulse audio volume opened while shorting pin, and alexa terminated, so I think I'm shorting right pin now
XD

ahmadihamid, 2016-05-04:
*pulse audio volume control

chilinski, 2016-05-04:
What are you using as a microphone?

ahmadihamid, 2016-05-04:
USB microphone

chilinski, 2016-05-04:
I use the one from Plugable and had to make a couple of changes, which I don't have with me right now at work. But I believe you need to change line 19 in main.py from:
device = "plughw:1" # Name of your microphone/soundcard in arecord -L
to
device = "Device"

I'm getting ready to leave work now, so I can check my settings when I get home and check in back here to see if just changing it to "Device" works for you. If you run arecord -L it will list the name you need to use.

ahmadihamid, 2016-05-04:
my arecord -L : 
plughw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Hardware device with all software conversions

I've tried : 
plughw:1 (no error not recording)
plughw:CARD=Device (no error not recording)
plughw:CARD=Device,DEV=0 (no error not recording)
Device (error : root@raspberrypi:~/AlexaPi# python main.py 
Checking Internet Connection
Connection OK
ALSA lib conf.c:4705:(snd_config_expand) Unknown parameters CARD=Device
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=Device
Traceback (most recent call last):
  File "main.py", line 153, in <module>
    start()
  File "main.py", line 119, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Invalid argument
)

chilinski, 2016-05-04:
First off, I am using the music version of this from Lenny's gitpage, but I don't think that should make a difference. I am using the Kinobo mic shown here http://www.amazon.com/Kinobo-Microphone-Desktop-Recognition-Software/dp/B00IR8R7WQ?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_detailpage_o02_s00. 

Okay, my arecord -L shows: plughw:CARD=Device, DEV=0 exactly the same as yours.
The line in main.py that says: device = "plughw:1" is changed to device = "Device" .

Please try it with just  device = "Device" and leave out all the plughw stuff. Just device = "Device" .

ahmadihamid, 2016-05-05:
Tried device = "Device", still no luck 
No recoding message 

Is there any conflict with pulseaudio package?

ahmadihamid, 2016-05-05:
I don't know why. I can hear "alexa's not so cute voice now" 
It works! Huge thanks all!

Using plughw:1

closed 2016-05-05 by sammachin