alexa-pi/AlexaPiDEPRECATED #75: Change Location / Settings

medoix, 2016-05-04:
Thank you so much for this awesome project. Works a treat hardware wise./

Is it at all possible to change the default location from Seattle to say Sydney? Hopefully this may also change the temp report from Fahrenheit to Celsius.

sammachin, 2016-05-05:
Not really, the location of each device is set in the alexa dashboard http://alexa.amazon.com however you can currently only select a US ZIP code.

Some people have successfully hacked th request to change the timezone although I've not managed that yet.

You can get local requests by adding the place name eg "Whats the weather tomorrow in Sydney" or "What time is it in London"

You can switch on 'metric' measurements in the settings tab on each device which might help.

closed 2016-05-05 by sammachin

doubledrat, 2016-05-05:
I was able to set the timezone to London by following this -

https://devdiner.com/answers/setting-timezone-amazon-echo-outside-us