alexa-pi/AlexaPiDEPRECATED #76: CherryPy Checker: The Application mounted at '' has an empty config.

FatherStorm, 2016-05-04:
I have confirmed tat AlexaPi is installed in the correct location, and have even tried on a fresh install of Raspbian on both a Pi 2 and Pi 3. 

runing auth_web.py never launches the browser,and when I manully open the browser to port 5000, and log in with my credentials, I get
 

500 Internal Server Error

The server encountered an unexpected condition which prevented it from fulfilling the request.

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/_cprequest.py", line 670, in respond
    response.body = self.handler()
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/lib/encoding.py", line 217, in __call__
    self.body = self.oldhandler(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/_cpdispatch.py", line 60, in __call__
    return self.callable(*self.args, **self.kwargs)
  File "./auth_web.py", line 35, in code
    line = 'refresh_token = "{}"'.format(resp['refresh_token'])
KeyError: 'refresh_token'

FatherStorm, 2016-05-05:
this finally resolved after removing second wireless adapter. first run succeeded, but after reboot, AlexaPi service wont lad. rerunning auth_web.py generates

insserv: warning: script 'S99alexa' missing LSB tags and overrides
insserv: warning: script 'alexa' missing LSB tags and overrides
insserv: There is a loop at service alexa if started
insserv: There is a loop between service plymouth and mountnfs if started
insserv:  loop involving service mountnfs at depth 7
insserv:  loop involving service networking at depth 6
insserv: There is a loop between service plymouth and mountnfs if started
insserv: There is a loop between service plymouth and mountall-bootclean if started
insserv:  loop involving service mountall-bootclean at depth 6
insserv:  loop involving service mountall at depth 5
insserv: There is a loop between service plymouth and mountall if started
insserv: There is a loop between service alexa and checkroot if started
insserv:  loop involving service checkroot at depth 3
insserv:  loop involving service hdparm at depth 2
insserv: There is a loop between service alexa and udev if started
insserv:  loop involving service udev at depth 1
insserv: Starting alexa depends on plymouth and therefore on system facility `$all' which can not be true!

sammachin, 2016-05-05:
The setup script isn't great in the current version, auth_web is only run once during install in order to complete the auth process, however if you need to re-auth for any reason you can run it manually from the command line `python auth_web.py`

The setup is improved in 1.1 which you might find easier to install, although its not quite 100% yet most of the issues are around the music playback which just doens't work at all in 1.0, give the 1.1 branch a try on a clean install.

dstinebaugh, 2016-05-16:
I actually get the page to load, however the refresh_token never get's generated and amazon's page returns an error (with no helpful output) so the token is never appended to the creds file. I'd love to check out 1.1 when it's ready!

REHolt, 2016-05-29:
dstinebaugh: i get the exact same error. I got this working on a RasPi B+, but when i try to install it on my RasPi 3 B, with start on boot, I get this same error. I have tried looking at the code and modifying some things but no joy. Hope 1.1 comes along soon:).

REHolt, 2016-05-29:
OK, I solved my problem. Turns out I forgot to go back to the "Web Settings" and change the local IP when I changed Raspberry Pi's. I also needed to get rid of the "https" in favor of just "http". it all works well now. I found the 1.1 version (couldn't see it or download it from the RasPi). I downloaded on my PC, copied it to a flash drive and installed in on the RasPi. 

Thanks for taking the time to put this together.

maso27, 2016-05-29:
You can use the "--branch" option in git clone to download a different branch.

renekliment, 2016-10-26:
Seems solved -> closing. Reopen if needed.

closed 2016-10-26 by renekliment