alexa-pi/AlexaPiDEPRECATED #78: AlexaPi on Win7 desktop?

mrx23dot, 2016-05-06:
Would it be possible to run AlexaPi on Win7 desktop?
that would be epic!

Hoki-gcav, 2016-05-06:
Not this version, because windows doesn't support all the libraries involved in making this. At some point some will make Alexa for Windows, but it won't be coded in python.

lennysh, 2016-05-06:
I have a version written in .Net, but it's not been upgraded to the new API yet...

sammachin, 2016-05-07:
Not easy to get this up & running on Win7 mainly due to the audio limitations and also the reliance on external GPIO buttons for control.

I have built a browser based version though which is using similar code to version 1.0, you can use it for free at https://alexaweb.herokuapp.com or download and run your own from http://github.com/sammachin/alexaweb

closed 2016-05-07 by sammachin

lennysh, 2016-05-07:
My version uses an on screen button, so mouse required.  Runs fine using VLC on Windows.