alexa-pi/AlexaPiDEPRECATED #79: "Alexa Stop" equivalent?

Hoki-gcav, 2016-05-06:
Lets say you ask alexa "How many calories are in a carrot" she will begin waffling on for about a minute going into vast detail over all the different variations of carrots and you just want to shut her up, there isn't really a way of doing that (to my knowledge) apart from muting my amp with the remote.

Maybe make it so that if the button to record is single pressed it will just stop, and like when turning off the radio, you can stop her from talking and get her to answer a new question if you press and hold down the record button.

This would be a nice feature, it seems to get used alot with the echo's from what I've seen in demos of it.

gregwa1953, 2016-05-06:
You can also tell her to "Stop", at least I've been able to.

GemBro, 2016-05-06:
**@ gregwa1953**

How did you achieve that with AlexaPi? ... pressing the button again does not stop Alexa ... also, pressing the button and sayin' 'stop' does nothing either ... but pressing the button again does stop audio music & radio streams, it then resets to wait for another command 

I think we maybe able to add a STOP feature in 'main.py' code, using another seperate button ... just needs to be in the same 'listening' loop, I think? ...

Hoki-gcav, 2016-05-06:
Id prefer it if it used the same button and just interrupted the while statement that stops it from going back to "Recording." state while Alexa talks.

GemBro, 2016-05-06:
Or that ;) ...

gregwa1953, 2016-05-07:
It was under the "other" Alexa (the java version).  I simply pressed the
"start listening" button on the screen and said "Stop".

I assumed that it would work on this version as well.

GemBro, 2016-05-07:
**@ gregwa1953**

Ahh ok ... no it doesn't seem to work on the Python version (at least not on 1.1) ... thanks for verifying ;) ...

gregwa1953, 2016-05-08:
@GemBro, sorry about that.

GemBro, 2016-05-09:
**@ gregwa1953**

Hey no worries ... glad we know which one it does work on ;) ...

ProfessionalParanoid, 2016-05-14:
I am running a version of this that flooie put together to address a url parsing issue, and I can stop by pressing the button a second time.  flooie's push is described under the TuneIn so close Issue#71.  My Alexa in action with stop: http://other.bagpipez.com/Projects/

flooie, 2016-05-14:
@ProfessionalParanoid I figure - if you're pushing the button you probably want to tell it something -

flooie, 2016-05-14:
@ProfessionalParanoid also the code i pushed is still missing a bunch of stuff -

flooie, 2016-05-15:
@ProfessionalParanoid - I think my new updated version handles all the issues (besides wake word) if someone is willing to try it out - I made it version1.1.1 - also this is my first time contributing to code online - so please let me know if I'm doing anything wrong - @GemBro @sammachin 

it still needs some clean up but it seems to handle everything I've thrown at it.

GemBro, 2016-05-16:
**@flooie** 

Sounds great mate ... I'll give it a go ... have you got a link to the code? ...

On the other note ... I'm not a moderator for any of the AlexaPi issues or threads ;) ... I'm just an active (and excited) person loving this project ...

But you could start your own GIT page and include your code in it ... similat to what Sam & Lenny have done ... then you can have your own thread system for feedback and possible inclusion into Sam's AlexaPi ... the others will give you pointers on how to go about this as I'm pretty new to GIT too ;) ...

Appreciate your input though ... although I'm I.T. orientated I wish I was a bit more confident in coding ... but I.T. & Electronic Design is my fung fu ... this is what has drawn me to this project ... :) ...

flooie, 2016-05-16:
@GemBro https://github.com/flooie/AlexaPi/tree/version1.1.1

GemBro, 2016-05-16:
**@flooie**

Thanks mate ;) ...

GemBro, 2016-05-16:
**@flooie**

I have just tried your 1.1.1 version and I'm still having a problem with my radio station ... as in getting an error ;) ... I get the same error I get with Sam's 1.1 ... it wants to play my station but the URL is wrong and so bombs out (or seems to hang) ... a CNTRL X get's it back to cursor ... 

As you can see it hasn't refactored the URL ... which needs to be either:

http://st8.webradioworld.net:8008
or
http://st8.webradioworld.net:8008/listen.pls

The "listen.pls" on the end is normally required for most shoutcast streams that are on dedicated servers ... WinAmp, iTunes and VLC use this for reconising audio streams ...

Does my station play for you? ...

**ERROR REPORT:**

Ready to Record.
Recording...

- I ask a command: "**play ... J ... F ... S ... Radio**" ...

Recording Finished.

Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Music#ACRI#DeviceTTSRendererV4_48078da5-0601-49f3-9c77-96704716033c", "audioContent": "cid:DeviceTTSRendererV4_48078da5-0601-49f3-9c77-96704716033c_1392776635"}}, {"namespace": "AudioPlayer", "name": "play", "payload": {"playBehavior": "REPLACE_PREVIOUS", "audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 900000}, "streamUrl": "http://opml.radiotime.com/Tune.ashx?id=s111822&formats=aac,mp3&partnerId=!EALLOjB&serial=AFIHU5JRSUSO3FIXP5L7MSTDFWQQ&latlon=33.94211,-118.41749", "offsetInMilliseconds": 0, "expiryTime": "2016-05-16T16:12:39+0000", "streamId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#c5492308-7e3e-4d9f-a5d5-288ce04d4a7b:1", "progressReportRequired": true}], "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#c5492308-7e3e-4d9f-a5d5-288ce04d4a7b:1"}, "navigationToken": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#c5492308-7e3e-4d9f-a5d5-288ce04d4a7b:1"}}]}, "messageHeader": {}}
Play_Audio Request for: /home/pi/AlexaPi/tmpcontent/DeviceTTSRendererV4_48078da5-0601-49f3-9c77-96704716033c_1392776635.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "main.py", line 265, in play_audio
    file = tuneinplaylist(file)
  File "main.py", line 256, in tuneinplaylist
    r = requests.get(req.content)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 60, in get
    return request('get', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/api.py", line 49, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 443, in request
    prep = self.prepare_request(req)
  File "/usr/lib/python2.7/dist-packages/requests/sessions.py", line 374, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 304, in prepare
    self.prepare_url(url, params)
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 354, in prepare_url
    scheme, auth, host, port, path, query, fragment = parse_url(url)
  File "/usr/lib/python2.7/dist-packages/urllib3/util/url.py", line 143, in parse_url
    raise LocationParseError(url)
LocationParseError: Failed to parse: st8.webradioworld.net:8008
http:

... hanging ... CNTRL X get's it back to cursor ...

flooie, 2016-05-16:
@GemBro try it now

GemBro, 2016-05-16:
Is this version 1.1.1b now? ... :) ... can I just overwrite or edit your previous **main.py** file? ... I'll have a play now ...

GemBro, 2016-05-16:
**@flooie**

Ok cool that worked :) ... nice work mate ... my funky station is playing ...

Those number (you have parsed in the code), 8008, 9008, etc, are the shoutcast stream port numbers ... there are many different ones and they can change for different stations ... that might cause a headache trying to add all of them ... ;) ... but it's a start ...

Now for the bad news ... (UK) BBC Radio 2, etc, do not work now ... as with BBC Radio London ... they go into a loop ... they do work fine with v1.1 though ... Kiss FM still works ... so maybe the code has gone way side for the UK stations? ...

Solved one ... but it did break others (as you said hoped it wouldn't) ... ;) ... but I think you're onto something flooie ;) ... this is probably similar to what Sam is doing? ...

flooie, 2016-05-16:
@GemBro - try now

GemBro, 2016-05-16:
That's the one flooie :) ... I checked the others and they are fine too (still) ... nice job ...

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment