alexa-pi/AlexaPiDEPRECATED #80: Timers and Alarms?

benwoodley, 2016-05-09:
I'm uncertain of how to add this on amazons API end, but I'd sure love to be able to set alarms and timers with Alexa. I use timers a lot for keeping my life organised and since I have Alexa quite near to where I sleep it would be convenient to use it as an alarm clock. I'm not sure what would be required to implement this functionality but it'd sure be interesting to see it work.

When currently asking Alexa to set an alarm there is no error of any kind just a message saying alarms are not supported, does this happen for anyone else?

GemBro, 2016-05-09:
**@Ben**

Yes it does ... to all versions of the Python version of AlexaPi ... for me anyhow ...

Amazon's Alexa app (alexa.amazon.com) doesn't seem to like the Python version much ... I have registered practically all the music apps but TuneIn seems the only one that works (I think that worked before I added that tbh) ...

I cannot set anything else in the App ...

I think the App only recognises Amazons' own JAVA AVS of Alexa and therefore has a lot more functions running then the Python version ...

I cannot get anything setup with that App for AlexaPi ... I'm waiting in the wings but designing some electronics for my Alexa, so that it's all ready for when (hopefully if & when) the guys are able to solve some more of the issues ... don't need everything but some extras will be nice ;) ...

tube0013, 2016-05-09:
Sam's Python version is based on the original api, which was limited and did not support timers and alarms, amongst other things.

The Java client Amazon released is based on the newer api.  Sam has said he's working on implementing that in a version 2 of his app.  The api change was fundamentally pretty big, and pretty much requires a full rewrite.  And dependency on other components like http2 client.

GemBro, 2016-05-09:
**@tube0013**

Sorry I didn't make that clear ... but we know Sam's working on v2 ... ;) ...

benwoodley, 2016-05-10:
Interesting.. well can't wait to see it working with the new API :) I'm probably gonna try out the java version at some point but I'll never switch from the python version on my main pi (Which is now welded to the bottom to the desk).

sammachin, 2016-05-10:
Yeah timers are in the v2016 api which will be the basis of AlexaPi V2, currently I'm still trying to wrangle the music playback stuff in v1.1 to work nicely but this code will also then be carried over to v2.

Don't really have a timescale for v2 yet, it depends on how much free time I've got, I'm off to the West Coast US for 2 weeks at the end of the month, so I'll either have loads of free time bored in hotels to work on or be rushing around with work!

closed 2016-05-10 by sammachin

gregwa1953, 2016-05-10:
Sam,
If you need help, please contact me.  I'll be glad to see what I can do.

Gregwa

flooie, 2016-05-18:
@gregwa1953 @sammachin - I'd also love to help get timers and alarms working - please let me know how I can help

amahlaka, 2016-06-19:
Here is  the migration guide to api v 2 https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/docs/migration-guide

suryapa1, 2018-02-25:
Hi flooie,

Could you share some example on timer/remainder setting in Node/Python version.

email me @ suryapa1@gmail.com