alexa-pi/AlexaPiDEPRECATED #81: alex function API response returning 204?

zachnoren, 2016-05-10:
Hello, I have been trying to debug this for a few hours now. Can confirm the microphone is working, plugged in credentials correctly, install script ran fine with no errors. Wasn't getting any response from Alexa so decided to print r.status_code from within alexa function. That is how I found out what was actually happening. It appears the API is responding with a 204. Running main.py as root. Has anyone had the same issue can anyone think of how to resolve this?

sammachin, 2016-05-10:
204 is the normal 'null' response, usually I get this when sending a very short or blank audio clip.

Its worth checking that you are recording the whole command, (remember to hold the button down while speaking) you should be able to listen to  the last recorded command sent to alexa in recording.wav

zachnoren, 2016-05-10:
I will give this a try later and reply back. Thanks

zachnoren, 2016-05-10:
Fixed faulty microphone. Thanks

closed 2016-05-10 by zachnoren